ghdl/ghdl #560: Function overloading not finding correct version of function.

CraigNoble, 2018-04-24:
As far as I know, this should be perfectly valid VHDL, but GHDL is throwing an error with it.

```
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

package reproducer_pkg is

    -- Functions
    function MIN(LEFT, RIGHT: unsigned) return unsigned;
    function MIN(LEFT, RIGHT: integer)  return integer;

end reproducer_pkg;

package body reproducer_pkg is
    
    function MIN(LEFT, RIGHT: unsigned) return unsigned is
    begin
        if LEFT < RIGHT then
            return LEFT;
        else
            return RIGHT;
        end if;
    end;
    
    function MIN(LEFT, RIGHT: integer) return integer is
    begin
        if LEFT < RIGHT then
            return LEFT;
        else
            return RIGHT;
        end if;
    end;
    
end reproducer_pkg;    
```

```
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

library work;
    use work.reproducer_pkg.all;

entity reproducer is
    port(
		inputA		: in  unsigned(7 downto 0);
		inputB		: in  unsigned(7 downto 0);
		inputC		: in  integer;
		inputD		: in  integer;
		OutputA		: out unsigned(7 downto 0);
		OutputB		: out integer
    );
end reproducer;

architecture rtl of reproducer is
begin

	OutputA <= min(inputA, inputB);
	OutputB <= min(inputC, inputD);
    
end rtl;
```

> ghdl -a --workdir=.\work --std=93 --no-vital-checks --ieee=synopsys .\reproducer_pkg.vhd

> ghdl -a --workdir=.\work --std=93 --no-vital-checks --ieee=synopsys .\reproducer.vhd
.\reproducer.vhd:22:22: cannot match function "min" with actuals
.\reproducer.vhd:22:23: can't associate 'inputa' with constant interface "left"
.\reproducer.vhd:22:23: (type of 'inputa' is a subtype of unsigned)
.\reproducer_pkg.vhd:9:18: (type of constant interface "left" is integer)

go2sh, 2018-04-24:
This actually might be a bug in ghdl as the LRM specifies in 4.2.2.1:
> The actual designator associated with a formal of class constant shall be an
expression.
`inputA` is a valid expression, so it should be fine. Further it's specified, that only the value is transferred into the subprogram for constants and variables.

tgingold, 2018-04-24:
Interesting case.  No, this is not a valid VHDL source, as the 'min' functions conflict with the 'min' physical unit.
However, there is also a bug in ghdl hence the weird error message.
I am trying to fix it.

tgingold, 2018-04-24:
I am testing a fix.  The error message is now:

```
reproducer.vhdl:55:19:error: no declaration for "min" (due to conflicts)
reproducer.vhdl:56:19:error: no declaration for "min" (due to conflicts)
```

Paebbels, 2018-04-24:
@tgingold  
Quartus has also naming problems with `min` as a function name, because type time uses `min` for secondary unit minute. According to ModelSim/QuestaSim, Active-HDL, Riviera-PRO, this is not a naming clash in VHDL. The language shall distinguish these cases, right?.

**On the other hand see this example:**

    signal temp : time;
    temp <= min;

**Interpretations:**
1. Physical units are allowed without an integer or real literal in front, meaning `1 min`.
2. `min` could be a parameterless function returning type `time`.

tgingold, 2018-04-25:
No, this is an error.  You can only overload subprogram (and enumeration literal) names.

Hence the reason why vhdl use minimum/maximum names instead of min/max.

Paebbels, 2018-04-25:
Ok, then I will file this as a bug in modelsim and Riviera :)

go2sh, 2018-04-25:
@tgingold Wouldn't it be nice, if the error is thrown at the declaration. Like with enumeration literals.

CraigNoble, 2018-04-25:
So, I'll just point out that ieee.numeric_std has a min function within its package body:
https://standards.ieee.org/downloads/1076/1076.2-1996/numeric_std-body.vhdl

Assuming that this is indeed invalid VHDL wouldn't that imply a bug within this package also?

Paebbels, 2018-04-25:
@JimLewis  
Please review this issue discussion for a possible bug in the LRM regarding function `min` in package `IEEE.numeric_std`.

tgingold, 2018-04-25:
@CraigNoble: no, it is fine to declare a min function.  The problem arises when you have a use clause (like use work.mypkg.all), as it makes the 'min' function potentially visible like the 'min' unit.

tgingold, 2018-04-25:
@go2sh only subprograms and enumeration literals can be overloaded; not object declarations.

tgingold, 2018-04-25:
@Paebbels no, there is no issue with the min function in numeric_std package body.

closed 2018-04-25 by tgingold

go2sh, 2018-04-26:
@tgingold Just 3 things: 
Since `time` is part of the `standard` package it is always visible. Even if you declare it in a package. So from what you have written, it shouldn't be possible at all to declare a min function as it always conflicts with the min unit. Or do I get something wrong?
I just reread the LRM and though that overloading a name with a subprogram and an enumeration literal might be wrong. Either its an enumeration literal or a subprogram, but both makes no sense and I didn't find any reading that this is allowed. Though, all simulators allow it.
Lastly, shouldn't it be allowed to overload units? I did come up with some examples where certain unit names might collide.

tgingold, 2018-04-26:
No, time is not always visible.  You can write `constant time : natural := 5;`, which hides the `time` declared in std.standard.

An enumeration literal is considered like a function. See LRM2018 5.2.2 Enumeration types.

Lastly, no: only subprograms and enum lit can be overloaded.  But you can override any name.

commit 1d7382e7 (2018-04-25): sem_scopes: correctly handle conflicting potential visibility.

Fix #560