ghdl/ghdl #561: error: waves.write_types: unhandled obj kind

slaweksiluk, 2018-04-26:
Hello, when I try dump data to ghw file in following design:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

package wishbone_pkg is

--./wb_demux_tb:internal error: waves.write_types: unhandled obj kind
type t_wishbone_slave_in is record
	dat : std_logic_vector;
end record;

--./wb_demux_tb:internal error: wave.create_type
--type t_wishbone_master_out is record
--	dat : std_logic_vector;
--end record;
--subtype t_wishbone_slave_in is t_wishbone_master_out;

end wishbone_pkg;

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
--use IEEE.NUMERIC_STD.ALL;

library work;
use work.wishbone_pkg.all;

entity wb_demux is
    Port (
    	wbs_i	: in t_wishbone_slave_in
    );
end wb_demux;
architecture full_regs of wb_demux is
begin

end full_regs;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.wishbone_pkg.all;

entity wb_demux_tb is
end entity;

architecture bench of wb_demux_tb is

	signal wbs_i : t_wishbone_slave_in(
		dat(32-1 downto 0)
	);

begin

	stimulus : process
	begin
		wbs_i.dat <= x"deadbeef";
		wait for 100 ns;
		report "pass" severity failure;
		wait;
	end process;

	dut : entity work.wb_demux
	port map (
		wbs_i => wbs_i
	);

end architecture;

```
```
ghdl -a --std=08 wishbone_pkg-2008.vhd
ghdl -a --std=08 wb_demux.vhd
ghdl -a --std=08 wb_demux_tb.vhd
ghdl -e --std=08 wb_demux_tb
echo "without ghw"
./wb_demux_tb
echo "with ghw"
./wb_demux_tb --wave=w.ghw
```
I receive following output:
```
--./wb_demux_tb:internal error: waves.write_types: unhandled obj kind
```
There is another error when I try to use subtype code in package:
```
--./wb_demux_tb:internal error: wave.create_type
```
Problem is related to unconstrained record use. Please find complete reproducer in my repo:
[reproducer](https://github.com/slaweksiluk/misc/tree/master/ghdl-playground/wave-create-error)

Paebbels, 2018-04-26:
This is related to issues:
* #453 - ghdl-llvm crash when compiling uvvm
* #459 - Constraining partial constrained types
* #473 - Unconstrained arrays; crash on elaboration
* #546 - Internal error when given multiple range constraints and initializer

slaweksiluk, 2018-08-10:
You're welcome and thank you for maintaining ghdl - still much faster support than in commercial simulators.