ghdl/ghdl #562: Misleading elab error

kraigher, 2018-04-26:
**ent.vhd**
```vhdl
entity ent is
  generic (
    max : natural);
  port (
    p : out natural range 1 to max := 3);
end entity;

architecture a of ent is
begin
end;
```

**ent2.vhd**
```vhdl
entity ent2 is
end entity;

architecture a of ent2 is
  constant max : natural := 2;
  signal p : natural range 1 to max;
begin

  inst : entity work.ent
    generic map (max => max)
    port map (p => p);
end;

```

**shell**
```
+ ghdl --version
GHDL 0.36-dev (v0.35-139-g0edf0a10) [Dunoon edition]
 Compiled with GNAT Version: 7.3.1 20180303 (Red Hat 7.3.1-5)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
+ ghdl -a ent.vhd
+ ghdl -a ent2.vhd
+ ghdl --elab-run ent2
ghdl:error: bound check failure at ent2.vhd:5
  from: work.ent2(a).inst.CMP_ELAB at ent2.vhd:5
ghdl:error: error during elaboration
```

The error is reported as occuring on ent2.vhd:5 but should be reported on ent.vhd:5

tgingold, 2018-04-27:
I can reproduce the issue.

kraigher, 2018-05-01:
I do not agree on the classification as "enhancement". The reproducer can be misleading regarding the error since ent2.vhd:5 seems related to ent.vhd:5. This is however just coincidence. If I insert 10 newlines above line 5 in ent2.vhd it will still complain about ent2.vhd:5 which is then only a empty line. Clearly the code has mixed up the file name and line numbers which I would classify as a bug.

OmarElfouly, 2024-10-23:
hello @tgingold and @Paebbels 
I tried to replicate this issue and it seems to have resolve itself as the error it reports is now:

```vhd
ent.vhd:5:39:error: value out of range
    p : out natural range 1 to max := 3);
                                      ^
```

Full terminal:
```
root@8df63971cf2d:/usr/testingElab# ghdl --version
GHDL 5.0.0-dev (4.1.0.r271.gec95e9130) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
root@8df63971cf2d:/usr/testingElab# ghdl -a ent.vhd    
root@8df63971cf2d:/usr/testingElab# ghdl -a ent2.vhd 
root@8df63971cf2d:/usr/testingElab# ghdl --elab-run ent2
ent.vhd:5:39:error: value out of range
    p : out natural range 1 to max := 3);
                                      ^
```

Steps followed:

Used the following Dockerfile to build ghdl from source then ran the commands mentioned in the issue on the files provided

```Dockerfile
# Use an official Ubuntu image as the base
FROM ubuntu:latest

# Install dependencies required for building GHDL
RUN apt-get update && \
    apt-get install -y gnat zlib1g-dev build-essential git make libgnat-10 g++ gcc

# Set the working directory for the build process
WORKDIR /usr/src/app

# Copy the source code from your current working directory into the container
COPY . .

# Run the build steps for GHDL from source
RUN ./configure --prefix=/usr/local && make

RUN make install

# Default command: open a shell so you can pass terminal commands after GHDL is built
CMD ["/bin/bash"]

```

dalex78, 2026-08-17:
# Issue #562 — Misleading elaboration error location

Analysis by an AI agent. Recommend (optionally merging the new test and) closing the ticket.

## High-level summary

`ent` declares a port whose default value (`3`) can be out of range once its `natural range 1 to max` constraint is resolved by a generic (`max`) passed in from an instantiating entity `ent2`. When that happens, GHDL correctly detected the bound-check violation during elaboration, but reported it at the instantiation site (`ent2.vhd:5`, the `inst : entity work.ent ...` line) instead of at `ent.vhd:5`, where the actual out-of-range default value is written. That made the error message point the user at the wrong file entirely.

## What was actually wrong in GHDL's implementation

Not applicable anymore — see below.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the issue (`ent.vhd` + `ent2.vhd`, `ghdl -a`, `ghdl -a`, `ghdl --elab-run`) against current `master`. The error is now reported correctly:

```
ent.vhd:5:39:error: literal out of range
    p : out natural range 1 to max := 3);
                                      ^
```

This points exactly at the offending literal `3` in `ent.vhd`, not at `ent2.vhd` as originally reported. The bound check itself, and its location tracking through generic-resolved subtype constraints, must have been fixed as part of general elaboration/error-reporting improvements since this was reported in 2018 (`GHDL 0.36-dev`). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue562/` (the two original source files plus a `testsuite.sh` regression test) that elaborates `ent2`, expects failure, and asserts the reported error location starts with `ent.vhd:5` and never with `ent2.vhd:`. This locks in the correct behavior and will catch any future regression that reintroduces the wrong-file error attribution.

closed 2026-08-17 by tgingold

dalex78, 2026-08-17:
There is an issue here. The commit seems to have been merged before the CI finishes, and the CI fails. The AI agent seems to have tested the fix only with mcode, like the ticket reports, where the error does not exist anymore. But the error still exists with GCC and LLVM backend.

The issue needs to be reopened I guess.

The AI agent is working on a fix.

## AI agent analysis of that remaining issue

What actually explains your failure signature: the `./ent2:error: bound check failure at ent2.vhd:5` / `instance:` format is characteristic of a runtime GRT bound-check from a compiled (gcc/llvm) backend executable — not mcode's `--elab-run`, which never prints a `./ent2` prefix. I confirmed this directly: running the exact same `ent.vhd`/`ent2.vhd` pair through the existing LLVM-backend GHDL build (`~/software/ghdl/llvm/issues-work`) reproduces your exact failure output byte-for-byte.

So the real story looks like: issue562's bound-check location bug is not actually fixed in general — it only reads as "fixed" on the mcode backend, because mcode's elaborator statically folds the generic-resolved default expression and catches it early (correctly, at `ent.vhd:5`). On gcc/llvm, that same check is deferred to a runtime bound-check inside the generated code, which misattributes the error to the instantiation context (`ent2.vhd:5`) — i.e., the original issue562 bug is still live there.

dalex78, 2026-08-17:
Sorry about that. I'll try to run all the tests with all the backends before proposing a fix.
I also suggest we don't merge before the CI finishes.

dalex78, 2026-08-18:
# Issue #562 — Misleading elaboration error location

Analysis by an AI agent. Recommend (optionally merging the new test and) closing the ticket.

## High-level summary

`ent` declares a port whose default value (`3`) can be out of range once its `natural range 1 to max` constraint is resolved by a generic (`max`) passed in from an instantiating entity `ent2`. When that happens, GHDL correctly detects the bound-check violation during elaboration, but the runtime error was reported at the instantiation site (`ent2.vhd:5`) instead of at `ent.vhd:5`, where the actual out-of-range default value is written.

## The issue does not reproduce on mcode — but is real on gcc/llvm

An earlier pass on this branch tested only against the `mcode` backend, found the location reported correctly, and concluded the bug was "already fixed on master" (see prior commit history / PR #3341). That conclusion was wrong: it only holds for `mcode`.

- **mcode**: `ghdl --elab-run` folds the generic-resolved bound statically during elaboration and reports the violation *before* simulation even starts, as a plain analysis-style message:
  ```
  ent.vhd:5:39:error: literal out of range
      p : out natural range 1 to max := 3);
                                        ^
  ```
  This message comes from a completely different code path (`src/synth/synth-vhdl_expr.adb`'s static elaborator) that never touches the buggy code below. The bug is simply never reached on this backend for this test case.

- **gcc / llvm**: `ghdl -e` compiles a design executable; the same violation is instead caught by a *runtime* bound check embedded in the compiled code, reported by GRT as:
  ```
  ./ent2:error: bound check failure at ent2.vhd:5
    instance:
  ./ent2:error: error during elaboration
  ```
  Confirmed by direct reproduction against both a stock LLVM-backend build and the exact source on this branch: the file is wrong (`ent2.vhd` instead of `ent.vhd`); the line number (`5`) happens to be right only because, by coincidence in this particular repro, the offending literal
  in `ent.vhd` and `ent2.vhd`'s own last declaration before the mis-blamed point are both on line 5.

## What is actually wrong in GHDL's implementation

`src/vhdl/translate/trans-chap6.adb`, `Gen_Bound_Error` (and, identically, `Gen_Direction_Error` and `Gen_Program_Error`):

```ada
procedure Gen_Bound_Error (Loc : Iir) is
   Constr    : O_Assoc_List;
   Name      : Name_Id;
   Line, Col : Natural;
begin
   Files_Map.Location_To_Position (Get_Location (Loc), Name, Line, Col);

   Start_Association (Constr, Ghdl_Bound_Check_Failed);
   Assoc_Filename_Line (Constr, Line);   -- only Line is passed on!
   New_Procedure_Call (Constr);
end Gen_Bound_Error;
```

It correctly computes `Name` (the file `Loc` actually belongs to) via `Files_Map.Location_To_Position`, but then **discards `Name`** and only forwards `Line` to `Assoc_Filename_Line`. That helper (`src/vhdl/translate/trans-helpers2.adb`) embeds the filename from `Current_Filename_Node` — a global that always points at whichever design
unit's code is *currently being generated*, not at the file `Loc` came from:

```ada
procedure Assoc_Filename_Line (Assoc : in out O_Assoc_List;
                               Line  : Natural) is
begin
   New_Association
     (Assoc, New_Address (New_Obj (Current_Filename_Node), Char_Ptr_Type));
   ...
```

For plain same-file checks (the overwhelming majority of call sites), `Loc` and the unit currently being compiled are always the same file, so this never shows up. It breaks specifically when the check's `Loc` belongs to a *different* file than the unit whose code is being generated right now — exactly this case: `p`'s default-value check is code-generated as part of `ent2`'s own elaboration procedure (because `port map (p => p)` makes GHDL collapse `ent`'s port `p` and `ent2`'s signal `p` into one signal, created in the instantiating unit's context), while `Loc` correctly points into `ent.vhd`. The line number survives (it travels through the call explicitly); the filename doesn't (it's read from the wrong ambient global) — hence "right line, wrong file".

## What the fix does

`src/vhdl/translate/trans-helpers2.ads`/`.adb`: added an overload, `Assoc_Filename_Line (Assoc : in out O_Assoc_List; Loc : Iir)`, that derives the filename from `Loc`'s own location (via `Files_Map.Location_To_Position` and a freshly created private string constant with Create_String`) instead of the ambient `Current_Filename_Node`.

`src/vhdl/translate/trans-chap6.adb`: `Gen_Bound_Error`, `Gen_Direction_Error` and `Gen_Program_Error` now call the new `Loc`-based overload instead of `Assoc_Filename_Line (Assoc, Get_Line_Number (Loc))`, so the filename embedded in the generated runtime check always matches the line number.

`testsuite/gna/issue562/testsuite.sh`: loosened the two `grep` assertions from anchored, mcode-specific patterns (`^ent.vhd:5:`) to plain substring matches (`ent.vhd:5` / `ent2.vhd:`), since the *correct* message legitimately has a different shape per backend (mcode's static
`ent.vhd:5:39:error: literal out of range` vs. gcc/llvm's runtime `./ent2:error: bound check failure at ent.vhd:5`) — the test should accept either, and only reject the file actually being wrong.

## Testing

Full local testsuite (sanity, gna, synth, vpi, vhpi, vests) run against all three backends, each built from this branch:

[... cut ...] t from unmodified master — a pre-existing environment limitation (a standalone C program can't link directly against this environment's `libghdl.so`), unrelated to this fix.
- **llvm**: `gna/issue562` confirmed fixed — now reports `ent.vhd:5` instead of `ent2.vhd:5`. All suites pass except the same pre-existing `gna/issue2005`.
- **gcc**: `gna/issue562` confirmed fixed, same as llvm. All suites pass except the same pre-existing `gna/issue2005`.

No other regressions on any of the three backends.

dalex78, 2026-08-18:
Note that Ì'll propose a fix for `issue2005` in another PR.