ghdl/ghdl #564: Compiling vendor libraries

Defossez, 2018-04-30:
Hi,

I try to compile Xilinx Vivado libraries on PC running Windows-10.
Vivado_2018.1 is installed and fully functional.
Change directory to <GHDL install>\lib\vendors
Have set in config.psm1 the settings:
	"XilinxVivado" =		@{
		"InstallationDirectory" = "C:\CaeTools\Xilinx\Vivado\2018.1\";
		"SourceDirectory" =       "data\vhdl\src";
		"DestinationDirectory" =  "Xilinx"
When running in powershell:  .\compile-xilinx-vivado.ps1 -All -VHDL2008

I get an error: 
   [ERROR]: Path '\data\vhdl\src' does not exist.

What is wrong since the path clearly exists.
I tried with an with '\' at the end of the InstallationDirectory path, nothing helps.

Can somebody give an hint?

Thanks

Marc

Paebbels, 2018-04-30:
According to the comment that I wrote in `config.psm1`, no trailing backslash is allowed after a path.
> Configure your tools here. Use absolute paths, without trailing directory
> delimiter. Empty strings indicate not installed tools

Please run the command again with option `-Debug` and paste the full console log.

Defossez, 2018-04-30:
Thanks for the fast answer.

Run powershell-ISE from WIndows-10 menu (must say I'm a complete novice in this).
cd  C;\CaeTools\Ghdl\lib\vendors
edit config.psm1 and changed Xilinx-Vivado path to looks as:
		"InstallationDirectory" = "C:\CaeTools\Xilinx\Vivado\2018.1";
		"SourceDirectory" =       "data\vhdl\src";
		"DestinationDirectory" =  "Xilinx"
Run the compile script with -Debug option as you asked:
PS C:\CaeTools\Ghdl\lib\vendors> .\compile-xilinx-vivado.ps1 -All -VHDL2008 -Debug
[ERROR]: Path '\data\vhdl\src' does not exist.
[DEBUG]: HARD EXIT

When I type - powershell presents a list with options I can select from, beside "-Debug" there was also "-Verbose". So I tried that but get the same result.

In case of: At the start, powershell did not want to run any script at all. 
So I had the change the execution policies.
These are now:
    PS C:\CaeTools\Ghdl\lib\vendors> Get-ExecutionPolicy
    RemoteSigned

Regards,

Paebbels, 2018-04-30:
Changing the execution policy is a necessary step and I expect that users of PowerShell know that they have to change this on their local machines :).

Please run the scripts from a real shell and not from within ISE (Integrated Scripting Environment).

What version of GHDL are you using?

Defossez, 2018-04-30:
I'm using Ghdl for a couple of year now on Ubuntu, without any issue! :-)
Recently installed it on Windows-10 for a project absolutely needing Windoze.
Version of GHDL:
    C:\Projects>ghdl --version
    GHDL 0.35 (tarball) [Dunoon edition]
    Compiled with GNAT Version: GPL 2017 (20170515-63)
    mcode code generator

Running the script in a 'real' powershell results in the same error message (with or without -Debug)

Defossez, 2018-06-13:
I've been debugging this issue a bit, I'm not a powershell guy, and found this:
First, using WIndows-10 Enterprise 64-bit

Changed config.psm script for Xilinx Vivado library compile;
```
	"XilinxVivado" =		@{
		"InstallationDirectory" = "C:\CaeTools\Xilinx\Vivado\2018.2";
		"SourceDirectory" =       "data\vhdl\src";
		"DestinationDirectory" =  "Xilinx"
	}
```
This is how it's going to be used:
    Directory used to compile the libraries in: C:\CaeTools\Ghdl\lib\XilSimLibs
    change directory, cd, and run from within that directory the library compiler
    scripts as:[PS] C:\CaeTools\Ghdl\lib\vendors\compile-xilinx-vivado.ps1 -All

When running the script the power shell shows this:
    **[ERROR]: Path '\data\vhdl\src' does not exist.**
    **[DEBUG]: HARD EXIT**

Examined the scripts to find where this error might come from.
Three scripts are used:
    "compile-xilinx-vivado.ps1" is calling "shared.psm1" and "config.psm1"
The ERROR and DEBUG messages are generated by some directory check in "shared.psm1".
At that place it goes wrong!

The original code is:
```
    ......
    .........
	94   $VendorToolInstallationDirectory =  Get-VendorToolInstallationDirectory
	95   $VendorToolSourceDirectory =        Get-VendorToolSourceDirectory
    96
	97   if ($Source -ne "")
	98   {	$SourceDirectory = $Source.TrimEnd("\")		}
	99   elseif ($EnvSource -ne "")
	100  {	$SourceDirectory = $EnvSource							}
	101  elseif ($VendorToolInstallationDirectory -ne "")
	102  {	$SourceDirectory = $VendorToolInstallationDirectory + "\" + $VendorToolSourceDirectory }
	103  else
	104  {	Write-Host "[ERROR]: $Module_VendorToolName is not configured in '$ScriptDir\config.psm1'." -ForegroundColor Red
	105  	Write-Host "  Use adv. options '-Source' and '-Output' or configure 'config.psm1'." -ForegroundColor Red
	106  	Exit-CompileScript -1
	107  }
    108
	109  if (-not (Test-Path $SourceDirectory -PathType Container))
	110  {	Write-Host "[ERROR]: Path '$SourceDirectory' does not exist." -ForegroundColor Red
	111  	Exit-CompileScript -1
	112  }
    113
	114  return Convert-Path (Resolve-Path $SourceDirectory)
    ........
    .....
```
The [ERROR] generated by the powershell is created in the code at line 109, 110 and 111.
The [ERROR] should report the *$SourceDirectory* variable.
The *$SourceDirectory* variable is following the code created from: 
        *$VendorToolInstallationDirectory + "\" + $VendorToolSourceDirectory*
But the [ERROR] only shows as output '\data\vhdl\src', being the *$VendorToolSourceDirectory* variable.

So I added one Wirte-Host line to the error generator code, as below
```
	if (-not (Test-Path $SourceDirectory -PathType Container))
	{	Write-Host "[ERROR]: Path '$SourceDirectory' does not exist." -ForegroundColor Red
        #1# Write-Host $SourceDirectory # or
        #2# Write-Host $VendorToolInstallationDirectory # or
        #3# Write-Host $VendorToolSourceDirectory # or
        #4# Write-Host $VendorToolInstallationDirectory + "\" + $VendorToolSourceDirectory
        # Remove #n# to enable a code line.
		Exit-CompileScript -1
	}
```
With #1# the output is *\data\vhdl\src*
With #2# the output is *C:\CaeTools\Xilinx\Vivado\2018.2*
with #3# the output is *\data\vhdl\src*
With #4# the output is *C:\CaeTools\Xilinx\Vivado\2018.2 + \ + data\vhdl\src*

Normally #1# should output: *C:\CaeTools\Xilinx\Vivado\2018.2\data\vhdl\src*
But that is not the case.

I think somebody really expert in powershell should check this out and correct it.

Something else (did not look that far down yet):
In the script "compile-xilinx-vivado.ps1", there is a code part looking as:
```
.....
........
112  function Get-XilinxVivadoDirectory
113  {	if (Test-Path env:XILINX_VIVADO)
114  	{	return $XILINX_VIVADO + "\" + (Get-VendorToolSourceDirectory)		}
115  	else
116  	{	$EnvSourceDir = ""
117  		foreach ($Drive in Get-DriveInfo)
118  		{	$Path = $Drive.Name + "Xilinx\Vivado"
119  			if (Test-Path $Path -PathType Container)
120  			{	foreach ($Major in 2018..2014)
121 				{	foreach ($Minor in 4..1)
........
.....
```
line 114 has the same code construct as 102 in previous code in file "shared.psm1".
Line 118 has a path set as *$Path = $Drive.Name + "Xilinx\Vivado"*, but what if the Xilinx
tools are installed in a different directory? like in my case C:/CaeTools/Xilinx/Vivado/....

Kind regards,

Marc

1138-4EB, 2018-06-13:
@Defossez, since I see that you used GHDL in Ubuntu for a couple of years before switching to Windows, would installing docker in Windows be a valid alternative for you?

I cannot help with PowerShell. We'll have to wait @Paebbels for that. But I can help you with the alternative.

Defossez, 2018-06-14:
@1138-4EB, I'm still using GHDL on Ubuntu, It now just for one project I need to use WIndows.
Never tried or though about docker, and will investigate of course.
What I did no to go forward is: compile libs on Ubuntu and copy them to Windows. 
That seems to work well for now but will be an issue when I need to deliver the project to the user.

@Paebbels, after writing the comment above, I continued looking into it. 
following Google searches, concatenation can be done many ways. So the way used might be correct.
The check in config.psm1 from line 97 till line 107 is if-elsif-elsif. the first elsif delivers as result for @EnvSource \data\vhdl\src.  Maybe the check does not even go that far as the second elsif, combining the path parts.
Don't know, yet, where the first elsif gets it's data from.

Marc

1138-4EB, 2018-06-14:
> Never tried or though about docker, and will investigate of course.

Using docker is not the best option for everyone, especially because it is not "natively" available in every Windows version. However, if you happen to use Windows 10 Pro, it is a relatively lightweight solution that allows you to run commands in a GNU/Linux environment as if it was a local binary.

Indeed, we provide ready-to-use docker images with GHDL. Also, there are a couple of images that include VUnit, and GtkWave. See `README.md` and `USE_CASES.md` in https://github.com/ghdl/docker for a general explanation. Also http://vunit.github.io/cli.html#continuous-integration-ci-environment. In #606 you'll find an example of a "new" user using one of these images. Other related issues are #579, #584 and #604.

> That seems to work well for now but will be an issue when I need to deliver the project to the user.

That should work, as long as i) GHDL is statically compiled or ii) the client/user has installed runtime dependencies. In the latter, you should be careful with the version of the dependencies.

In the end, I use docker intensively so that I don't have to worry about these dependency/version issues. I use these images either in GNU/Linux (typically Fedora/CentOS, Arch or Ubuntu), in Windows and/or in play-with-docker, and I know that the environment/context seen by GHDL is exactly the same. Therefore I spend no time in sysadmin tasks, and focus on source code development.

Please, accept my apologies if you find this too off-topic, even though I thought it might be useful as a workaround until the Windows workflow is fixed. Anyway, please jump to the [gitter channel](https://gitter.im/ghdl1/Lobby) or open a new issue if you want to further talk about the 'docker way'.

Paebbels, 2018-06-14:
I'm currently on a business trip. I'll try to look into that issue on the weekend.

@Defossez 

* Can you please check what if / elsif branch gets gets triggered?
* What is the content of `$Envsource`?

Defossez, 2018-06-14:
@1138-4EB,
Thank you very much for the explication.
I read your issue: GHDL: a complete VHDL parser #111 (Jul 2016) and found it very interesting.
I'm using Windows 10 Enterprise, so will need to check if it's going to be lightweight too.
It is not "too off-topic" at all and I also like the July 2016 ideas.
I will certainly dig in the gitter channel.

Defossez, 2018-06-14:
@Paebbels 
```
    ......
    .........
	97   if ($Source -ne "")
	98   {	$SourceDirectory = $Source.TrimEnd("\")		}
	99   elseif ($EnvSource -ne "")
	100  {	$SourceDirectory = $EnvSource							}
	101  elseif ($VendorToolInstallationDirectory -ne "")
	102  {	$SourceDirectory = $VendorToolInstallationDirectory + "\" + $VendorToolSourceDirectory }
    ........
    .....
```
When I just put a Write-Host line for $SourceDirectory in the check then the output is: \data\vhdl\src
```
if (-not (Test-Path $SourceDirectory -PathType Container))
	{	Write-Host "[ERROR]: Path '$SourceDirectory' does not exist." -ForegroundColor Red
                Write-Host $SourceDirectory 
 		Exit-CompileScript -1
	}

The first elsif, elseif ($EnvSource -ne "")  outputs as $SourceDirectory : \data\vhdl\src
The second elsif, elseif ($VendorToolInstallationDirectory -ne "") should output the full path
but I think that sinvce the first elsif already has output it never reaches to the second elsif.

Defossez, 2018-06-27:
@1138-4EB 

Just wanted to let you know:
I've applied your docker solution on one of my machines and it works fine.
I've proposed it to the customer but for them it is too complex/cumbersome. 
So I keep running GHDL installed on the Windows machine as is, using libraries compiled on a Linux box.

Kind regards,

Paebbels, 2021-01-15:
@ZirconfleX sorry for the very very long delay. I'm very sorry.

-----------
I found now time to update these scripts. It has been tested with Vivado 2020.2.

eine, 2021-01-15:
@ZirconfleX let me update the status of some of the other things we commented:

- GHDL is available through official MSYS2 repositories now. You might want to take that into account for future clients you may have.
- [hdl/containers](https://github.com/hdl/containers) was created, which provides Docker images including other open source EDA tools built from sources.

ZirconfleX, 2021-01-16:
Hi,

Thanks for the notification.
Case closed I would say.

Regards,

Marc

closed 2021-01-16 by eine