ghdl/ghdl #565: MinGW on Windows produces '*.dll.exe' files

Paebbels, 2018-05-01:
I'm trying to improve parts of the MinGW64 compilation results. While doing so, I found that normal compilation is creating these files:

* `ghdl_llvm.exe` - OK
* `ghdl1-llvm.exe` - OK
* `libghdl.dll.exe`
* `libghdl-0_36_dev.dll.exe`

I assume the last two files should have the extension `*.dll` instead of `*.dll.exe`, right?

Paebbels, 2018-11-26:
This bug still exists.

```
[...]
gnatbind -aI. -aOC:\git\GitHub\Paebbels\ghdl\build\mingw64-llvm70\pic -aI../../src -aI../../src/vhdl -aI../../src/grt -aI../../src/psl -aI../../src/vhdl/translate -aI../../src/vhdl/python -aI../../src/ghdldrv -shared -Llibghdl_ -z -x C:\git\GitHub\Paebbels\ghdl\build\mingw64-llvm70\pic\libghdl.ali
gnatlink C:\git\GitHub\Paebbels\ghdl\build\mingw64-llvm70\pic\libghdl.ali -shared-libgcc -o libghdl-0_36_dev.dll.exe -g -shared
```

eine, 2020-01-14:
@Paebbels, I think that #1010 was a duplicate of this issues, but we didn't see it. Anyway, this should be fixed. Can you confirm?

closed 2020-05-10 by eine