ghdl/ghdl #566: Migration from travis-ci.org to travis-ci.com

1138-4EB, 2018-05-02:
Travis is starting to merge travis-ci.org and travis-ci.com. GHDL is not affected, yet, but it will be in the following months:

> However, open source repositories will be migrated to travis-ci.com gradually, beginning at the end of Q2 2018

https://docs.travis-ci.com/user/open-source-on-travis-ci-com/#Existing-Open-Source-Repositories-on-travis-ci.org

@tgingold, do we want to keep the build history? If not, you can send an e-mail and have the migration done now.

Apart from the domain, they are also going to change how authentication works. We should keep an eye on it:

> This creates the foundation for us to remove the repo-scoped OAuth login token, which is a common request.

https://blog.travis-ci.com/2018-05-02-open-source-projects-on-travis-ci-com-with-github-apps

tgingold, 2018-05-05:
I don't care about the build history.  I will send the email.

1138-4EB, 2018-05-21:
@tgingold, just because of curiosity, any update about this?

This 'checks' feature seems interesting: https://blog.travis-ci.com/2018-05-07-announcing-support-for-github-checks-api-on-travis-ci-com

tgingold, 2018-05-26:
No update right now.  I have migrated some of my repo, but there is no option to migrate ghdl/ghdl.  I will investigate that if I have time, or I will wait until the migration automatically occurs...

1138-4EB, 2019-05-20:
Related to the latest announces by GitHub:

- https://github.com/features/package-registry
- https://github.com/features/actions

It might be worth changing from Travis to GitHub Actions, if these are available before the forced migration to `.com`.

tgingold, 2019-05-20:
Each time I try to read the github/travis doc, I get quickly confused...  Maybe I should do it very seriously.

1138-4EB, 2019-05-20:
> Each time I try to read the github/travis doc, I get quickly confused... Maybe I should do it very seriously.

The fact is that there is no clear documentation about how the migration works. You need to send an e-mail asking for the migration. They will reply asking for confirmation and proceed manually.

I do not suggest to read the docs seriously. Let's wait until GitHub Actions is out of beta, and we will then discuss different alternatives (CircleCI, GitLab, Bitbucket...).

closed 2021-01-13 by tgingold

eine, 2021-01-13:
+2.5y since this issue was opened... but we made it in the end! :tada:

PR #929 (2021-01-13): Migrate from Travis and AppVeyor to GitHub Actions
Close #566.

This is work in progress to deprecate Travis and AppVeyor as CI systems and *embrace* GitHub Actions.

- All the scripts in ` dist/travis` and ` dist/windows/appveyor` are moved to `dist/.deprecated/appveyor` and `dist/.deprecated/travis`, respectively, and none of them is used anymore. @tgingold, let me know if you prefer to just have them removed straightaway.

Relevant feature differences compared to the current Travis/AppVeyor solution:

- Startup time for Windows jobs is slightly slower, because MSYS2 is not installed by defa [... cut]