ghdl/ghdl #568: Add support for GCC 8

andreygursky, 2018-05-08:
GCC 8.1 has been released on May 2 [1].

[1] GCC 8.1 Released
   https://gcc.gnu.org/ml/gcc/2018-05/msg00017.html

1138-4EB, 2018-05-08:
@andreygursky, do you mean to use GCC 8 to build GHDL with mcode backend? Or to use GCC 8 as the GHDL backend, building it with any other installed GCC version?

If you tried any of these and got some error, it'd be great if you could post it here.

I just tried replacing [Makefile.in#L219](https://github.com/ghdl/ghdl/blob/master/Makefile.in#L219) with:

```
7.* | 8 | 8.*)      gcc_ortho_lang=ortho-lang-7.c ;; \
```

and I get the following error when building the [gcc-8_1_0](https://github.com/gcc-mirror/gcc/archive/gcc-8_1_0-release.tar.gz) backend in Debian Buster with [these](https://github.com/ghdl/ghdl/blob/master/dist/linux/docker/build/buster#L7) dependencies:

``` bash
../../gcc-srcs/gcc/vhdl/ortho-lang.c: In function 'tree_node* type_for_mode(mach
ine_mode, int)':
../../gcc-srcs/gcc/vhdl/ortho-lang.c:707:61: error: cannot convert 'poly_uint16
{aka poly_int<1, short unsigned int>}' to 'unsigned int' for argument '1' to 'tr
ee_node* type_for_size(unsigned int, int)'
     return type_for_size (GET_MODE_BITSIZE (mode), unsignedp);
```

andreygursky, 2018-05-08:
> Or to use GCC 8 as the GHDL backend

Yes.

I've just noticed recent PR https://github.com/ghdl/ghdl/pull/567 address building with GCC 8.

vicencb, 2018-05-08:
Hi,
with #567 ghdl can be built under gcc 8.1 (with gcc 7.3), but the generated compiler is not working :( ...

```sh
$ cat e.vhd
entity e is end entity;
architecture a of e is begin end architecture;

$ ghdl -a e.vhd 

$ ghdl -e e
/usr/lib/ghdl/libgrt.a(run-bind.o): In function `grt_main__Tsec_default_sized_stacksBIP':
(.text+0xc4): undefined reference to `system__secondary_stack__ss_stackIP'
/usr/lib/ghdl/libgrt.a(run-bind.o): In function `grt_init':
(.text+0xea): undefined reference to `__gnat_default_ss_size'
(.text+0xf4): undefined reference to `__gnat_binder_ss_count'
(.text+0x106): undefined reference to `__gnat_default_ss_pool'
collect2: error: ld returned 1 exit status
ghdl:error: compilation error
```

```
$ ghdl --version
GHDL 0.36-dev (v0.35-168-g252a9169) [Dunoon edition]
 Compiled with GNAT Version: 8.1.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Both x86_64 and arm64 fail the same way.

Regards,
&nbsp;&nbsp;Vicenç.

andreygursky, 2018-05-08:
Vicenç, thanks for testing! It seems again new src/ortho/gcc/ortho-lang-X.c is needed.

vicencb, 2018-06-03:
This is indeed a step forward, but the OP, in the third comment, stated that this issue refers to supporting gcc 8 as the backend.

commit ef5f47fa (2018-06-04): Add support for gcc 8.1 as a backend.

Fix #568
closed 2018-06-04 by tgingold