ghdl/ghdl #572: Documentation broken reference

gritbub, 2018-05-19:
I'm cleaning up the documentation at my [fork][1]. Besides editing simple typos and grammatical nuances, I also came across a broken reference **_:ref:'GHDL_options'_** that is used 3 times in [Invoking GHDL][2]. Look at [readthedocs][3] to see that the link is broken.

I can't figure out where this is supposed to refer back to. As a side note, **_GHDL_options_** (without the reference tag **_:ref:_**) is found 3 times in [ghdl.texi][4].

I am new to git. I'm also new to these file types (**_.md_**, **_.rst_**, and also **_readthedocs_** stuff), so I apologize in advance if I'm doing something wrong or just plain missing something obvious.

[1]: https://github.com/gritbub/ghdl
[2]: https://github.com/ghdl/ghdl/blob/master/doc/using/InvokingGHDL.rst
[3]: http://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#analysis-a
[4]: https://github.com/ghdl/ghdl/blob/master/doc/ghdl.texi

Paebbels, 2018-05-19:
`*.md` are MarkDown files. A simple markup language with simple formatting: headlines, bold/italic font face, identifier markup, embedded source code, tables, images and links. It's used by e.g. StackOverflow. 

`*.rst` are ReStructured Text files. A translation tool like Sphinx can translate it into an intermediate format and from there it can produce HTML, PDF, ePub reader and other output formats. It's more powerful than MarkDown. ReadTheDocs is a cloud service that offers Sphinx compile time and a webhosting service `*.readthedocs.org`. More over it can host different versions (and languages) of a documentation.

@tgingold was `:ref:'GHDL_options'` intended to reference to `ghdl --options-help` ?

```
C:\> ghdl.exe --options-help
Main options:
  --work=LIB         use LIB as work library
  --workdir=DIR      use DIR for the file library
  -PPATH             add PATH in the library path list
  --std=87/93/00/02/08  select vhdl 87/93/00/02/08 standard
  --std=93c          select vhdl 93 standard and allow 87 syntax
  --[no-]vital-checks  do [not] check VITAL restrictions
Warnings:
  -Wbinding          warns for component not bound
  -Wreserved         warns use of 93 reserved words in vhdl87
  -Wlibrary          warns for redefinition of a design unit
  -Wvital-generic    warns of non-vital generic names
  -Wdelayed-checks   warns for checks performed at elaboration
  -Wbody             warns for not necessary package body
  -Wspecs            warns if a all/others spec does not apply
  -Wunused           warns if a subprogram is never used
  -Werror            turns warnings into errors
Extensions:
  -fexplicit         give priority to explicitly declared operator
  -frelaxed-rules    relax some LRM rules
  -C  --mb-comments  allow multi-bytes chars in a comment
  --bootstrap        allow --work=std
  --syn-binding      use synthesis default binding rule
  -fpsl              parse psl in comments
Compilation list:
  -l[sca]            after semantics, canon or annotation
  --lall             -lX options apply to all files
  -lv                verbose list
  -v                 disp compilation stages
Compilation dump:
  -d[psa]            dump tree after parse, semantics or annotate
  --dall             -dX options apply to all files
  --dump-drivers     dump processes drivers
```

1138-4EB, 2018-05-19:
@gritbub, first of all, thanks a lot for cleaning the docs. Many of those typos and grammatical nuances where introduced by me. But I could hardly spot them, due to my limited knowledge of the language.

About `:ref:'GHDL_options'`, I think that it should point to subsection 'Options' in the same page, i.e., http://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#options Therefore, we should add `.. _REF:GHDL_options:` just before the corresponding subheader (line 203 in https://github.com/ghdl/ghdl/blob/master/doc/using/InvokingGHDL.rst).

About md and rst formats, note that GitHub is able to properly display any of those. But in the issues and PRs markdown is used.

Overall, LGTM. Are you building it locally? Did you set up a fork of RTD? Are you just editing the sources without building the docs?

Paebbels, 2018-05-19:
I recommend to use `:` instead of `_` as a delimiter symbol in references. `-` and/or `_` is used by extensions in Sphinx auto generating references.

gritbub, 2018-05-20:
@1138-4EB, happy to help, and happy to learn. Your English is pretty good for someone who doesn't speak it as their first tongue.

I have my own fork of ghdl, I've only been editing the sources locally without building the docs. At least so far. I do not have a fork of RTD.

I push my commits up to github regularly. I'll do a PR when I get through.

@Paebbels thanks for the tips, I'll go with `:`. Let me know if I need to fix anything else.

Also, @Paebbels, @1138-4EB, @tgingold:
Most of my edits are only textual. However, on one of my edits, found [here][1], there is the possibility that I am incorrectly changing something. Just wanted you to be aware of that edit _specifically_ for in the future when I submit my PR.

Thanks

[1]: https://github.com/ghdl/ghdl/commit/9c82234f9d6398a40d50a934dc599ec0bb2739c4

1138-4EB, 2018-05-20:
> However, on one of my edits, found here, there is the possibility that I am incorrectly changing something.

The change is correct. The meaning should be "outdated" or "not updated". The previous double negation was wrong.

> I have my own fork of ghdl, I've only been editing the sources locally without building the docs. At least so far. I do not have a fork of RTD.

I didn't word it well: you don't need a fork of RTD itself, because it is a service; you need to set up a new project. However, that's only one of the possibilities to test your changes:

1. You can just install `python` and `sphinx` (a python lib/package) locally. Then, all you need to do is execute `sphinx-build -T -b html -D language=en . <output_dir>`.
2. You can go to https://readthedocs.org/, sign up and set a new project (e.g. `GHDL-gritbub.readthedocs.org`) which is automatically synced with your fork (`github.com/gritbub/ghdl`).
  - For example: http://ghdl-devfork.readthedocs.io/en/latest/ , http://readthedocs.org/projects/ghdl-devfork/ Indeed, if you want, you can open a PR to the master branch of my fork (github.com/1138-4EB/ghdl). Then I will activate that version in ghdl-devfork.
3. You can use https://github.com/buildthedocs/btd to do the same as 1, but with docker containers, so that you don't need to install python in the host.
4. You can use buildthedocs in travis-ci, and have the result pushed to a gh-pages branch in you fork, in order to have it automatically served in https://gritbub.github.io/ghdl.
  - For example: https://1138-4eb.github.io/ghdl/doc/ is automatically built when I push changes to some of the branches in my fork.

Anyway, none of these are required. You can continue just editing the sources and we will have it checked afterwards. I just put it here, because you said that you are learning and I thought that it would be interesting for you to be aware of the bigger picture of the workflow.

Regards

tgingold, 2018-05-20:
I am closing this issue as the question was answered (thanks to @1138-4EB).  But feel free to continue the discussion.

closed 2018-05-20 by tgingold

gritbub, 2018-05-20:
Thanks for your help, @1138-4EB, I'll probably test with at least one of those methods. I appreciate your willingness to teach. You're correct, I am definitely interested in grasping the bigger picture.