ghdl/ghdl #576: conditional assignment depending on generic causes overflow

T045T, 2018-05-23:
I have a project that assigns different things to different parts of a signal depending on a generic - in essence, it zero-pads if the target register is larger than the source, and truncates the source if the target is smaller.

I implemented that using an if statement inside a process (a synchronous one in my actual code, a combinational for the reproduction case).

The code works fine and synthesizes (and runs on an FPGA) without problem using Vivado, but GHDL chokes on the if statement during elaboration:
```
$ ghdl -i --workdir=work *.vhdl && ghdl -m --workdir=work foo_tb && ghdl -r --workdir=work foo_tb
```

```
ghdl:error: overflow detected
  from: [unknown caller]
  from: work.foo_tb(tb).dut.CMP_ELAB at foo_tb.vhdl:10
  from: work.foo_tb(tb).DECL_ELAB at foo_tb.vhdl:18
  from: Elaboration of design
ghdl:error: error during elaboration
```

Am I doing something wrong here, or is this actually a bug in GHDL?

Output of `ghdl -v`:

```
GHDL 0.35 (v0.35-dirty) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

To repro:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity foo is
  generic(
    IN_WIDTH : positive := 8;
    OUT_WIDTH : positive := 8);
  port(
    A : in std_logic_vector(IN_WIDTH - 1 downto 0);
    X : out std_logic_vector(OUT_WIDTH - 1 downto 0));
end entity foo;

architecture repro of foo is

begin  -- architecture repro
  set_output: process (A) is
  begin  -- process set_output
    if OUT_WIDTH > IN_WIDTH then
      X(OUT_WIDTH - 1 downto IN_WIDTH) <= (others => '0');
      X(IN_WIDTH - 1 downto 0) <= A;
    else
      X <= A(OUT_WIDTH - 1 downto 0);
    end if;
  end process set_output;
end architecture repro;
```

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity foo_tb is
end entity foo_tb;

architecture tb of foo_tb is

  -- component generics
  constant IN_WIDTH  : positive := 8;
  constant OUT_WIDTH : positive := 4;

  -- component ports
  signal A : std_logic_vector(IN_WIDTH - 1 downto 0);
  signal X : std_logic_vector(OUT_WIDTH - 1 downto 0);

  -- clock
  signal Clk : std_logic := '1';
begin  -- architecture tb

  -- component instantiation
  DUT: entity work.foo
    generic map (
      IN_WIDTH  => IN_WIDTH,
      OUT_WIDTH => OUT_WIDTH)
    port map (
      A => A,
      X => X);

  -- clock generation
  Clk <= not Clk after 10 ns;

  -- waveform generation
  WaveGen_Proc: process
  begin
    -- insert signal assignments here
    A <= "00001111";
    wait until Clk = '1';

    assert (X = "1111");
  end process WaveGen_Proc;

end architecture tb;

configuration foo_tb_tb_cfg of foo_tb is
  for tb
  end for;
end foo_tb_tb_cfg;
```

tgingold, 2018-05-23:
I confirm the issue and I am investigating.

diogratia, 2018-05-23:
After concatenating all design units into a a single design file named foo_tb.vhdl a newer llvm version of ghdl (GHDL 0.36-dev (v0.35-168-g252a916)) gives a different message:

> ghdl -i --workdir=work *.vhdl && ghdl -m --workdir=work foo_tb && ghdl -r --workdir=work foo_tb
> analyze foo_tb.vhdl
> elaborate foo_tb
> ./foo_tb:error: bound check failure at foo_tb.vhdl:20
> ./foo_tb:error: error during elaboration

where the bit about bound check failure gives a better indication along with the line number:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity foo is
  generic(
    IN_WIDTH : positive := 8;
    OUT_WIDTH : positive := 8);
  port(
    A : in std_logic_vector(IN_WIDTH - 1 downto 0);
    X : out std_logic_vector(OUT_WIDTH - 1 downto 0));
end entity foo;

architecture repro of foo is

begin  -- architecture repro
  set_output: process (A) is
  begin  -- process set_output
    if OUT_WIDTH > IN_WIDTH then
      X(OUT_WIDTH - 1 downto IN_WIDTH) <= (others => '0');
      X(IN_WIDTH - 1 downto 0) <= A;
    else
      X <= A(OUT_WIDTH - 1 downto 0);   -- LINE NUMBER 20
    end if;
  end process set_output;
end architecture repro;
```

and we see that you're missing an if statement choice that will allow matching left and right elements in the assignment statement found on line 20.  

Your generic values are incompatible with your if statement noting the the X range is  (OUT_WIDTH  - 1 downto IN_WIDTH) for IN_WIDTH = 8 and OUT_WIDTH = 4 declared as constants in architecture tb of foo_tb and passed as generic values, which gives (4 -1 downto 8) which is a null array (no elements) and the slice of A in the expression on the right hand side of the assignment gives an A slice with a range  (OUT_WIDTH -1 downto 0) (the range of A) which is (8 - 1 downto 0).

The bounds indeed don't match, the implication is that you're doing something wrong and it's not a bug in ghdl.

tgingold, 2018-05-23:
First, there is a problem with the error line.  The error occurs in foo.vhdl line 20, ie:
``   X(IN_WIDTH - 1 downto 0) <= A;``
During elaboration, drivers are created.  And because this slice is not within the bounds of the index of X, there is a failure.

You could argue that there is no need to create a driver for this line as it is never executed.  That's true, but this is not what VHDL says (unless I am wrong).

You'd better to use an if-generate construct.

T045T, 2018-05-23:
Thanks for the quick response! An if-generate construct does sound like the cleaner way to do this, thanks for the suggestion. I'm still learning VHDL, so that didn't immediately come to mind.

I figured since the condition of the if statement just consists of a comparison of constants, VHDL might specify that only the selected branch is generated.
Apparently the Vivado toolset does just that, but I'm not nearly familiar enough with the standard to argue one way or the other.

That said, providing a more accurate error line for cases like this would be great :)

CraigNoble, 2018-05-24:
Every day is a school day, even when you've been working with VHDL for years there will still be times when you come across something that you hadn't known before.

Vivado should optimise the output from the normal if-else into the same optimal logic that would be produced by the if-generate (VHDL-2008 even allows if-generate-else-generate), but that's not how I would write this.

If it were me, I'd use the resize function from numeric_std as that function does exactly what you're looking for, but with signed/unsigned rather than std_logic_vector.

I'm a believer in fewer lines means fewer bugs and resize would allow you to do this on one line. I'd also do this as a function or procedure rather than an entity but that's neither here nor there.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is
  generic(
    IN_WIDTH : positive := 8;
    OUT_WIDTH : positive := 8);
  port(
    A : in std_logic_vector(IN_WIDTH - 1 downto 0);
    X : out std_logic_vector(OUT_WIDTH - 1 downto 0));
end entity foo;

architecture repro of foo is

begin  -- architecture repro

    X <= std_logic_vector(resize(unsigned(A), OUT_WIDTH));
     
end architecture repro;
```

You could even write a resize function for std_logic_vector and use that instead of an entity:

As a wrapper around the numeric_std function:
```vhdl
  function RESIZE (ARG: STD_LOGIC_VECTOR; NEW_SIZE: NATURAL) return STD_LOGIC_VECTOR is
  begin
    return STD_LOGIC_VECTOR(RESIZE(UNSIGNED(ARG), NEW_SIZE));
  end RESIZE;
```

Or as a stand-alone:
```vhdl
  function RESIZE (ARG: STD_LOGIC_VECTOR; NEW_SIZE: NATURAL) return STD_LOGIC_VECTOR is
    constant NAU : STD_LOGIC_VECTOR(0 downto 1) := (others => '0');
    constant ARG_LEFT: INTEGER := ARG'LENGTH-1;
    alias XARG: STD_LOGIC_VECTOR(ARG_LEFT downto 0) is ARG;
    variable RESULT: STD_LOGIC_VECTOR(NEW_SIZE-1 downto 0) := (others => '0');
  begin
    if (NEW_SIZE < 1) then return NAU;
    end if;
    if XARG'LENGTH =0 then return RESULT;
    end if;
    if (RESULT'LENGTH < ARG'LENGTH) then
      RESULT(RESULT'LEFT downto 0) := XARG(RESULT'LEFT downto 0);
    else
      RESULT(RESULT'LEFT downto XARG'LEFT+1) := (others => '0');
      RESULT(XARG'LEFT downto 0) := XARG;
    end if;
    return RESULT;
  end RESIZE;
```

Brian-Drummond, 2018-05-30:
Not necessarily relevant to this specific case, but...The mcode backend can detect some classes of overflow that (a) are genuine and (b) remain undetected by most other simulators - sometimes including the other ghdl backends.