ghdl/ghdl #577: New 'docker' repo in the organization?

1138-4EB, 2018-05-24:
Since we get questions about compiling GHDL with GCC backend once in a while, I cleaned the branch I had in my fork, in order to provide a working example: https://travis-ci.org/1138-4EB/ghdl/builds/383427719

As you see, I have to check GCC 5.5.0 in Stretch, but GCC 7.2.0 in Buster works. I was thinking about opening a PR to add just the latter. However, I think that adding a 40min job is too much. Furthermore, I'd like to add Fedora 27 too.

Therefore, I think we might create a 'ghdl/docker' repo and move [`dist/linux/docker`](https://github.com/ghdl/ghdl/tree/master/dist/linux/docker), [`dist/linux/create.sh`](https://github.com/ghdl/ghdl/blob/master/dist/linux/create.sh), [`dist/linux/deploy2dockerhub.sh`](https://github.com/ghdl/ghdl/blob/master/dist/linux/deploy2dockerhub.sh), and [`dist/linux/travis-enc-dockerhub.sh`](https://github.com/ghdl/ghdl/blob/master/dist/linux/travis-enc-dockerhub.sh) there.

Doing so would allow us to create multiple branches/subdirs there (up to one for each platform), while keeping the main repo clean. Also, because docker images need not to be updated with each commit, having a separate repo would help run CRON jobs. For example, we can everyday build the images that are used by the main repo. But then, run the other platforms, GCC, ghdlsynth... once a week.

Indeed, we are working like this in practice, because this repo does not have credentials to push images to dockerhub, so it is my fork the one that triggers it. Consequently, I would not mind to keep doing it in the proposed new repo.

This proposal is related to #477 and to the slightly increased complexity of #569.

CraigNoble, 2018-05-25:
`I'd like to add Fedora 27 too.`

Why not Fedora 28?

1138-4EB, 2018-05-25:
@CraigNoble sure. I didn't know it was out already. Shall a different repo be created, there would be no problem to have *lots* of versions (3-4 Ubuntu, 3-4 Fedora, 3-4 Debian...). Moreover, it'd allow users to just clone the branch they want and have their binary built either in travis, in play-with-docker or locally. However, I don't feel like putting all this in the main repo.

CraigNoble, 2018-05-25:
Sounds great. :+1:

tgingold, 2018-05-26:
Looks like a good idea!  So do you need me to create the ghdl/docker repo ?

1138-4EB, 2018-05-26:
Yes, please. Create `ghdl/docker` and give me write permissions in that repo only. I will progressively move the content there.

1138-4EB, 2018-05-26:
The repo is created already, and I have access to it. However, I'd like to keep this issue open until I set it up. I'll make a PR close this automatically.

closed 2018-05-27 by Paebbels

CraigNoble, 2018-05-28:
@1138-4EB Thanks for the effort and the update. :-)

1138-4EB, 2018-06-03:
@CraigNoble , you are welcome.

See [568 (comment)](https://github.com/ghdl/ghdl/issues/568#issuecomment-394177374)

PR #585 (2018-05-27): move docker-related resources to ghdl/docker
Close #577.

This PR removes sources that are not required anymore, because they are now moved to [ghdl/docker](https://github.com/ghdl/docker). Also `dist/linux` is renamed to `dist/travis`, because all of the sources there are travis-specific.