ghdl/ghdl #578: compiler error

faf5678, 2018-05-24:
cannot execute ghdl -a generic.vhd

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (v0.35) [Dunoon edition]
Compiled with GNAT Version: 4.6
Target: x86_64-pc-linux-gnu
In directory: /home/user/test/ghdl/
Command line:
/home/user/ghdl/bin/ghdl1-llvm -P/home/user/ghdl/lib/ghdl/v93/std/ -P/home/user/ghdl/lib/ghdl/v93/ieee/ -c -o generics.o generics.vhd
Exception STORAGE_ERROR raised
Exception information:
Exception name: STORAGE_ERROR
Message: stack overflow (or erroneous memory access)
Call stack traceback locations:
0x7f31eee0033e 0xf6a8fb 0x48236e 0x5e7cf9 0x5615df 0x56179e 0x565920 0x5add81 0x5c321c 0x5c3323 0x5aca6a 0x5acc1d 0x5c31d1 0x5c3323 0x5aabf8 0x5aaca2 0x5c3195 0x5c3359 0x5efd21 0x5d3451 0x5f0ef1 0x589307 0x601df1 0x6024ef 0x48741b 0x44a0fa 0x7f31edb04ec3 0x44990a 0xfffffffffffffffe
******************************************************************
/home/user/ghdl/bin/ghdl:error: compilation error

generics.vhd attached as generics.txt

[generics.txt](https://github.com/ghdl/ghdl/files/2037332/generics.txt)

diogratia, 2018-05-25:
Without incrementally commenting out particular functions to find the one causing the problem requires tracing execution.  The author of the package would be best able to apply incremental commenting out.  (The package body order of functions isn't helpful, you can depend on declarations in the package declaration to incorporate use of function specifications appearing later in the package body.)

Some of the functions/procedures aren't called in any subprogram specification in the package body and can easily be commented out. Some not so easily.

Anyway the problem can be reduced to a minimal, complete and verifiable example:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

package crash_pkg is
    function reorder_vector(inp : std_logic_vector) return std_logic_vector;
end package crash_pkg;

package body crash_pkg is
    function reorder_vector(inp : std_logic_vector) return std_logic_vector is
        variable ret : std_logic_vector(inp'reverse_range);
    begin
        return inp;
        if inp'left < inp'right then
            for i in inp'range loop
                ret(inp'right - i) := inp(i);
            end loop;
        elsif inp'left > inp'right then
            for i in inp'range loop
                ret(inp'left - i) := inp(i);
            end loop;
        else
            ret(inp'left) := inp(inp'left);
        end if;
        return ret;
    end function;
end package body crash_pkg;
```
```

(Note the lack of error diagnostics here is a ghdl issue.)

tgingold, 2018-05-26:
Thank you for the issue, I will push  a fix soon.

tgingold, 2018-05-27:
My bad!

jahagirdar, 2018-06-24:
I am having a similar error where the design consumes around 30 GB of memory before crashing with a storage error message. My server has around 128GB RAM, so free memory is not an issue. Is it related to this issue?
Console log produced by running /bin/time -v make is as follows

Execution terminated by unhandled exception
Exception name: STORAGE_ERROR
Message: grt-table.adb:60 explicit raise
Call stack traceback locations:
0x464030 0x4640be 0x466214 0x4690b2 0x7f965655f2ee
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory `/home/centos/interconnect/tb'
make: *** [sim] Error 2
Command exited with non-zero status 2
	Command being timed: "make"
	User time (seconds): 69.82
	System time (seconds): 14.47
	Percent of CPU this job got: 100%
	Elapsed (wall clock) time (h:mm:ss or m:ss): 1:24.29
	Average shared text size (kbytes): 0
	Average unshared data size (kbytes): 0
	Average stack size (kbytes): 0
	Average total size (kbytes): 0
	Maximum resident set size (kbytes): 29668076
	Average resident set size (kbytes): 0
	Major (requiring I/O) page faults: 0
	Minor (reclaiming a frame) page faults: 7241100
	Voluntary context switches: 455
	Involuntary context switches: 244
	Swaps: 0
	File system inputs: 0
	File system outputs: 8
	Socket messages sent: 0
	Socket messages received: 0
	Signals delivered: 0
	Page size (bytes): 4096
	Exit status: 2

Paebbels, 2018-06-25:
Please open a new issue and add a reproducer so we can investigate your code that is causing such an error.

jahagirdar, 2018-06-25:
I am getting this from a design which has around 50K lines of generated code instantiating and connecting some 3K modules at the toplevel. It will take me some days/weeks to eliminate rest of the design and create a minimal failing code... As and when I manage to do so, I will file a bug report.

Paebbels, 2018-06-25:
Is the code classified? Can you send a zip file or a link to a Git repository?

jahagirdar, 2018-06-25:
Regretfully  no, I cannot send the current code across as it was written under NDA.
I will try to root cause the failure at a later date and file a bug report then :(

diogratia, 2018-07-22:
The fix doesn't appear to have worked on the OS X platform.

diogratia@Macbook: $GHDL --version
GHDL 0.36-dev (v0.35-231-g6a24c6d) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
Written by Tristan Gingold.

(The .git subdirectory has to remain intact now to get the commit in the version. It's a pity, the .git is chock full of recent documentation changes.)

diogratia@Macbook: echo $PATH
/Users/diogratia/share/clang+llvm-3.8.0-x86_64-apple-darwin/bin:/Users/diogratia/share/gnat-gpl-2015-x86_64-darwin-bin/bin:.:/Users/diogratia/bin:/usr/bin:/usr/sbin:/bin:/sbin:/usr/local/bin:/usr/X11/bin/:/usr/libexec

Adacore gnat-gpl-2015-x86_64-darwin-bin

llvm.org clang+llvm-3.8.0-x86_64-apple-darwin

Bombed out in the testsuite:

GNA dir issue563:
try to analyze repro.vhdl
repro.vhdl:44:31:error: unexpected token ')' in a primary
repro.vhdl:44:34:error: "<=" or ":=" expected instead of ')'
repro.vhdl:48:5:error: 'process' is expected instead of "arch_tb_counter"
try to analyze repro2.vhdl
repro2.vhdl:2:1:error: context clause not allowed before vhdl 08
repro2.vhdl:2:1:error: design file is empty (no design unit found)
analyze counter.vhdl
try to analyze tb_counter.vhdl
tb_counter.vhdl:23:26:error: empty interface list not allowed
tb_counter.vhdl:42:22:error: empty association list is not allowed
tb_counter.vhdl:45:31:error: unexpected token ')' in a primary
tb_counter.vhdl:45:34:error: "<=" or ":=" expected instead of ')'
Remove work library
Test successful

GNA dir issue578:
analyze crash2.vhdl

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: erroneous memory access
Load address: 0x10a608000
Call stack traceback locations:
0x10b354ae7 0x7fff8adc6528
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/build/ghdl_llvm:error: compilation error
#################################################################
######### FAILURE: issue578
#################################################################

diogratia@Macbook: cd gna/issue578
diogratia@Macbook: ls
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/testsuite/gna/issue578
crash2.vhdl    generics.vhdl  testsuite.sh*
diogratia@Macbook: more testsuite.sh
#! /bin/sh

. ../../testenv.sh

#export GHDL_STD_FLAGS=--std=08
analyze crash2.vhdl
analyze generics.vhdl

clean

echo "Test successful"
diogratia@Macbook: more crash2.vhdl
library ieee;
use ieee.std_logic_1164.all;

package crash_pkg is
    function reorder_vector(inp : std_logic_vector) return std_logic_vector;
end package crash_pkg;

package body crash_pkg is
    function reorder_vector(inp : std_logic_vector) return std_logic_vector is
        variable ret : std_logic_vector(inp'reverse_range);
    begin
        return inp;
        if inp'left < inp'right then
            for i in inp'range loop
                ret(inp'right - i) := inp(i);
            end loop;
        elsif inp'left > inp'right then
            for i in inp'range loop
                ret(inp'left - i) := inp(i);
            end loop;
        else
            ret(inp'left) := inp(inp'left);
        end if;
        return ret;
    end function;
end package body crash_pkg;
diogratia@Macbook: $GHDL -a crash2.vhdl

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: erroneous memory access
Load address: 0x10c79d000
Call stack traceback locations:
0x10d4e9ae7 0x7fff8adc6528
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/build/ghdl_llvm:error: compilation error
diogratia@Macbook:

Commenting out the return inp; immediately found in the function body still eliminates the error.