ghdl/ghdl #579: Issue runner

1138-4EB, 2018-05-25:
It is quite common in this repo to request users to provide Minimal Working Examples (MWEs) or reproducers of the errors they get. I wrote a couple of scripts that allow to execute such MWEs with a single command, so that developers are not required to copy sources manually and set the testing environment.

https://github.com/1138-4EB/issue-runner

---

I am using this issue as an example. This is the *issue-runner friendly* version of ghdl/ghdl#562.

``` vhdl
#>> ent.vhd
entity ent is
  generic (
    max : natural);
  port (
    p : out natural range 1 to max := 3);
end entity;

architecture a of ent is
begin
end;

#>> ent2.vhd
entity ent2 is
end entity;

architecture a of ent2 is
  constant max : natural := 2;
  signal p : natural range 1 to max;
begin

  inst : entity work.ent
    generic map (max => max)
    port map (p => p);
end;

#>> run.sh
ghdl --version
ghdl -a ent.vhd
ghdl -a ent2.vhd
ghdl --elab-run ent2
```

1138-4EB, 2018-05-25:
This is the result:

``` bash
$ curl -L https://raw.githubusercontent.com/1138-4EB/issue-runner/master/runner.sh | sh -s ghdl/ghdl#579
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   881  100   881    0     0  11441      0 --:--:-- --:--:-- --:--:-- 11746
cmd: /usr/bin/python extract_sources.py ghdl/ghdl#579
GHDL 0.36-dev (v0.35-132-gab9391ea) [Dunoon edition]
 Compiled with GNAT Version: 6.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
ghdl:error: bound check failure at ent2.vhd:5
  from: work.ent2(a).inst.CMP_ELAB at ent2.vhd:5
ghdl:error: error during elaboration
```

I'm closing this issue, because it is just an example.

closed 2018-05-25 by 1138-4EB

tgingold, 2018-05-26:
Interesting, but how to make sure that submitters use that format ?

1138-4EB, 2018-05-26:
We cannot make sure, but we can strongly suggest it with issue templates:

- https://blog.github.com/2016-02-17-issue-and-pull-request-templates/
- https://help.github.com/articles/about-issue-and-pull-request-templates/
- https://help.github.com/articles/manually-creating-a-single-issue-template-for-your-repository/

Then, we just need to add one or more `issue_template.md` to the main repo. I'd say that adding it/them to a new subdir named `.github` is the cleanest solution.

I'll have a better look at it, and I'll propose a PR.

Example: go to https://github.com/filebrowser/filebrowser and open a new issue. You'll see that it is filled with https://github.com/filebrowser/filebrowser/blob/master/.github/ISSUE_TEMPLATE.md.

Paebbels, 2018-05-26:
I think GitHub can nowadays handle multiple issue / pull request templates. The file path is `.github\ISSUE_TEMPLATE\<template name>.md`. This should show a drop-down like on GitLab.com.

1138-4EB, 2018-05-26:
@Paebbels, indeed, that's what I am doing. I created:

- `.github/`
  - `ISSUE_TEMPLATE/`
  - `PULL_REQUEST_TEMPLATE`

https://blog.github.com/2018-01-25-multiple-issue-and-pull-request-templates/

Now, I am checking template examples:

- https://github.com/stevemao/github-issue-templates
- https://github.com/devspace/awesome-github-templates
- https://github.com/cezaraugusto/github-template-guidelines
- https://github.com/liuderchi/gh-template-bot

1138-4EB, 2018-05-26:
After merging #583, using 1138-4EB/issue-runner is suggested in the 'Bug report' template: https://github.com/ghdl/ghdl/issues/new?template=bug_report.md