ghdl/ghdl #580: GHDL compiler bug

GustavSVJ, 2018-05-25:
Bug occurring when utilizing VUnit with GHDL. The code was originally written in Xilinx ISE 14.7 and can be compiled there.  The source files are attached as a zip file. Can somebody tell me what goes wrong?

Thanks,
Gustav

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: C:\workspace\1091\fw\FPGA\head\hw_rev_u4\fpga\GHDL_bug_report\bug1\Debug\src\VUnit\
Command line:
C:\Program Files\GHDL\bin\ghdl.exe -a --workdir=C:\workspace\1091\fw\FPGA\head\hw_rev_u4\fpga\GHDL_bug_report\bug1\Debug\src\VUnit\vunit_out\ghdl\libraries\fpga_lib --work=fpga_lib --std=08 -PC:\works
pace\1091\fw\FPGA\head\hw_rev_u4\fpga\GHDL_bug_report\bug1\Debug\src\VUnit\vunit_out\ghdl\libraries\vunit_lib -PC:\workspace\1091\fw\FPGA\head\hw_rev_u4\fpga\GHDL_bug_report\bug1\Debug\src\VUnit\vunit
_out\ghdl\libraries\fpga_lib --ieee=synopsys C:\workspace\1091\fw\FPGA\head\hw_rev_u4\fpga\GHDL_bug_report\bug1\Debug\src\master_Bug.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
******************************************************************

[Debug.zip](https://github.com/ghdl/ghdl/files/2039266/Debug.zip)

tgingold, 2018-05-26:
I think the crash is due to an unsupported vhdl 2008 feature: aggregates can now be locally static.
I plan to implement that feature.

eine, 2021-01-17:
Had to adjust the `root` and `add_source_files` args in the `run_debug.py` file, but I could reproduce the bug with a recent build of GHDL:

```
# python3 run_debug.py
Compiling into fpga_lib:  ../master_Bug.vhd                                                                                            
C:/msys64/mingw64/lib/python3.8/site-packages/vunit/sim_if/ghdl.py:256: Warning: 'ghdl.flags' is deprecated and it will be removed in future releases; use 'ghdl.a_flags' instead
  warn(
failed
=== Command used: ===
C:/msys64/mingw64/bin/ghdl -a --workdir=D:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/fpga_lib --work=fpga_lib --std=08 -PD:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/vunit_lib -PD:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/fpga_lib --ieee=synopsys D:/ghdl/ghdl/temp/master_Bug.vhd

=== Command output: ===
build_constant: cannot handle IIR_KIND_AGGREGATE (D:/ghdl/ghdl/temp/master_Bug.vhd:52:73)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1270.gb65ee8c3) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
D:\data-dev\eine\ghdl\ghdl\temp\VUnit\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --workdir=D:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/fpga_lib --work=fpga_lib --std=08 -fsynopsys -PD:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/vunit_lib\ -PD:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/fpga_lib\ -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o D:/ghdl/ghdl/temp/VUnit/vunit_out/ghdl/libraries/fpga_lib\master_Bug.o D:/ghdl/ghdl/temp/master_Bug.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x4a78bf 0x4e3b5e 0x4eeeb4 0x4eef74 0x558d5e 0x5591b9 0x57243f 0x572515 0x573546 0x572553 0x573546 0x5736b3 0x57421c 0x574983 0x574bc1 0x574d03 0x534787 0x53b2d7 0x52263f 0x636ee6 0x638bba 0x63a495 0x402706 0x4013bf 0x4014f4 0x7ffaa4177032 0x7ffaa549d0cf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error

Compile failed
```

dalex78, 2026-09-03:
# Issue #580 — `build_constant: cannot handle IIR_KIND_AGGREGATE`

Analysis by an AI agent. No longer reproducible; recommend merging the regression test and closing.

## The reproducer, without the zip

The report attached `Debug.zip` and a `DebugLog.txt`. The zip is eleven VHDL files and a VUnit script; the log narrows it to one line:

```
Compiling into fpga_lib:  ..\master_Bug.vhd    failed
build_constant: cannot handle IIR_KIND_AGGREGATE (…\master_Bug.vhd:52:73)
raised TYPES.INTERNAL_ERROR : errorout.adb:246
```

Nobody needs the zip. `master_Bug.vhd` declares a `std_logic_vector` signal whose default value is a constant from `pkg_ext_i2c`, and later uses that same constant as a **case choice**. This is the whole thing:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;

package pkg_ext_i2c is
  constant led_drv_ic11_i2C_address : std_logic_vector(6 downto 0) :=
    std_logic_vector(conv_unsigned(16#60#, 7));
end package;

library ieee;
use ieee.std_logic_1164.all;
use work.pkg_ext_i2c.all;

entity repro is
end entity;

architecture behav of repro is
  signal led_device : std_logic_vector(6 downto 0) := (others => '0');
  constant led_device_zero : std_logic_vector(6 downto 0) := (others => '0');
begin
  process (led_device)
  begin
    case led_device is
      when led_device_zero =>
        null;
      when led_drv_ic11_i2C_address =>
        null;
      when others =>
        null;
    end case;
  end process;
end architecture;
```

Analysed with `--std=08 --ieee=synopsys`, exactly as the reporter's VUnit run did.

## What is wrong with the source

`led_drv_ic11_i2C_address` is initialised by a call to `conv_unsigned`, so it is globally static but **not locally static**, and LRM 10.9 requires a case choice to be locally static. The design is invalid. `led_device_zero`, initialised by an aggregate of literals, is locally static and is fine — which is why only the second choice is rejected.

## Status on master

**The internal error is gone.** GHDL now says what is actually wrong, at the right place:

```
repro.vhd:31:12:error: choice must be locally static expression
      when led_drv_ic11_i2C_address =>
           ^
```

and the same for the reporter's own files, straight out of the zip:

```
master_Bug.vhd:99:18:error: choice must be locally static expression
```

Confirmed on mcode, llvm and llvm-jit, with and without `-frelaxed`.

This was fixed some time before **2.0.0**: that release, built here from the `v2.0.0` tag, already gives the same message. The exact commit was not pinned — reproducing the original would need 0.35 or 0.36, and sources of that age no longer compile with a current GNAT (real errors in `grt-signals.adb` and elsewhere, not just warnings).

One thing worth noting for whoever closes this: `Error_Kind ("build_constant", Val)` is still there at [vhdl-evaluation.adb:292](src/vhdl/vhdl-evaluation.adb#L292). What changed is that `sem` now rejects this input before evaluation is asked to fold a non-static aggregate, so `Build_Constant` is no longer reached *by this design*. The internal error itself has not been converted into a diagnostic, so it may still be reachable from some other input.

## What this change adds

Nothing in `src/`. `testsuite/gna/issue580` analyses the reproducer above and asserts three things: that it is rejected, that the message is the locally-static one, and that no `GHDL Bug occurred` box appears.

Recommend closing #580 as fixed.

closed 2026-09-04 by tgingold