ghdl/ghdl #581: Generic Package with access type

TripRichert, 2018-05-26:
I got a "Bug Occurred" crash message during elaboration when trying to build a generic package with an access type.

I'm using the v035 release (hash eb5499b185)
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (v0.35) [Dunoon edition]
Compiled with GNAT Version: 7.2.0
Target: x86_64-linux-gnu
In directory: /home/trip/Documents/projects/genericIssue/
Command line:
ghdl -e --std=08 Test_LinkedListPkg
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-rtis.adb:1655 range check failed
Call stack traceback locations:
0x55b0976813f2 0x55b097681f99 0x55b09767fb5d 0x55b097681edf 0x55b09768265c 0x55b09768447a 0x55b097685b44 0x55b097684967 0x55b0976859ea 0x55b097686e28 0x55b0976b0a92 0x55b097622180 0x55b09761fe77 0x55b09762cc79 0x55b0976d8a90 0x55b0976c9369 0x55b0976b8d5f 0x55b0976e4ae4 0x55b0973fa66f 0x7f77ab7441bf 0x55b0973f9498 0xfffffffffffffffe
******************************************************************

I tried to make it as minimal as I could and attached it.

[genericIssue.zip](https://github.com/ghdl/ghdl/files/2041677/genericIssue.zip)

Paebbels, 2018-05-26:
Generic types in packages should be supported by GHDL.

**Related issues:**
* [Crash on type generic](https://github.com/ghdl/ghdl/issues/412) in entities (open)
* [Exception raised trying to use simple generic package](https://github.com/ghdl/ghdl/issues/406) (solved)

tgingold, 2018-05-29:
I cannot reproduce the crash with the development version, so it looks like the bug has been fixed.

Thank you for the report.

closed 2018-05-29 by tgingold