ghdl/ghdl #584: Issue runner (attached zipfile)

1138-4EB, 2018-05-26:
**Description**
This issue is an example. It is the issue-runner friendly version of ghdl/ghdl#581. I created it to test tarball/zipfile support in issue-runner, and to test the new 'Bug report' template.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ ] Built from sources. Commit SHA:

**How to reproduce?**

```
#>> sim.sh
ghdl -a LinkedListPkg.vhd
ghdl -a Test_LinkedListPkg.vhd
ghdl -e --std=08 Test_LinkedListPkg

#>> run.sh
docker run --rm -tv /$(pwd):/src -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Files**

- [genericIssue.zip](https://github.com/ghdl/ghdl/files/2041677/genericIssue.zip)

**Checklist**

- [ ] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [ ] Try the latest version

1138-4EB, 2018-08-13:
Since a couple of days ago, mcode builds on Debian Buster and Ubuntu 18 are failing because of this issue. However, tests pass for any other platform:

- https://travis-ci.com/ghdl/docker/builds/81572106
- https://travis-ci.com/ghdl/docker/builds/81592104
- https://travis-ci.com/ghdl/docker/builds/81616596

@tgingold, any guess about what might be going on here?

tgingold, 2018-12-13:
I think the initial problem is now fixed.  Reopen this issue if this is not the case.

closed 2018-12-13 by tgingold