ghdl/ghdl #586: Use clause of specific type in package does not make it visible for following entity declaration.

pmarkkula, 2018-05-28:
**Description**
When trying to use type that was made visible with use clause in entity declarations generic list fails analysis with following error:

`bug.vhdl:6:26:error: no declaration for "zero"`

**Expected behaviour**
Type should be visible to entity declaration. 

**Context**

- OS: Linux (Ubuntu Xenial)
- Origin:
  - [ ] Built from sources. Commit SHA: 6a9d095dbe85e725639e6dc3080be9cdb94db4ef

**How to reproduce?**
Following code illustrates the problem.

```
#>>bug_pkg.vhdl
package bug_pkg is
  type t_zero_one is (zero, one);
end package bug_pkg;

#>>bug.vhdl
use work.bug_pkg.t_zero_one;            -- .all works
entity bug is
  generic (
    test : t_zero_one := zero
    );
end entity bug;

#>> sim.sh
ghdl -a bug_pkg.vhd.
ghdl -a bug.vhdl

#>> run.sh
docker run --rm -tv /$(pwd):/src -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

tgingold, 2018-05-29:
The behaviour of use clauses was modified by vhdl 2008.  So if you use --std=08 option, there is no error.  If you stay with vhdl 93 (or vhdl 02), there is an error.

closed 2018-05-29 by tgingold

pmarkkula, 2018-05-29:
Do you mean the VHDL standard was modified or does GHDL handle use clauses differently with --std=08?

The example code work fine with --std=08, but my design fails, because of rol-operator overloading.

tgingold, 2018-05-29:
> Do you mean the VHDL standard was modified or does GHDL handle use clauses differently with --std=08?

Both: the VHDL standard was modified for the 2008 revision, and --std=08 follows the change.

> The example code work fine with --std=08, but my design fails, because of rol-operator overloading.

You have to decide which standard you want to follow!