ghdl/ghdl #590: CONSTRAINT_ERROR occurs

LogicSolutionsBrundler, 2018-05-30:
**Description**
It looks like I am running into an internal error. From the error message, I am not able to narrow-down the root-cause.
I am not very active on GitHub, so please contact me on oliver.bruendler[at]psi.ch if you have any questions, feeback, etc.
Of course I am willing to spend time to get the issue fixed too.

**Context**
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: i686-pc-mingw32
In directory: D:\git\Issus\GHDLTest\FailLinApprox\
Command line:
C:\Program Files (x86)\GHDL\bin\ghdl.exe --elab-run --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_fix_lin_approx_sin18b_tb -gStimuliDir_g=. --ieee-asserts=disable
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:2566 access check failed
******************************************************************

- OS: Windows10 - 64 bit

**How to reproduce?**
I provide a ZIP file including all files required and a batch file. Just run the batch file to reproduce the issue.
[FailLinApprox.zip](https://github.com/ghdl/ghdl/files/2052908/FailLinApprox.zip)

tgingold, 2018-05-31:
I can reproduce the crash, and it is due to not fully implemented vhdl 2008 features.

LogicSolutionsBrundler, 2018-06-01:
Thank you for the update.

I guess this is just one featrue request amongst others and it will take some time to get implemented, right? If so, do you know which VHDL statement causes the crash? I could then check if I can work around the issue by avoiding that statement.

eine, 2021-01-17:
Tested with a recent version of GHDL:

```
# ./reproduce_bat.bat
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_tb_txt_util.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_tb_textfile_pkg.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_common_math_pkg.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_common_logic_pkg.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_fix_pkg.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_fix_lin_approx_calc.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_fix_lin_approx_sin18b.vhd 
D:\ghdl\ghdl\temp>ghdl -a --std=08 -frelaxed-rules -Wno-shared --work=psi_fix psi_fix_lin_approx_sin18b_tb.vhd 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1270.gb65ee8c3) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
D:\ghdl\ghdl\temp\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -frelaxed-rules -Wno-shared --work=psi_fix -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o psi_fix_lin_approx_sin18b_tb.o psi_fix_lin_approx_sin18b_tb.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap8.adb:3164
Call stack traceback locations:
0x179dce6 0x61a271 0x616b50 0x61bb08 0x6226a8 0x622709 0x6264f4 0x62b8d1 0x5e59c6 0x5d7eaa 0x637218 0x638bba 0x63a495 0x402706 0x4013bf 0x4014f4 0x7ffaa4177032 0x7ffaa549d0cf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

dalex78, 2026-09-04:
# Issue #590 — individual association of an array element with dynamic bounds

Analysis by an AI agent. The crash the title describes is fixed; the same design now hits a second one, which this turns into a diagnostic.

## The reproducer, without the zip

The report attached `FailLinApprox.zip` — eight VHDL files, a 78 KB stimulus file and a `.bat` that drives them. None of it is needed. The construct is the call at `psi_fix_lin_approx_sin18b_tb.vhd:87`:

```vhdl
appy_textfile_content(  Clk       => Clk,
                        Rdy       => PsiTextfile_SigOne,
                        Vld       => InVld,
                        Data(0)   => InData,          -- <-- this
                        Filepath  => StimuliDir_g & "/stimuli.txt",
                        ClkPerSpl => 1);
```

where the formal is declared as

```vhdl
type TextfileData_t is array (natural range <>) of std_logic_vector;
...
procedure appy_textfile_content (... signal Data : out TextfileData_t; ...);
```

`Data(0) => InData` associates **one element of an array formal**, and the array's element subtype is unbounded, so its bounds are only known from the actual. In 40 lines:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

package repro_pkg is
  --  Array of an unbounded element: the bounds are dynamic.
  type slv_array_t is array (natural range <>) of std_logic_vector;

  procedure drive (signal data : out slv_array_t);
end package;

package body repro_pkg is
  procedure drive (signal data : out slv_array_t) is
  begin
    data (0) <= (data (0)'range => '1');
  end procedure;
end package body;

library ieee;
use ieee.std_logic_1164.all;
use work.repro_pkg.all;

entity repro is
end entity;

architecture a of repro is
  signal s : std_logic_vector (7 downto 0) := (others => '0');
begin
  process
  begin
    --  Individual association of an array element with dynamic bounds.
    drive (data (0) => s);
    wait for 1 ns;
    assert s = x"ff" report "bad value" severity failure;
    wait;
  end process;
end architecture;
```

## The original crash is gone; a different one took its place

The reported signature was `trans-chap3.adb:2566`, while analysing `psi_fix_lin_approx_calc.vhd`. On master that file analyses cleanly, and so do all seven others. The crash has moved to the **testbench**, and it is a different one:

```
raised ADA.ASSERTIONS.ASSERTION_ERROR : trans-chap8.adb:3304
```

on llvm while analysing, and on mcode and llvm-jit while elaborating. Line 3304 is

```ada
            if Dynamic_Individual_Assoc /= Null_Iir then
               --  With dynamic bounds.
               --  FIXME: only records are supported.
               pragma Assert (Get_Kind (Formal) = Iir_Kind_Selected_Element);
```

`Iir_Kind_Selected_Element` is a record element. The formal here is an `Iir_Kind_Indexed_Name`, so the assertion fails. The FIXME says exactly what is going on: individual association with dynamic bounds is implemented for records and not for arrays.

## What the fix does

It does not implement arrays. Doing so is real work and not a line or two: the record path takes the element bounds straight from the formal's element (`Chap3.Record_Bounds_To_Element_Bounds`) because a record's shape is fully known from the association list, whereas for an array the **index range** of the whole object has to be derived from the individual associations first, and nothing does that today -- the dynamic branch of the `Association_Element_By_Individual` case only calls `Allocate_Unbounded_Composite_Bounds` and leaves the index range to be filled in later.

So it reports it:

```
repro.vhd:36:24:error: individual association of an element of an array with dynamic bounds is not supported
    drive (data (0) => s);
                        ^
```

and stops. The stop matters: the code after the assertion builds the object out of the per-element bounds, and letting it run without them produced, on mcode, a *second* internal error one step further on --

```
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:462
```

-- so the diagnostic raises `Errorout.Compilation_Error` rather than falling through.

On the reporter's own files, straight out of the zip, both `Data(0) =>` calls are now reported by name and line, with no internal error on any backend.

## Testing

`testsuite/gna/issue590` is the reduction above. It asserts that the construct is rejected, that the message is the not-supported one, and that no `GHDL Bug occurred` box appears -- and it adapts to the backend, since gcc and llvm report while analysing and the pre-elaborated backends while elaborating.

On master the test fails on all three backends with the assertion; with the change all three pass, and `sanity`, `gna` and `synth` pass on mcode, llvm and llvm-jit.

## Recommendation

The title of #590 -- "crash on record individual assoc with dynamic bounds" -- is now fixed, and the array case is the part that is left. Either close it and open a feature request for arrays, or re-scope it to that; what should not stay is the internal error, and that is what this removes.