ghdl/ghdl #592: hwrite and to_hstring yield "no declaration"

Ravenwater, 2018-06-03:
trying to print a std_logic_vector in hex, but can't find the functions in GHDL. Where do they live?

Ravenwater, 2018-06-03:
Thanks for the pointer. Still having a cow though:

$ ghdl -a ./reuse/component/posit/tb/arch/bhv/pau_add_tb.vhdl --std=08
C:\eda\ghdl\bin\ghdl.exe: cannot open --std=08

as you can see I am using ghdl on Windows.
$ ghdl --version
GHDL 0.35 (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

1138-4EB, 2018-06-03:
@Ravenwater try putting `--std=08` before the file path, either before or after `-a`.

Paebbels, 2018-06-03:
The language option `--std=08` needs to be added before the file names.

`ghdl.exe -<command> <options> <filenames>`

`ghdl.exe -a --std=08 ./reuse/component/posit/tb/arch/bhv/pau_add_tb.vhdl`

Ravenwater, 2018-06-03:
Darn, I misread the help: it is right there: so sorry for wasting your time.

```
$ ghdl --help
usage: C:\eda\ghdl\bin\ghdl.exe COMMAND [OPTIONS] ...
COMMAND is one of:
-a [OPTS] FILEs    Analyze FILEs
-e [OPTS] UNIT [ARCH]  Elaborate UNIT
-r,--elab-run [OPTS] UNIT [ARCH] [RUNOPTS]  Run UNIT
-c [OPTS] FILEs -r UNIT [ARCH] [RUNOPTS]  Compile, elaborate and run UNIT
-m [OPTS] UNIT [ARCH]  Make UNIT
```

closed 2018-06-03 by Paebbels