ghdl/ghdl #594: Bound Check Failure

LogicSolutionsBrundler, 2018-06-04:
**Description**
When I run my simulation, GHDL reports a "bound check failure". I checked the code-line that is reported and to me it looks correct. The line assigns the output of a function to a variable. Both are standard logic vector and have the same width.
Of course there could be a bug inside the function but I would assume that GHDL would report the code-line within the function in that case. I also know that the code is properly working from modelsim simulations.

**Context**
C:\Program Files (x86)\GHDL\bin\ghdl.EXE:error: bound check failure at D:/git/Libraries/VHDL/psi_fix/testbench/psi_fix_lin_approx_tb/sin18b/psi_fix_lin_approx_sin18b_tb.vhd:77
  from: process psi_fix.psi_fix_lin_approx_sin18b_tb(sim).P0 at psi_fix_lin_approx_sin18b_tb.vhd:77
  from: [unknown caller]
  from: process psi_fix.psi_fix_lin_approx_calc(rtl).p_comb at psi_fix_lin_approx_calc.vhd:131

- OS: Win10 - 64-bit
- Version: 
[FailSin18b.zip](https://github.com/ghdl/ghdl/files/2066966/FailSin18b.zip)
GHDL release: 0.35 (tarball) [Dunoon edition]

**How to reproduce?**
I will attach a ZIP file containing the error case.

diogratia, 2018-06-04:
```
ghdl -r --std=08 psi_fix_lin_approx_sin18b_tb
./psi_fix_lin_approx_sin18b_tb:error: bound check failure at psi_fix_lin_approx_sin18b_tb.vhdl:76
./psi_fix_lin_approx_sin18b_tb:error: simulation failed
```
Where line 76 is:

```vhdl
InData <= std_logic_vector(to_unsigned(SigIn(0), InData'length));
```

Replace that with a process:

```vhdl
	-- InData <= std_logic_vector(to_unsigned(SigIn(0), InData'length));
    process (SigIn)
        variable  intermed: unsigned(InData'range);
    begin
        report "InData'length = " & integer'image(InData'length);
        report "intermed'range = (" & integer'image(intermed'left) &
        " downto " & integer'image(intermed'right) & ")";
        intermed := to_unsigned(SigIn(0), InData'length);  -- this is line 83
        InData <= std_logic_vector(intermed);
    end process;
```
And produces:

```
ghdl -a --std=08 psi_fix_lin_approx_sin18b_tb.vhdl
ghdl -e --std=08 psi_fix_lin_approx_sin18b_tb          
ghdl -r --std=08 psi_fix_lin_approx_sin18b_tb
psi_fix_lin_approx_sin18b_tb.vhdl:80:9:@0ms:(report note): InData'length = 20
psi_fix_lin_approx_sin18b_tb.vhdl:81:9:@0ms:(report note): intermed'range = (19 downto 0)
./psi_fix_lin_approx_sin18b_tb:error: bound check failure at psi_fix_lin_approx_sin18b_tb.vhdl:83
./psi_fix_lin_approx_sin18b_tb:error: simulation failed
```
Where we can see the variable assignment failed. 

I'll cut to the chase. The to_unsigned conversion failed because InData(0) (type integer) is not contained in the natural range of `arg` in package declaration numeric_std:

```vhdl
  -- Id: D.3
  function TO_UNSIGNED (ARG, SIZE : NATURAL) return UNRESOLVED_UNSIGNED;
  -- Result subtype: UNRESOLVED_UNSIGNED(SIZE-1 downto 0)
  -- Result: Converts a nonnegative INTEGER to an UNRESOLVED_UNSIGNED vector with
  --         the specified SIZE.
```
SigIn(0) has a default value of integer'left which is not a natural range value, it's instead the maximum negative integer value.

LogicSolutionsBrundler, 2018-06-05:
Thank you for the help. This solved the issue.

Wouldn't it be very useful if the error message provided a bit more information such as:
- Simulation time the error occurs
- Expected range for  the bound-check that failed
- Actual value for the bound-check that failed

This would allow users to find their errors easier (and hence reduce the amount of support work as required for this case... sorry :-|).

abyszuk, 2020-09-19:
I've just run into very similar issue.
It would be very helpful if GHDL could output more information just like `ghost` proposed

tgingold, 2020-09-20:
Do you have a test case ?  I think the initial case has now more information.