ghdl/ghdl #595: Which debian package provides cmath for llvm build

svenn71, 2018-06-04:
**Description**

Trying to compile LLVM version of ghdl on a computer which I have been doing this for years now. After a dist-upgrade, I suddenly get error message about a missing cmath include file which has nothing to do with ghdl code. I am trying to track down which debian package is providing the cmath which llvm is looking for, but somehow my search lore with apt and google is failing me.

```
svennb@no-mrsim:~/work/prg/git/ghdl$ ./configure --with-llvm-config=llvm-config-6.0
Creating ghdl.gpr
Creating Makefile
for d in src/std src/synopsys src/openieee std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 src/ieee src/vital95 src/vital2000 src/mentor src/ieee2008 mentor/v93 ieee/v08 synopsys/v08; do
\
  mkdir -p lib/ghdl/$d; \
done
Generate default_paths.ads

svennb@no-mrsim:~/work/prg/git/ghdl$ make
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.36-dev/" < src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNAT_FLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="llvm-config-6.0" CXX="clang++" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/svennb/work/prg/git/ghdl'
clang++ -c `llvm-config-6.0 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
clang: warning: -Wl,--no-keep-files-mapped: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--no-map-whole-files: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-fuse-ld=gold' [-Wunused-command-line-argument]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
In file included from src/ortho/llvm4-nodebug/llvm-cbindings.cpp:18:
In file included from /usr/lib/llvm-6.0/include/llvm-c/Target.h:22:
In file included from /usr/lib/llvm-6.0/include/llvm-c/Types.h:17:
/usr/lib/llvm-6.0/include/llvm/Support/DataTypes.h:33:10: fatal error: 'cmath' file not found
#include <cmath>
         ^~~~~~~
1 warning and 1 error generated.
make[1]: *** [src/ortho/llvm4-nodebug/Makefile:18: llvm-cbindings.o] Error 1
make[1]: Leaving directory '/home/svennb/work/prg/git/ghdl'
make: *** [Makefile:304: ghdl1-llvm] Error 2
```

svenn71, 2018-06-04:
I have Debian Sid and when I re-run the same commands I get:

```
$ clang++ --version
clang version 4.0.1-10+b1 (tags/RELEASE_401/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```
which looks the same except for the +b1 on version

```
$ llvm-config-6.0 --cxxflags
-I/usr/lib/llvm-6.0/include -std=c++0x -fuse-ld=gold -Wl,--no-keep-files-mapped -Wl,--no-map-whole-files -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wno-maybe-uninitialized -Wdelete-non-virtual-dtor -Wno-comment -ffunction-sections -fdata-sections -O2 -DNDEBUG  -fno-exceptions -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS
```
which looks the same

And then comes the error.
```
$ clang++ `llvm-config-6.0 --cxxflags` -c -o test.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
clang: warning: -Wl,--no-keep-files-mapped: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--no-map-whole-files: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-fuse-ld=gold' [-Wunused-command-line-argument]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
In file included from src/ortho/llvm4-nodebug/llvm-cbindings.cpp:18:
In file included from /usr/lib/llvm-6.0/include/llvm-c/Target.h:22:
In file included from /usr/lib/llvm-6.0/include/llvm-c/Types.h:17:
/usr/lib/llvm-6.0/include/llvm/Support/DataTypes.h:33:10: fatal error: 'cmath' file not found
#include <cmath>
         ^~~~~~~
1 warning and 1 error generated.
```
I've been trying to find out where that <cmath> file should be located. The ones I find are
```
$  locate -A --regex usr '\<cmath$'

/usr/include/boost/compatibility/cpp_c_headers/cmath
/usr/include/boost/tr1/tr1/cmath
/usr/include/c++/4.7/cmath
/usr/include/c++/4.7/tr1/cmath
/usr/include/c++/4.9/cmath
/usr/include/c++/4.9/ext/cmath
/usr/include/c++/4.9/tr1/cmath
/usr/include/c++/5/cmath
/usr/include/c++/5/ext/cmath
/usr/include/c++/5/tr1/cmath
/usr/include/c++/6/cmath
/usr/include/c++/6/ext/cmath
/usr/include/c++/6/tr1/cmath
/usr/include/c++/7/cmath
/usr/include/c++/7/ext/cmath
/usr/include/c++/7/tr1/cmath
/usr/lib/rstudio/resources/libclang/builtin-headers/libc++/3.5/cmath
```

andreygursky, 2018-06-04:
> which looks the same except for the +b1 on version

Rebuild for ncurses update. Shouldn't be relevant.

You can try with -v flag:
```
$ clang++ -v `llvm-config-6.0 --cxxflags` -c -o test.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
clang version 4.0.1-10 (tags/RELEASE_401/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/i586-linux-gnu/4.8
Found candidate GCC installation: /usr/bin/../lib/gcc/i586-linux-gnu/4.8.5
Found candidate GCC installation: /usr/bin/../lib/gcc/i586-linux-gnu/4.9
Found candidate GCC installation: /usr/bin/../lib/gcc/i586-linux-gnu/4.9.3
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/5
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/5.4.1
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/7
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/7.3.0
Found candidate GCC installation: /usr/bin/../lib/gcc/i686-linux-gnu/8
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.7
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.7.4
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.8
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.8.5
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.9
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.9.3
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/5
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.1
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/6
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/6.4.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/i586-linux-gnu/4.8
Found candidate GCC installation: /usr/lib/gcc/i586-linux-gnu/4.8.5
Found candidate GCC installation: /usr/lib/gcc/i586-linux-gnu/4.9
Found candidate GCC installation: /usr/lib/gcc/i586-linux-gnu/4.9.3
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/5
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/5.4.1
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/7
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/7.3.0
Found candidate GCC installation: /usr/lib/gcc/i686-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.7
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.7.4
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.8.5
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.9
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.9.3
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/5
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/5.4.1
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/6
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/6.4.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7.3.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/8
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0
Candidate multilib: .;@m64
Candidate multilib: 32;@m32
Candidate multilib: x32;@mx32
Selected multilib: .;@m64
clang: warning: -Wl,--no-keep-files-mapped: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--no-map-whole-files: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-fuse-ld=gold' [-Wunused-command-line-argument]
 "/usr/lib/llvm-4.0/bin/clang" -cc1 -triple x86_64-pc-linux-gnu -emit-obj -disable-free -disable-llvm-verifier -discard-value-names -main-file-name llvm-cbindings.cpp -mrelocation-model pic -pic-level 2 -mthread-model posix -fmath-errno -masm-verbose -mconstructor-aliases -munwind-tables -fuse-init-array -target-cpu x86-64 -momit-leaf-frame-pointer -v -dwarf-column-info -debugger-tuning=gdb -ffunction-sections -fdata-sections -coverage-notes-file test.gcno -resource-dir /usr/lib/llvm-4.0/bin/../lib/clang/4.0.1 -I /usr/lib/llvm-6.0/include -D NDEBUG -D _GNU_SOURCE -D __STDC_CONSTANT_MACROS -D __STDC_FORMAT_MACROS -D __STDC_LIMIT_MACROS -internal-isystem /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0 -internal-isystem /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/x86_64-linux-gnu/c++/7.3.0 -internal-isystem /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/x86_64-linux-gnu/c++/7.3.0 -internal-isystem /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0/backward -internal-isystem /usr/include/clang/4.0.1/include/ -internal-isystem /usr/local/include -internal-isystem /usr/lib/llvm-4.0/bin/../lib/clang/4.0.1/include -internal-externc-isystem /usr/include/x86_64-linux-gnu -internal-externc-isystem /include -internal-externc-isystem /usr/include -O2 -Werror=date-time -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wno-long-long -Wno-maybe-uninitialized -Wdelete-non-virtual-dtor -Wno-comment -pedantic -std=c++1 [... cut ...]
ignoring duplicate directory "/usr/include/clang/4.0.1/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/llvm-6.0/include
 /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0
 /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/x86_64-linux-gnu/c++/7.3.0
 /usr/bin/../lib/gcc/x86_64-linux-gnu/7.3.0/../../../../include/c++/7.3.0/backward
 /usr/include/clang/4.0.1/include
 /usr/local/include
 /usr/include/x86_64-linux-gnu
 /usr/include
End of search list.
1 warning generated.
```

svenn71, 2018-06-05:
Diffing the two logs revealed that my instance called gcc-8 while the working one called gcc-7.3.0 so I took the easy way out and apt removed gcc-8 and apt autoremoved the leftovers. This caused clang++ to pick up gcc-7.3.0 in my environment, too, and compilation succeeded. Thumbs up!! I never asked for gcc-8, and nothing I have installed seems to be depending on its prescense, so I leave it at this.

Thanks a lot for the help, and the very valuable debug-the-sanity-of-your-build-platform steps.

closed 2018-06-05 by svenn71

andreygursky, 2018-06-05:
> Diffing the two logs revealed that my instance called gcc-8 while the working one called gcc-7.3.0

Yes. And you have been missing the libstdc++-8-dev package. Actually one would expect that C++ compiler wouldn't pick up other compiler paths with missing C++ STL headers.