ghdl/ghdl #598: to_string on std_logic_vector constant causes IIR_PREDEFINED_ARRAY_CHAR_TO_STRING

mjerabek, 2018-06-05:
**Description**
Calling to_string() on std_logic_vector declared as constant causes internal compiler error.
If the `constant` is changed to `variable`, everything works.

**Expected behaviour**
It works identically in both cases.

**Context**
```
internal error: eval_monadic_operator: IIR_PREDEFINED_ARRAY_CHAR_TO_STRING

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (tarball) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /home/mjerabek/src/                                                                                                                                                                                                                                               
Command line:                                                                                                                                                                                                                                                                   
/opt/ghdl/libexec/gcc/x86_64-pc-linux-gnu/6.3.0/ghdl1 --std=08 -P/opt/ghdl/lib/ghdl/v08/std/ -P/opt/ghdl/lib/ghdl/v08/ieee/ -quiet -o tb.s tb.vhd                                                                                                                               
Exception TYPES.INTERNAL_ERROR raised                                                                                                                                                                                                                                           
Exception information:                                                                                                                                                                                                                                                          
raised TYPES.INTERNAL_ERROR : errorout.adb:781                                                                                                                                                                                                                                  
Call stack traceback locations:                                                                                                                                                                                                                                                 
0x637821 0x653042 0x64eb68 0x64fff3 0x707b25 0x715c95 0x71349d 0x6fcb9b 0x6fd07d 0x6fd7be 0x6fd7f9 0x6fdf45 0x6fd8cc 0x7222ed 0x6628d3 0x6f49a5 0x62970d 0x5fda0e 0xa9d6ed 0x5f7f04 0x602b19 0x629769 0x5f6658 0x7f8af0fbc2df 0x5fa488 0xfffffffffffffffe                       
******************************************************************
```
- OS: Linux
- Origin: Built from sources. Tarball: v0.35

GHDL 0.35 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 6.3.0
 GCC back-end code generator

**How to reproduce?**

```vhdl
#>> tb.vhd
library ieee;
use ieee.std_logic_1164.all;

entity tb is
end entity;

architecture tb of tb is
begin
    process
        constant a : std_logic_vector(3 downto 0) := x"A";
        variable b : std_logic_vector(3 downto 0) := x"B";
    begin
        report to_string(b); -- OK
        report to_string(a); -- fails
    end process;
end architecture;

#>> sim.sh
ghdl -a --std=08 tb.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

1138-4EB, 2018-06-06:
I can reproduce it with the issue-runner:

```
internal error: eval_monadic_operator: IIR_PREDEFINED_ARRAY_CHAR_TO_STRING

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-200-g2627c62f) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
ghdl -a --std=08 tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:781
Call stack traceback locations:
0x557d3aa07280 0x557d3a9f565c 0x557d3a9fd080 0x557d3a9fd239 0x557d3a9fd791 0x557d3aa331c6 0x557d3a9ddc30 0x557d3a9debec 0x557d3a9abc06 0x557d3a9ad1ac 0x557d3a9ad317 0x557d3a9adf68 0x557d3a9ae44e 0x557d3a9ae76f 0x557d3a9df7db 0x557d3a9e60a9 0x557d3a9a16f4 0x557d3ab30f0e 0x557d3ab20a03 0x557d3ab4cc56 0x557d3a85c5c3 0x7f60f7484a85 0x557d3a85b3e8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2018-06-06:
I can reproduce the crash and I will fix it.

commit 26800bcd (2018-06-08): evaluation: handle to_string for arrays.

Fix #598
closed 2018-06-08 by tgingold