ghdl/ghdl #600: Support generic-derived constants as "numbers" in PSL

lainy, 2018-06-07:
**Is your feature request related to a problem? Please describe.**
Writing a core where the minimum time between changes (fastest toggle rate) of one of its outputs is determined by a generic. It'd be nice to use a PSL assertion to validate that, such as:
```vhdl
-- psl toggle_rate_01: assert always { s = '0'; s = '1' } |=> { (s = '1')[*DelayCycles] };
-- psl toggle_rate_10: assert always { s = '1'; s = '0' } |=> { (s = '0')[*DelayCycles] };
```
where `DelayCycles` is a constant computed from some generic.

However, when attempting the above, I get a compilation failure saying "number expected" at the location of `DelayCycles`. Indeed it works if I replace all the uses of generic-derived constants with numeric literals.
The same applies when using `next_a`.

**Describe the solution you'd like**
Support for generic-derived constants in PSL as above.

IEEE 1850-2010 (PSL) ambiguously defines "number" as "A non-negative integer value, and a statically computable expression yielding such a value." I'd say a generic is statically computable, but that only applies _per instance_, so it is unclear to me if my desired use case should be legal or not.

**Describe alternatives you've considered**
Have not tried any alternatives yet, open to suggestions here.

**Additional context**
None.

tgingold, 2018-06-07:
Generics are not locally static (but they are globally static).  I have to investigate what 'statically computable expressions' means in IEEE 1850.

tmeissner, 2018-06-07:
I support the feature request. I can test the examples with Questa later at work if of interest.

rjordans, 2020-04-14:
This would be nice to have indeed, I just ran into this in some of my own code as well.

Topi-ab, 2025-03-08:
This would be really useful for generalizing checkers.