ghdl/ghdl #603: VHDL access C code

artintest, 2018-06-10:
Hello,
I am not that experienced with ghdl. I am trying to test a sample VHDL code that accesses C code. The code is provided here: http://ygdes.com/GHDL/access_c/
I get the following error when I try to use `-Wl` option with elaboration `-e` command:
`ghdl:error: option -Wl is not available when ghdl is not configured with gcc or llvm`

It seems like my ghdl is configured in "mcode", and I have tried to build and configure it on "gcc" with the following sequence of commands:
`cd <ghdl>`
`mkdir build`
`cd build`
`../configure --prefix=PREFIX`
`make`
`make install`

The `make` raises a couple of Warnings and then the `make install` does the installation but gives the following feedback:
`install: cannot stat 'lib/ghdl/src/openieee/*': No such file or directory`

And at the end I have my ghdl configured with "mcode" still.

Could you please kindly help me?
Thank you one million.

1138-4EB, 2018-06-10:
Hi @artintest! You are following the build instructions for the default built-in backend, which is mcode. You should visit the [Building](http://ghdl.readthedocs.io/en/latest/building/Building.html) section in the docs. There, you will find some hints and links to [build-llmv](http://ghdl.readthedocs.io/en/latest/building/llvm/index.html#build-llvm) and [build-gcc](http://ghdl.readthedocs.io/en/latest/building/gcc/index.html#build-gcc). I suggest to use the LLVM version, as it is easier to build.

Alternatively, you can check [dist/travis/build.sh](https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh). That's the build script that is used in the CI environment to build any of the three supported backends.

Lats, there are ready-to-use docker images available at https://hub.docker.com/r/ghdl/ghdl/tags/. Just pick the one you prefer. If you want prebuilt packages (in scratch images), you can find them at https://hub.docker.com/r/ghdl/pkg/tags/.

Hope this helps you. Please let us know if you have any further doubts. I'd be pleased to know if you get to have it done.

artintest, 2018-06-11:
@1138-4EB Thank you very much for your thorough response. I decided to go with llvm, and installed the llvm package, then tried to configure the ghdl with llvm by running ``` ./configure --with-llvm-config``` in my ghdl directory, but the configuration didn't change, and it is still 'mcode'. Would you please, am I doing it wrong?

artintest, 2018-06-11:
Because compiling from source as mentioned in the link you provided, took a huge space, and I was unable to do that. I tried to install the llvm package and llvm-4.0-dev, and then ran the above mentioned ./configure command.

1138-4EB, 2018-06-11:
Could you, please, provide some more info?
- Platform/architecture
- OS/distribution and version
- Version of LLVM/clang+
- Exact sequence of commands: git clone, configure, prefix...

artintest, 2018-06-12:
@1138-4EB , Thanks for you response. Yes, the details are as following:

Platform/architecture:
```
Architecture:          i686
CPU op-mode(s):        32-bit, 64-bit
```
Os/distribution and version: LinuxMint 18.3 Sylvia 32 bit
```
Distributor ID:	LinuxMint
Description:	Linux Mint 18.3 Sylvia
Release:	18.3
Codename:	sylvia
Linux 4.10.0-38-generic i686
```
The following LLVM packages are installed right now:
```
llvm/xenial-updates,now 1:3.8-33ubuntu3.1 i386 [installed]
llvm-3.8/xenial-updates,now 1:3.8-2ubuntu4 i386 [installed,automatic]
llvm-3.8-runtime/xenial-updates,now 1:3.8-2ubuntu4 i386 [installed,automatic]
llvm-4.0/xenial-updates,xenial-security,now 1:4.0-1ubuntu1~16.04.2 i386 [installed,automatic]
llvm-4.0-dev/xenial-updates,xenial-security,now 1:4.0-1ubuntu1~16.04.2 i386 [installed]
llvm-4.0-runtime/xenial-updates,xenial-security,now 1:4.0-1ubuntu1~16.04.2 i386 [installed,automatic]
llvm-5.0/xenial-updates,now 1:5.0-3~16.04.1 i386 [installed]
llvm-5.0-runtime/xenial-updates,now 1:5.0-3~16.04.1 i386 [installed,automatic]
llvm-runtime/xenial-updates,now 1:3.8-33ubuntu3.1 i386 [installed,automatic]
```

The Clang+ version is: 
```clang version 3.8.0-2ubuntu4```

And the sequence of actions and commands, that I ended up here after running them is:

- I downloaded .zip file, and unzipped it. (Tried this for .tar.gz and also tried to git clone, and the results were the same)

-  Then cd to the root folder containing ghdl

-  Then ``` ./configure --with-llvm-config```

-  And ``` make``` brought me to the point in which I am stuck.

I hope this has provided the required details that you asked about, and I am looking forward to hearing from you.

1138-4EB, 2018-06-13:
Glad you solved it! I was about to start tinkering with it, when I saw that you updated the comment. Congratulations!

Just for completeness, note that you had to install `llvm-3.8-dev` because you are using `./configure --with-llvm-config`, which picks the default LLVM version in your machine (3.8). Should you want to use LLVM 5.0 (i.e. `llvm-5.0-dev`), you need to explicitly set it with: `./configure --with-llvm-config=llvm-config-5.0 CXX=clang++-5.0`.

See https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh#L102-L124

For me to know, did the example (http://ygdes.com/GHDL/access_c/) work as is? Or did you have to make any modification to the sources?

Last, please close this issue, as it is solved now. Feel free to reopen it in the future or to open a new one if your have further issues.

artintest, 2018-06-13:
Hi @1138-4EB , Thanks for your message. 

Yes, the example at (http://ygdes.com/GHDL/access_c/) worked for me.

Peace & Blessings.

closed 2018-06-13 by artintest

janslab, 2019-11-21:
Hello,
I tried to build the llvm version of GHDL. I got the same error as diogratia (fatal errror: 'llvm-c/Target.h' file not found). As you suggest I specified the llvm and clang version, but the error remained.

The output is as follows: 
```
Cloning into 'ghdl'...
remote: Enumerating objects: 471, done.
remote: Counting objects: 100% (471/471), done.
remote: Compressing objects: 100% (235/235), done.
remote: Total 39883 (delta 312), reused 359 (delta 236), pack-reused 39412
Receiving objects: 100% (39883/39883), 54.80 MiB | 27.03 MiB/s, done.
Resolving deltas: 100% (30676/30676), done.
Use full IEEE library
Build machine is: x86_64-linux-gnu
Debugging is not enabled with llvm 6.0.0
create pic/ subdirectory
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
  mkdir -p lib/ghdl/$d; \
done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads
if test -d /home/janslab/Downloads/src/ghdl/.git \
   && desc=`cd /home/janslab/Downloads/src/ghdl; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.37-dev/" < /home/janslab/Downloads/src/ghdl/src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=/home/janslab/Downloads/src/ghdl/src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI/home/janslab/Downloads/src/ghdl/src -aI/home/janslab/Downloads/src/ghdl/src/vhdl -aI/home/janslab/Downloads/src/ghdl/src/grt -aI/home/janslab/Downloads/src/ghdl/src/psl -aI/home/janslab/Downloads/src/ghdl/src/vhdl/translate -aI/home/janslab/Downloads/src/ghdl/src/ghdldrv -aI/home/janslab/Downloads/src/ghdl/src/ortho -aI/home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug -aI/home/janslab/Downloads/src/ghdl/src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="llvm-config-6.0" CXX="clang++-6.0" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/janslab/Downloads/src/llvm-ghdl-build'
rm -f -f /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  DO NOT MODIFY - this file was generated from:" > /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  ortho_nodes.common.ads and ortho_llvm.private.ads" \
  >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--" >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
sed -e '/^private/,$d' \
  < /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.private.ads >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  Start of common part" >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
sed -e '1,/^package/d' -e '/^private/,$d' < /home/janslab/Downloads/src/ghdl/src/ortho/ortho_nodes.common.ads >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  End of common part" >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
sed -n -e '/^private/,$p' \
  < /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.private.ads >> /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
chmod a-w /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.ads
clang++-6.0 -c `llvm-config-6.0 --cxxflags` -o llvm-cbindings.o /home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/llvm-cbindings.cpp
clang: warning: -Wl,--no-keep-files-mapped: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--no-map-whole-files: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-fuse-ld=gold' [-Wunused-command-line-argument]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean
      '-Wno-uninitialized'? [-Wunknown-warning-option]
/home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/llvm-cbindings.cpp:18:10: fatal error: 
      'llvm-c/Target.h' file not found
#include "llvm-c/Target.h"
         ^~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
/home/janslab/Downloads/src/ghdl/src/ortho/llvm4-nodebug/Makefile:18: recipe for target 'llvm-cbindings.o' failed
make[1]: *** [llvm-cbindings.o] Error 1
make[1]: Leaving directory '/home/janslab/Downloads/src/llvm-ghdl-build'
Makefile:343: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
mkdir -p /home/janslab/bin/ghdl-llvm
mkdir -p /home/janslab/bin/ghdl-llvm/bin
mkdir -p /home/janslab/bin/ghdl-llvm/lib
[... 46 lines cut ...]
cd $WORKDIR
git clone https://github.com/ghdl/ghdl.git

# create the build directory
mkdir llvm-ghdl-build
cd llvm-ghdl-build

# configure GHDL
$WORKDIR/ghdl/configure --with-llvm-config=llvm-config-${LLVM_VERSION} CXX=clang++-${LLVM_VERSION} --prefix=${INSTALLDIR}
make
make install
```

Version of llvm and clang:
```console
janslab@machine4:~$ llvm-config --version
6.0.0
janslab@machine4:~$ clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Where do I go wrong.

1138-4EB, 2019-11-21:
@janslab, it seems to me that you installed `llvm-6.0` (or `llvm`). To build GHDL, you need `llvm-6.0-dev` (or `llvm-dev`). That's why you are missing `llvm-c/Target.h`. You do have runtime dependencies, but you need sources too.

janslab, 2019-11-21:
Thank you @1138-4EB , that solved the problem. I had no idea that I needed a -dev version of it. The manual didn't say anything about such a version. Am I such a noob at Linux that I should have figured that out, or should the manual have a hint about this?

1138-4EB, 2019-11-21:
It is a noob issue, but we've all been there not only once, so it's perfectly ok to have these mistakes. The point to remember is that, whenever you get a complain about missing header files (`*.h`), you either forgot to install some `-dev` package, or your are not providing the compiler the paths it requires.

Note that, to make it a little *fun*, packages with header files are not always guessed by appending `-dev` to the *regular*/*runtime* package. It is a convention in Debian; so you can expect it on Ubuntu, Mint, etc. However, other distributions use `-devel`. Furthermore, naming versions is neither consistent. For example, on Alpine Linux it's named `llvm6-dev`. Last, on some distros, such as Arch, the logic is inverted: *regular* packages include sources, and *extras* might exist which contain pre-built runtime libs. That's why it might be painful even for experienced users to find the set of packages that are required for a not-packaged/not-maintained tool.

Apart from checking the recipes for ghdl that are used in Debian (or other distros), if your focus is on Mint, you can use the following dockerfiles as a reference:

- https://github.com/ghdl/docker/blob/master/dockerfiles/build_debian
- https://github.com/ghdl/docker/blob/master/dockerfiles/run_debian

Say you want to build ghdl with GCC backend. You need:

https://github.com/ghdl/docker/blob/178cdbcdf195b5b0c10540c959d7fc6f4b005dae/dockerfiles/build_debian#L16-L19

https://github.com/ghdl/docker/blob/178cdbcdf195b5b0c10540c959d7fc6f4b005dae/dockerfiles/build_debian#L43-L51

Or, say you want to run a pre-built ghdl with LLVM backend. You need:

https://github.com/ghdl/docker/blob/178cdbcdf195b5b0c10540c959d7fc6f4b005dae/dockerfiles/run_debian#L15-L18

https://github.com/ghdl/docker/blob/178cdbcdf195b5b0c10540c959d7fc6f4b005dae/dockerfiles/run_debian#L28

https://github.com/ghdl/docker/blob/178cdbcdf195b5b0c10540c959d7fc6f4b005dae/dockerfiles/run_debian#L50