ghdl/ghdl #604: Docker image ghdl/ext:vunit ImportError: No module named vunit

katutxakurra, 2018-06-11:
**Description**
New to Docker this could be a  problem with my workflow, sorry if that is the case.

Pull ghdl/ext:vunit docker image
Run command to test vunit code
Python doesn't find vunit module
$ docker pull ghdl/ext:vunit
$ git clone https://github.com/VUnit/vunit/
$ docker run --rm -t -v /$(pwd)://work -w //work ghdl/ext:vunit sh -c VUNIT_SIMULATOR=ghdl; for f in $(find ./vunit/examples/vhdl/user_guide/ -name 'run.py'); do python $f; done
Traceback (most recent call last):
  File "./vunit/examples/vhdl/user_guide/run.py", line 8, in <module>
    from vunit import VUnit
ImportError: No module named vunit

However, if I use bash in the container first:
$ docker run --rm -it -v /$(pwd)://work -w //work ghdl/ext:vunit bash
root@155a7fadf179:/work# for f in $(find vunit/examples/vhdl/user_guide/ -name 'run.py'); do python $f; done

It works.

**Expected behaviour**
Expected to find and run VUnit tests. Instead Python couldn't find the vunit module

**Context**
$ docker run --rm -t -v /$(pwd)://work -w //work ghdl/ext:vunit sh -c VUNIT_SIMULATOR=ghdl; for f in $(find ./vunit/examples/vhdl/user_guide/ -name 'run.py'); do python $f; done
Traceback (most recent call last):
  File "./vunit/examples/vhdl/user_guide/run.py", line 8, in <module>
    from vunit import VUnit
ImportError: No module named vunit

- OS: Linux in play-with-docker.com
- Origin:
$ docker pull ghdl/ext:vunit
vunit: Pulling from ghdl/ext
f2aa67a397c4: Pull complete
862a29fe9d1e: Pull complete
3227a4ed3d61: Pull complete
3174c48f7eb1: Pull complete
18529cf5d1ec: Pull complete
426c344e7e29: Pull complete
c07d49d8e8dc: Pull complete
ed989e5d2772: Pull complete
Digest: sha256:37119f1effcd8d53cbe90292eaa81ab7f92832dfe2ef5dca612c5633b6fd7eb6
Status: Downloaded newer image for ghdl/ext:vunit

**Additional context**

**How to reproduce?**
Follow commands in description section.

**Files**
N/A

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [ ] Try the latest version

1138-4EB, 2018-06-11:
Kaixo! Could you, please, put code and logs in code blocks, so that it is easier for us to read it? For example:

```
    ``` bash
    this is some dummy code
    ```
```

I'll then have a look at it ASAP.

katutxakurra, 2018-06-11:
Kaixo ;)
Sorry about that. Report updated.

1138-4EB, 2018-06-11:
No problem at all; we all need to learn ;).

It is much easier to spot the bug now. You just need to add quotes:

```
docker run --rm -t -v /$(pwd)://work -w //work ghdl/ext:vunit sh -c "VUNIT_SIMULATOR=ghdl; for f in $(find ./vunit/examples/vhdl/user_guide/ -name 'run.py'); do python $f; done"
```
Let me rewrite it, so that it is easier for you to see why quotes are required:
```
docker run \
  --rm \
  -t \
  -v /$(pwd)://work \
  -w //work \
  ghdl/ext:vunit \
  sh -c "VUNIT_SIMULATOR=ghdl; for f in $(find ./vunit/examples/vhdl/user_guide/ -name 'run.py'); do python $f; done"
```
As you see, the syntax is `sh -c your_command`. In this case `your_command` is `VUNIT_SIMULATOR=ghdl; for f in $(find ./vunit/examples/vhdl/user_guide/ -name 'run.py'); do python $f; done`. If you don't use quotes, each part of your command is considered to be an argument to sh, which is not correct.

---

Furthermore, if you want to execute multiple commands inside the container, I suggest to use the same approach that is shown in the [bug report template](https://github.com/ghdl/ghdl/blob/master/.github/ISSUE_TEMPLATE/bug_report.md):

First, put all your commands in a bash script, say `sim.sh`:
```
VUNIT_SIMULATOR=ghdl
for f in $(find ./vunit/examples/vhdl/user_guide/ -name 'run.py'); do
  python $f;
done
```
Then, pass the script as the argument when you run docker:
```
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ext:vunit sh -c ./sim.sh
```

I think that this approach is easier to understand for newcomers, even if it requires you to write an additional file.

---

As a side note, since a couple of weeks ago there is a specific repo for these docker related issues: https://github.com/ghdl/docker. Please, consider opening further similar issues there, so that we can keep this one for GHDL specific questions. Note that the docker image you used is based on GHDL, but this problem had nothing to do with it.

---

Last, if you are happy with the reply, please close this issue, as I have no privileges to do so.

1138-4EB, 2018-06-11:
@katutxakurra , just because of curiosity, may I know which path did you follow to try using this image in play-with-docker? Someone told you? Did you read some issue here? Did you find [this](https://github.com/ghdl/docker/blob/master/USE_CASES.md)?

There is some pending documentation work, and it would help us know how new user reach us.

katutxakurra, 2018-06-11:
@1138-4EB  thanks for the quick and very detailed solution. It makes perfect sense. 
I will keep in mind the correct location for the docker related issues.
I got to this image and play-with-docker from the VUnit documentation https://vunit.github.io/cli.html 
As you noticed, I made a mistake and I removed the quotes when copying the command.
Thanks for your help!

closed 2018-06-11 by katutxakurra