ghdl/ghdl #605: Documentation: More description for including custom libraries

iduhetonas, 2018-06-12:
Hello!

I would suggest that there should be a small update to the documentation, to be more explicit about how to include external libraries. My issue was, as follows:

- I have a huge project that expects to be in "work", and uses an external library named "igloo".
- I have the .vhd from this vendor, to which I can compile and make a library just by using --work=igloo

It turns out, the answer is just to include the extra library with the -P option. But this was buried deep in the documentation, and it wasn't explicit. I ended up using two days just to finally find this.

To find out how to include the library in my project, I ran into several dead-ends:
-No information here, but it seemed relevant at first: http://ghdl.free.fr/ghdl/Using-vendor-libraries.html#Using-vendor-libraries
-No information here: http://ghdl.free.fr/ghdl/GHDL-options.html
-Doing "ghdl --help" didn't give any indication
-I found the path where the standard libraries were included on my machine through --disp-config, and attempted adding it to my build system (not a good solution), and this didn't work.
-There was a very, very tiny clue here: https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#library-commands but I had to go to here: https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-workdir to see it mentioned as a side-note. Only after actually testing it did I understand it. It's also hard to know that -P is correct, because I would never expect -P to indicate "external library".

So my small request is as follows:

Add a section here: https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#library-commands that explicitly states how to include more libraries.

Apologies if this is not the right place for documentation requests.

1138-4EB, 2018-06-12:
Hi @iduhetonas. Thanks for the suggestion. I agree that it might be more explicit, but I don't know how it might be improved there (besides explicitly linking to the -P option).

I'd rather add a practical example to the Quick Start Guide or to a longer User Guide.

Do you want to help us by opening a PR so that we have a reference to work on? I don't mind whether it is just adding a couple of sentences where you said, or a more elaborated example. It is up to you to decide how much additional details are required.

You can find the sources of the documentation in this same directory, under folder 'docs'.

iduhetonas, 2018-06-12:
Hi @1138-4EB,

Thank you for the response!

I assume by PR, you mean a pull request. I will try, although I'm not much of a power-user when it comes to Github. :) Hopefully I can get something for you soon.

1138-4EB, 2018-06-12:
Yes, I mean a Pull Request. Don't worry about not being a power-user. You can just open the file in the browser and click edit: https://github.com/ghdl/ghdl/blob/master/doc/using/InvokingGHDL.rst It will automatically create a fork and a new branch for you (probably named `patch-1`). Then, you can just continue editing the file online. When you are ready to go, let us know and we will walk the remaining path together.