ghdl/ghdl #606: Exception TYPES.INTERNAL_ERROR : trans-chap9.adb:1922

mbrobbel, 2018-06-13:
**Description**
`ghdl -i` + `ghdl -m` fails with an exception.

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-218-gd69ff3ef) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/synopsys/v93/ -P/usr/local/lib/ghdl/ --elab unit_tb -l e~unit_tb.lst -c -fpic -o e~unit_tb.o e~unit_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap9.adb:1922
Call stack traceback locations:
0x59f5bd 0x623b4e 0x623cbd 0x6262ca 0x56a122 0x566f1e 0x5756bd 0x62970f 0x62b548 0x408e67 0x7f2c48392b95 0x408278 0xfffffffffffffffe
******************************************************************
ghdl: compilation error
```
- OS: ubuntu
- Origin:
  - [X] Docker
```
docker --version && docker images | grep ghdl
Docker version 18.05.0-ce, build f150324
ghdl/ghdl                             fedora28-gcc-8.1.0   5625ab441934        14 hours ago        902MB
ghdl/ghdl                             ubuntu18-llvm-5.0    0a70e782bd05        15 hours ago        323MB
```

**Additional context**
I forked https://github.com/johanpel/fletcher in https://github.com/mbrobbel/fletcher where I made some modifications in a `simulation` branch. My goal is to use ghdl for simulation.

There is an entity (`ColumnReader`) which takes a configuration string in the generic map (https://github.com/johanpel/fletcher/blob/master/hardware/vhdl/arrow/ColumnReader.vhd#L67), and based on this generates the required logic (https://github.com/johanpel/fletcher/blob/master/hardware/vhdl/arrow/ColumnReaderLevel.vhd#L146).

However, as can be seen here:
https://github.com/johanpel/fletcher/blob/master/hardware/vhdl/arrow/ColumnReaderLevel.vhd#L219
and here:
https://github.com/johanpel/fletcher/blob/master/hardware/vhdl/arrow/ColumnReaderArb.vhd#L355

`ColumnReaderLevel` can add another `ColumnReaderArb` which adds another `ColumnReaderLevel` which can add another `ColumnReaderArb` which adds another `ColumnReaderLevel` and so forth.

My guess is that this logic somehow is causing the exception. This is the condition triggering the exception:
https://github.com/ghdl/ghdl/blob/master/src/vhdl/translate/trans-chap9.adb#L1918-L1922

I'm using the `ghdl/ghdl` docker images to run `ghdl` like this:
```docker run --rm -v `pwd`:/src ghdl/ghdl:ubuntu18-llvm-5.0 bash -c "shopt -s globstar && ghdl -i /src/**/*.vhd && ghdl -m -v --ieee=synopsys ..."```

**How to reproduce?**
```
git clone https://github.com/mbrobbel/fletcher.git
cd fletcher
git checkout simulation
docker run --rm -v `pwd`:/src ghdl/ghdl:ubuntu18-llvm-5.0 bash -c "shopt -s globstar && ghdl -i /src/examples/**/*.vhd /src/hardware/**/*.vhd && ghdl -m -v --ieee=synopsys ColumnReaderLevel"
```
This seems relevant: https://github.com/ghdl/ghdl/issues/2

diogratia, 2018-06-14:
I cloned your repository twice and got different versions of the hardware/vhdl/utils/Utils.vhd file, varying by whether or not bodies for functions matched their declarations (the bodies missing the explicit mode **in** that had been added to the declarations.

 VHDL IEEE Std 1076-2008 requires they match:

4.3 Subprogram bodies

>The declaration of a subprogram is optional. In the absence of such a declaration, the subprogram specification of the subprogram body acts as the declaration. For each subprogram declaration, there shall be a corresponding body. If both a declaration and a body are given, **_the subprogram specification of the body shall lexically conform (see 4.10) to the subprogram specification of the declaration_**. Furthermore, both the declaration and the body shall occur immediately within the same declarative region (see 12.1).

4.10 Conformance rules

>Whenever the language rules either require or allow the specification of a given subprogram to be provided in more than one place, the following variations are allowed at each place:  
>>— A numeric literal can be replaced by a different numeric literal if and only if both have the same value.  
>>— A simple name can be replaced by an expanded name in which this simple name is the suffix if, and only if, at both places the meaning of the simple name is given by the same declaration.  
>  
>Two subprogram specifications are said to _lexically conform_ if, apart from comments and the preceding allowed variations, both specifications are formed by the same sequence of lexical elements and if corresponding lexical elements are given the same meaning by the visibility rules.

Mind this could be due to Github server distribution around the globe. After adding the missing mode **in**s (see attached vhdl.zip)
```
ghdl -i vhdl/*/*.vhd
ghdl -m --ieee=synopsys ColumnReader
```
will demonstrate the error.

```
elaborate columnreader

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-168-g252a916) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: /Users/diog_gratia/Desktop/fletcher_mbrobbel/hardware/
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/synopsys/v93/ -P/usr/local/lib/ghdl/ --elab columnreader -l e~columnreader.lst -c -o e~columnreader.o e~columnreader
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: trans-chap9.adb:1922
Load address: 0x106f48000
Call stack traceback locations:
0x1070b8cbd 0x1071137db 0x10711395f 0x107116391 0x10709f35f 0x107111045 0x1070aac48 0x1071185a8 0x106f93227 0x106f4ba4c
******************************************************************
ghdl:error: compilation error
```
Without quite so much abstraction.

While it's interesting to see complex designs this one looks like it could be simplified a bit further  to demonstrate the issue more concisely.

[vhdl.zip](https://github.com/ghdl/ghdl/files/2100700/vhdl.zip)
[LICENSE.txt](https://github.com/ghdl/ghdl/files/2100706/LICENSE.txt)

mbrobbel, 2018-06-14:
> I cloned your repository twice and got different versions of the hardware/vhdl/utils/Utils.vhd file, varying by whether or not bodies for functions matched their declarations (the bodies missing the explicit mode in that had been added to the declarations.

Did you checkout the `simulation` branch? I made the required modifications on that branch: https://github.com/mbrobbel/fletcher/commit/7d640b1feb2f0cdcb152416cd9f168d5e0999c09

> While it's interesting to see complex designs this one looks like it could be simplified a bit further to demonstrate the issue more concisely.

Agree. I'll try to find some time to extract the problematic logic to produce a minimal example. My thinking was that maybe someone can already point me towards a probable cause based on the exception.

tgingold, 2018-06-16:
I can reproduce the crash, and I am investigating.

tgingold, 2018-06-17:
The crash is due to the use of indirect recursion; not a common pattern.  Working on a fix.

tgingold, 2018-06-18:
I now have a simple reproducer.  Work in progress...

mbrobbel, 2018-06-19:
Thanks for the quick fix. Awesome :+1:

tgingold, 2018-06-19:
You're welcome.  I am often surprised by issues like this, that shows bugs that were present since the beginning.