ghdl/ghdl #607: allow passing arguments on the command line

faf5678, 2018-06-15:
made available through some kind of magic predefined function name.

__ghdl_argv_get(i);

1138-4EB, 2018-06-15:
Hi @faf5678! Could you please elaborate the feature proposal a bit further? GHDL currently supports both command line arguments and passing generics as such.

faf5678, 2018-06-15:
command line arguments that are passed through to the vhdl code.

you know like main(int argc, char ** argv)

Paebbels, 2018-06-15:
@faf5678  
This feature is not part of VHDL.  
Similar features are upcoming with VHDL-2018:
* API to environment variables: [LCS2016_006e](www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_006e).

   Package `ENV` will offer new functions:
   ```vhdl
     impure function GETENV(Name : STRING) return STRING;
     impure function GETENV(Name : STRING) return LINE;
   ```

faf5678, 2018-06-15:
why does it matter what the standard says.

i have a legitimate use case for it, so the standard is bad. if you care about compliance, hide useful extensions behind a enabler switch, so that you can opt in to the goodies.

now that's just a thought.

Paebbels, 2018-06-15:
@faf5678  
Maybe you should participate in the [IEEE P1076 working group](https://gitlab.com/gitlab-com/support-forum/issues/2576) and explain your use case. Then we can discuss it and get it integrated.

You can also propose a `GHDL` package and describe your feature. Please note, a simulator is called with lots of parameters, you don't want to see them all in VHDL ...

faf5678, 2018-06-15:
the environment variables proposal already covers my use case.

anyway, i am writing a ucf constraint file with report statements and i don't want this during normal simulation. and since i am not working with a waveform viewer, but only the log file to debug my code, i do care about what gets written there. i mean i can just use grep and be done with, but it just makes sense to be able to pass arguments to programs. every other programming language allows it.

diogratia, 2018-06-16:
You'd think writing to STD_OUTPUT would be easier than using report statements to produce information for a constraints file. It would allow you to filter lines by embedded prefix (or lack thereof) in whatever automated debugging you're doing.

Without changing the language, your debugging methodology, or ghdl there's at least a couple of ways to accomplish what you're trying to do (prevent report statements used to  produce a constraint file from executing during initialization or simulation when debugging).

You could move whatever report statements you have dealing used to generate a constraint file inside a process or processes into a generate statement conditionally elaborated based on a generic which can be passed on the command line (a tool implementation option in VHDL supported by ghdl) to prevent reporting. The generic could have whatever default value seems appropriate.

You could likewise use in a process without a sensitivity list while having a final wait statement. The process would execute once after elaboration during initialization and could write the constraint file directly using textio sidestepping the issue of STD_OUTPUT and console output going to the same output stream used for debugging.

1138-4EB, 2018-06-16:
The main point is that VHDL is not "another programming language", but a Hardware Description Language. As time goes by, multiple programming features are being added. But many of these are meant for simulation/metaprogramming only, and have little to no meaning from a hardware perspective. For example, environment variables are hard to conceive in a synthesised circuit.

The natural equivalent to command line args in any other language are generics, as these allow to parametrise your description, thus modifying the behaviour of the "program" with high level constants. GHDL allows to set generics as arguments.

tgingold, 2018-06-16:
Add a boolean generic to your top-level entity and use that generic to control the output.  Use command line option -gGEN=val to set the value of it during simulation.  This is quiet portable.

faf5678, 2018-06-16:
thx everyone for pointing out my stupidity.

had i known about -gGEN=val i wouldn't have come here.

you can close this now. thx for the tech support guys.

closed 2018-06-16 by Paebbels

faf5678, 2018-06-16:
@Paebbels 

as of ghdl0.35 -g does not appear with -r foo_tb --help

faf5678, 2018-06-16:
@1138-4EB

nah, i am a season professional, i am not hindered by background bias.

i was under the impression that -g would not exist, and that would be a foolish limitation.

Paebbels, 2018-06-16:
@1138-4EB  please open a new documentation enhancement issue, because this issue was related to passing command line arguments like in C. You can link to this issue as a follow up.