ghdl/ghdl #608: Add reference to `-gGENERIC=value` in the docs

1138-4EB, 2018-06-16:
Coming from #607. See [this](https://github.com/ghdl/ghdl/issues/607#issuecomment-397800186) comment.

@tgingold where do you think that we should put it? Is it an elaboration or a run option? Does this collide with the `-g` argument related to debugging options?

tgingold, 2018-06-16:
Currently it is a run option; but in the future it will be an elaboration option only.
Maybe it should be allowed as an elaboration option before being official ?

It doesn't collide with debug option -g as it contains an '='.

1138-4EB, 2018-06-16:
> Maybe it should be allowed as an elaboration option before being official?

Yes. Let me know when you implement it, and I will update the docs accordingly.

Bamban, 2019-03-29:
> in the future

Hello GHDL team.
How far is this future ? :)
I am currently (trying to) using cocotb to simulate my design. I have a small blocking point: my generics are generally set by a wrapper of my DUT. This needs for a wrapper is removed by cocotb. But I would like to set the generics necessary for elaboration (not runtime) at a higher level than my DUT. I think there is no way to currently do this, apart from the -g at elab option. Am I wrong ?

1138-4EB, 2019-03-29:
> I think there is no way to currently do this, apart from the -g at elab option, which is to be developed. Am I wrong ?

The `-g` option is not to be developed. It is already implemented and available. This issue is just about adding it to the docs, so that users like you know about it.

> But I would like to set the generics necessary for elaboration (not run-time) at a higher level than my DUT.

You might want to use [JSON-for-VHDL](https://github.com/Paebbels/JSON-for-VHDL) instead. This allows to pass multiple parameters as a single JSON encoded generic string, or to read the parameters from a JSON file. There is an example with VUnit: https://github.com/VUnit/vunit/tree/master/examples/vhdl/json4vhdl. But you can also use it without VUnit.

Bamban, 2019-03-29:
> The `-g` option is not to be developed. It is already implemented and available.
OK perfect. Now that I take a closer look, I can clearly see that -g is used without error with ```ghdl -e ...```. Problem of cocotb I have to solve: the EXTRA_ARG makefile variable is also passed to ```ghdl -a ...```, causing an error afterward. I'll contact them about that.

JSON-for-VHDL is a little bit too overkill for me. Having the generics set in the Makefile is enough for me right now. But thanks anyway for the pointer, it might prove useful.

tgingold, 2019-03-30:
@Bamban , what do you mean by 
> at a higher level than my DUT

Bamban, 2019-04-01:
How it worked with Qsys (using modelsim for BFM, hence I want to get rid of the BFMs and move to cocotb for writing them). Generics were set when generating bench with Qsys:
[image]

How it work now with cocotb, and the generics are set in the DUT. Each time I change them, it bothers me seeing git showing changes were there are none:
[image]

Bamban, 2019-04-01:
> > I think there is no way to currently do this, apart from the -g at elab option, which is to be developed. Am I wrong ?
> 
> The `-g` option is not to be developed. It is already implemented and available. This issue is just about adding it to the docs, so that users like you know about it.

Sorry to insist @1138-4EB , but -g is not an **elab** option yet, according to some simple test.
Here is my test file:
```vhdl
ibrary ieee;
use ieee.std_logic_1164.all;

entity dff is
	generic (
						dw : natural := 8
					);
	port(
				 c : in  std_logic;
				 d : in  std_logic;
				 q : out std_logic := '0');
end entity dff;

architecture rtl of dff is

	function init_report(i : natural) return natural is
	begin
		report "dff says" &
		" dw =" & natural'image(dw) severity note;
		return 0;
	end function;

	constant the_init_report : natural := init_report(0);

begin
	q <= d when rising_edge(c);
end architecture rtl;
```
And here is the steps to reproduce:
```bash
$ ghdl --version
GHDL 0.37-dev (v0.36-16-g0c27469d) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ ghdl -e -gdw=12 dff #crash
unknown option '-gdw=12'
ghdl:error: cannot open e~dff.lst
$ ghdl -e dff #does not crash
$ ./dff
dff.vhd:18:16:@0ms:(report note): dff says dw =8
$ ./dff -gdw=133
dff.vhd:18:16:@0ms:(report note): dff says dw =133

```

1138-4EB, 2019-04-01:
@Bamban, although I cannot test it ATM, this is probably because you are using LLVM as a backend. Building and running a simulation with GHDL involves three steps:

- Analyze.
- Elaborate.
- Run.

With mcode backend, elaborate and run are merged in a single stage, so `-g` does make sense as an argument to `-e`. However, with LLVM backend, elaboration does not need to know the actual value of a generic. Therefore, I am guessing that `-g` is not accepted for `-e` with LLVM. You should provide it to `ghdl -r` or when you execute the resulting binary directly, as if it was a cli arg to an app of yours.

1138-4EB, 2019-04-01:
> (using modelsim for BFM, hence I want to get rid of the BFMs and move to cocotb for writing them)

Just because of curiosity, what does cocotb allow you which you cannot do with VUnit?

tgingold, 2019-04-01:
Right now, you can only override the generics of the top level entity.

Bamban, 2019-04-02:
> elaboration does not need to know the actual value of a generic.

Is it? I have some `generate` statements that depends on generics. Are there no impact on `ghdl -e`? Are the generates sort of instantiated during runtime?
Because if it is the case, I can indeed move my generic setting as late as runtime.
I think I was misunderstanding the role of generics in the analyse/elaborate/run the whole time >_<

> Just because of curiosity, what does cocotb allow you to do which you cannot do with VUnit?

I'm not entirely sure right now, I'm just beginning to use it and don't clearly see the differences. I would say that I'm glad that the testbench writing is removed with cocotb. Plus I found that I can plug directly my Avalon interfaces with drivers/monitors/memories (my main concern). Just my two cents after 3 days discovering cocotb! :)

> Right now, you can only override the generics of the top level entity.

Thanks for the precision.

1138-4EB, 2019-04-02:
> Is it? I have some `generate` statements that depends on generics. Are there no impact on `ghdl -e`? Are the generates sort of instantiated during runtime?
> Because if it is the case, I can indeed [...]

GHDL is a compiler/simulator, not a synthesizer. Generates are nothing but 'if/else' or 'for' from a software perspective. So, no, you do not need the actual value of a generic for elaboration. You need to know that a generic exists, and which is the type. During compilation/elaboration, all the possibilities will be included in the binary. Then, at runtime, the value of the generic will decide which branches are taken during the internal execution. 'Instantiating' something in a simulator is just traversing a code block or branching to ignore it.

Should GHDL be a synthesizer, the 'runtime' stage would not exist, so generics would be resolved during elaboration or just after elaboration.
 
> I'm not entirely sure right now, I'm just beginning to use it and don't clearly see the differences. I would say that I'm glad that the testbench writing is removed with cocotb. Plus I found that I [...]

That's precisely why I asked. You seem to be new to cocotb, and I feel that it might not be the tool you need. Cocotb is a coroutine co-simulation framework, will VUnit is more of a verification toolset. If all you want is to use BFMs or Verification Components:

- http://vunit.github.io/verification_components/user_guide.html
- https://github.com/VUnit/vunit/blob/master/examples/vhdl/array_axis_vcs/src/test/tb_axis_loop.vhd#L141-L184

Unless you want/need to write your own BFMs (with custom logic), I think that VUnit is the way to go here. Related to generics and configurations, VUnit provides several mechanisms to make exploration easier: https://vunit.github.io/python_interface.html#configurations

Bamban, 2019-04-04:
Thank you for the precision. I updated my bench accordingly, setting generics at runtime.

Maybe I'll give VUnit a try if I'm not satisfied with cocotb! It's indeed interesting (and it's VHDL flavour). For now I have something that almost work so I'll stick with it.

umarcor, 2019-04-07:
Coming from VUnit/vunit#466, and in regard to comment *"currently it is a run option; but in the future it will be an elaboration option only*" by @tgingold ([above](https://github.com/ghdl/ghdl/issues/608#issuecomment-397822352)):

- Generics being a runtime option allows to generate a single binary (GCC/LLVM backends) and then execute it multile times with different generics.
- Would making generics an elaboration option only remove this feature? I.e., will a different binary be required for each set of generics (configuration)?

tgingold, 2019-04-07:
It's a trade off.  Some optimizations can be performed only after elaboration.  Hence the future need to generate code after elaboration.

PR #806 (2019-04-30): Update doc: announce v0.36, add info about interfacing to other languages
<strike>Depends on #805, because config option `--default-pic` is documented here.</strike>

This PR contains several modifications to the docs:

* announce `v0.36`
* add hint about `--synth` and `ghdlsynth`
* add option `-gGENERIC=VALUE` (Close #608)
* add hint about `--list-link` (#640) (#801)
* add hint about `--bind` (#781)
* move 'Interfacing to other languages' to it's own section, and extend it:
  * 'Linking GHDL to Ada/C'
  * 'Dynamically loading GHDL' (#805)
* fix shields not being shown in contribute nor in lic [... cut]
closed 2019-04-30 by tgingold