ghdl/ghdl #610: ASSERT_FAILURE error when compiling UVVM avalon_mm testbench

sinitame, 2018-06-18:
Hi,
I have a problem when I try to compile a testbench using UVVM Avalon MM VVC and I can't figure out how to fix this error :

```
 ******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-218-gd69ff3ef) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /home/sinitame/Documents/fanna/device/sim/simu/ghdl/ddr/
Command line:
/usr/local/bin/ghdl1-llvm --warn-no-vital-generic --std=08 -frelaxed-rules --workdir=work --work=work -P/home/sinitame/UVVM_COMP/ -P/usr/local/lib/ghdl/synopsys/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o work/tb_read_loop_uvvm.o ../../../testbenchs/ddr/tb_read_loop_uvvm.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs_utils.adb:1157
Call stack traceback locations:
0x7f2d0abc18d1 0x4d42e2 0x5ee770 0x5fdaca 0x57a87d 0x57adbc 0x57bb81 0x5b689c 0x5b1a42 0x5b22d4 0x61207f 0x59c928 0x617913 0x61814c 0x48b1b1 0x4451a8 0x7f2d0a0692df 0x4445a8 0xfffffffffffffffe
****************************************************************** 
```

I compiled UVVM libs in UVVM_COMP using UVVM v2.4.1 and the compile-uvvm.sh script provided in ghdl sources as UVVM compilation failed for the last release as mentionned in  #453.
Any idea ?

My testbench code: 

```vhdl 
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library bitvis_vip_avalon_mm;
use bitvis_vip_avalon_mm.vvc_methods_pkg.all;
use bitvis_vip_avalon_mm.td_vvc_framework_common_methods_pkg.all;

entity tb_read_loop_uvvm is
end tb_read_loop_uvvm;

architecture simu of tb_read_loop_uvvm is

	--generics
	constant ADDR_W : natural := 32; -- address width
	constant SYMB_W : natural := 8; -- symbol width
	constant N_SYMB : natural := 4; -- number of symbols
	constant BCNT_W : natural := 5; -- number of symbols
	constant BRST_S : natural := 4; --maximum number of bursts

	constant NB_OUT : natural := 12; -- output number

	-- Testbench signal declarations
	signal rstn     : std_logic := '0';
	signal clk      : std_logic := '1';
	signal clk_next : std_logic := '0';
	signal stop     : std_logic := '0';
	--Signals declaration
	signal pin              : std_logic_vector(ADDR_W-1 downto 0)           := (others => '0');
	signal nin              : std_logic_vector(SYMB_W*N_SYMB-1 downto 0)    := (others => '0');
	signal dout             : std_logic_vector(SYMB_W*N_SYMB-1 downto 0)    := (others => '0');
	signal tb_ivalid        : std_logic := '0';
	signal tb_oready        : std_logic;
	signal tb_done          : std_logic;
	signal tb_ovalid        : std_logic;
	signal tb_iready        : std_logic := '0';
	signal tb_readdata      : std_logic_vector(N_SYMB*SYMB_W-1 downto 0);
	signal tb_readdatavalid : std_logic;
	signal tb_waitrequest   : std_logic;
	signal tb_address       : std_logic_vector(ADDR_W-1 downto 0);
	signal tb_read          : std_logic;
	signal tb_byteenable    : std_logic_vector(N_SYMB*SYMB_W/8-1 downto 0);
	signal tb_burstcount    : std_logic_vector(BCNT_W-1 downto 0);
	signal tb_enable        : std_logic;
	signal tb_writeack      : std_logic; -- this signal is purely ignored

begin
	i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;

	dut: entity work.read_loop
		port map (
			clk               => clk,
			resetn            => rstn,
			pointer_in        => pin,
			n_input           => nin,
			ivalid            => tb_ivalid,
			oready            => tb_oready,
			data_out          => dout,
			iready            => tb_iready,
			ovalid            => tb_ovalid,
			ami_readdata      => tb_readdata,     
			ami_readdatavalid => tb_readdatavalid,
			ami_waitrequest   => tb_waitrequest,  
			ami_address       => tb_address,      
			ami_read          => tb_read,         
			ami_byteenable    => tb_byteenable,   
			ami_burstcount    => tb_burstcount,   
			ami_enable        => tb_enable,       
			ami_writeack      => tb_writeack     
			);
	

	avalon : entity bitvis_vip_avalon_mm.avalon_mm_vvc
		
		port map(	
		    	clk                                       => clk
			);

	simulation : process 
	begin
		-- Wait for UVVM to finish initialization
  		await_uvvm_initialization(VOID);

    		-- Print the configuration to the log
    		report_global_ctrl(VOID);
    		report_msg_id_panel(VOID);

    		--enable_log_msg(ALL_MESSAGES);
    		disable_log_msg(ALL_MESSAGES);
    		enable_log_msg(ID_LOG_HDR);
    		enable_log_msg(ID_SEQUENCER);
    		enable_log_msg(ID_UVVM_SEND_CMD);

	end process simulation;
end simu; 
```

Best regards,
Emrick

tgingold, 2018-06-20:
This is currently WIP.  Latest versions of UVVM are using unbounded arrays/records that aren't fully supported by ghdl.

sinitame, 2018-06-20:
Do you know when it will be available for release ?

tgingold, 2018-06-21:
I know how to implement that feature, but it requires a little bit of time.

commit 1d566ee1 (2018-11-22): Handle assignment of aggregate to an unbounded variable.

Fix #610
closed 2018-11-22 by tgingold