ghdl/ghdl #611: Vunit Segmentation fault

slaweksiluk, 2018-06-20:
**Description**
Segmentation fault when trying to allocate 1MB buffer with Vunit allocate() function. It works for smaller buffers eg 512KB.

**Expected behaviour**
Memory should be allocated or error should be raised.

**Context**
```bash
(gdb) run
Starting program: /home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_memory_seg.Test_1MB_alloc_f1612f5755ee3bc71afb17ee200c1d381d277e09/ghdl/tb_memory_seg-a 
               0 fs - runner               -    INFO - Test case: Test 1MB alloc

Program received signal SIGSEGV, Segmentation fault.
0x00000000004b2d3b in vunit_lib.integer_vector_ptr_pkg.integer_vector_ptr_storage_t.resize ()
(gdb) where
#0  0x00000000004b2d3b in vunit_lib.integer_vector_ptr_pkg.integer_vector_ptr_storage_t.resize ()
#1  0x00000000004b3915 in vunit_lib.integer_vector_ptr_pkg.resize ()
#2  0x000000000053d9a1 in vunit_lib.memory_pkg.allocate ()
#3  0x00000000005443d6 in vunit_lib__tb_memory_seg__ARCH__a__main( const &&&*) ()
#4  0x000000000055f769 in grt.processes.run_processes (postponed=false) at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:792
#5  0x000000000055fc9a in grt.processes.simulation_cycle () at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:986
#6  0x000000000055fff1 in grt.processes.simulation_main_loop () at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:1128
#7  0x0000000000560057 in grt.processes.simulation () at /home/slawek/git/github/ghdl/src/grt/grt-processes.adb:1168
#8  0x000000000056a2b0 in __ghdl_run_through_longjump (func=0x560042 <grt.processes.simulation>) at /home/slawek/git/github/ghdl/build/../src/grt/config/jumps.c:232
#9  0x0000000000569f76 in grt.main.run_simul () at /home/slawek/git/github/ghdl/src/grt/grt-main.adb:183
#10 0x000000000056a03b in grt.main.run () at /home/slawek/git/github/ghdl/src/grt/grt-main.adb:216
#11 0x000000000057d180 in ghdl_main (argc=1, argv=0x7fffffffdb18) at /home/slawek/git/github/ghdl/src/grt/ghdl_main.adb:59
#12 0x000000000056a5a0 in main (argc=1, argv=0x7fffffffdb18) at /home/slawek/git/github/ghdl/build/../src/grt/main.adb:31
```

- OS: Fedora 27
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 0007498

**Additional context**
The problems occurs inside Vunit library and it is not easy to extract relevant functions into single file MWE. Currently I created minimal reproducer within Vunit library at my repository.

**How to reproduce?**
```bash
git clone -b allocate-segfault https://github.com/slaweksiluk/vunit.git
cd vunit
git submodule update --init --recursive
cd vunit/vhdl/verification_components/
PYTHONPATH=../../../ python run.py "*tb_memory_seg*1MB*" --log-level debug
```
Vunit wraps ghdl executable output and prints return code only:
```
  DEBUG - Process with pid=27530 terminated with code=255
  DEBUG - Process with pid=27530 terminated with code=255
```
It is possible to run standalone executable (it works without arguments):
```
$cd vunit_out/test_output/vunit_lib.tb_memory_seg.Test_1MB_alloc_f1612f5755ee3bc71afb17ee200c1d381d277e09/ghdl
$./tb_memory_seg-a 
               0 fs - runner               -    INFO - Test case: Test 1MB alloc
Segmentation fault (core dumped)
```
Please let me know if MWE is essential or some other info is needed.

**Checklist**
- [x] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version

tgingold, 2018-06-21:
I can reproduce the crash. Investigating...

qarlosalberto, 2018-10-11:
Any news with this? The same test for Modelsim works but for GHDL it does not :( :(

tgingold, 2018-10-11:
I am currently changing the implementation of records in the aim of unbounded vhdl 2008 records...

slaweksiluk, 2018-10-11:
As a possible workaround maybe it is worth to try it on llvm or mcode backend?

qarlosalberto, 2018-10-15:
@slaweksiluk It is difficult for us to change to LLVM or mcode because we would lose the code coverage :(

tgingold, 2018-11-12:
Due to the use of a large array...

tgingold, 2018-12-08:
So the solution is to increase the size of the stack (using ulimit on Linux).
Large allocations are now detected, so the crash doesn't occur anymore.

closed 2018-12-08 by tgingold

1138-4EB, 2019-05-20:
> Large allocations are now detected, so the crash doesn't occur anymore.

It is still reproducible with:

``` bash
#!/bin/sh

$(command -v winpty) docker run --rm -it ghdl/ext:vunit bash -c "\
apt update -y && apt install -y git && \
git clone -b allocate-segfault https://github.com/slaweksiluk/vunit.git && \
cd vunit && \
git submodule update --init --recursive && \
cd vunit/vhdl/verification_components/ && \
python3 run.py "*tb_memory_seg*1MB*" --log-level debug"
```

But the exit code is -11 instead of 255:

```
Starting vunit_lib.tb_memory_seg.Test 1MB alloc
Output file: vunit_out/test_output/vunit_lib.tb_memory_seg.Test_1MB_alloc_f1612f5755ee3bc71afb17ee200c1d381d277e09/output.txt
  DEBUG - Started process with pid=720: '/usr/local/bin/ghdl --elab-run --std=08 --work=vunit_lib --workdir=/vunit/vunit/vhdl/verification_components/
vunit_out/ghdl/libraries/vunit_lib -P/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/vunit_lib -P/vunit/vunit/vhdl/verification_com
ponents/vunit_out/ghdl/libraries/osvvm tb_memory_seg a -grunner_cfg=active python runner : true,enabled_test_cases : Test 1MB alloc,output path : /vun
it/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_memory_seg.Test_1MB_alloc_f1612f5755ee3bc71afb17ee200c1d381d277e09/,tb path :
 /vunit/vunit/vhdl/verification_components/test/,use_color : true --assert-level=error'
  DEBUG - Process with pid=720 terminated with code=-11
fail (P=0 S=0 F=1 T=1) vunit_lib.tb_memory_seg.Test 1MB alloc (0.3 seconds)

  DEBUG - TestRunner: Leaving
==== Summary ==================================================
fail vunit_lib.tb_memory_seg.Test 1MB alloc (0.3 seconds)
===============================================================
pass 0 of 1
fail 1 of 1
===============================================================
Total time was 0.3 seconds
Elapsed time was 0.3 seconds
===============================================================
Some failed!
```