ghdl/ghdl #612: Missing Version Number in Windows Build of GHDL

Paebbels, 2018-06-21:
**Description**
The version number is not compiled into the ghdl.exe binary. It shows some configure template `@VER@`.

**Expected behaviour**
Display the correct version number

**Context**
```
C:\git\GitHub\Paebbels\ghdl [paebbels/master ≡ +1 ~0 -0 !]> ghdl.exe --version
GHDL @VER@ (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

- OS: Windows
- Origin: Built from sources.  
  Commit SHA: c00074985783c13dce54de3814c39e76cf269268

**Additional context**
I think something was changed in the configure and/or make files, but the Windows Build scripts haven't been adapted.

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [x] Try the latest version

Paebbels, 2018-11-19:
The regexp in `targets.psm1` is broken.

It looks like a global search and replace for `sed` or a similar command broke two lines in the PowerShell scripts:

```powershell
$FileContent = $FileContent -Replace "\s@VER@\s", $GHDLVersion
$FileContent = $FileContent -Replace "\s\(tarball\)\s", " (commit: $GitCommitDataString;  git branch: $GitBranchName';  hash: $GitCommitHash) "
```

This will be fixed when MR #693 gets merged including commit 3254746a.

dalex78, 2026-08-17:
# Issue #612 — Missing version number in Windows build of GHDL

Analysis by an AI agent. Recommend merging the fix and closing.

## High-level summary

`ghdl.exe --version` on a Windows build showed the literal, unsubstituted template placeholder `@VER@` instead of the actual version string (e.g. `GHDL @VER@ (tarball) [Dunoon edition]`).

## What is actually wrong in GHDL's implementation

`src/version.in` is the template for `src/version.ads`:

```ada
   Ghdl_Ver     : constant String := "@VER@";
```

On Unix, `Makefile.in`'s `version.tmp` rule does the substitution with `sed -e "s#@VER@#$(ghdl_version)#"` — a plain literal-token replace, so it matches `"@VER@"` regardless of what surrounds it.

The (now-orphaned) native-Windows equivalent, `Invoke-PatchVersionFile` in `scripts/windows/targets.psm1`, instead used:

```powershell
$FileContent = $FileContent -Replace "\s@VER@\s", $GHDLVersion
```

`\s` matches whitespace, but `@VER@` in `version.in` is immediately surrounded by double quotes (`"@VER@"`), not whitespace. The regex can therefore never match anything in the actual template file, the `-Replace` is a silent no-op, and `@VER@` is written out to `version.ads` unchanged — exactly the symptom reported.

Note `Invoke-PatchVersionFile` also never substitutes `@DESC@`/`@REF@`/`@HASH@` at all (no corresponding `-Replace` calls exist for them), so those placeholders have the same "never substituted" problem. That is out of scope for this ticket, which is specifically about `@VER@`/the version number, but is worth a maintainer's attention if this code path is ever revived.

## What the fix does

`scripts/windows/targets.psm1`: change the regex from `"\s@VER@\s"` to plain `"@VER@"`, mirroring the Makefile's literal-token `sed` substitution so it actually matches the quoted placeholder in `version.in`.

## Caveat: this code path is currently unreachable and untested here

While investigating, `scripts/windows/targets.psm1` (and its `Invoke-PatchVersionFile` function) turned out to have **no caller anywhere in the current repository** (`grep -rl Import-Module scripts/windows` finds nothing) — the native PowerShell+MSVC Windows build it belongs to was replaced by an MSYS2-based build (see `Build-MSYS2.yml`), which drives the ordinary `configure`/`make` flow and therefore already gets `@VER@` substitution right via the ordinary `sed`-based Makefile rule.

This sandbox also has no `pwsh`/`powershell` interpreter available, so this fix could not be executed against an MWE here. The fix was verified by inspection only (the corrected regex is a plain literal-token match, directly mirroring the working Makefile `sed` rule for the same substitution), not by running the script. No `testsuite/` entry was added, since there is nothing in this repository that currently invokes this code path to test.

closed 2026-08-17 by tgingold