ghdl/ghdl #613: Large Time Values

CogPy, 2018-06-22:
**Description**

A time above approximately 9222 seconds leads to a runtime error.

**Expected behaviour**

Accept any time given.

**Context**

```
ent.vhd:5:47:warning: physical literal out of range [-Wruntime-error]
```

**Additional context**

A description of the problem is given here: 
https://www.aldec.com/en/support/resources/documentation/articles/1165/print_page

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> ent.vhd
entity ent is
end entity;

architecture a of ent is
  constant SimulationTime_c  : time    := 10000 sec;
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

Note that `run.sh` is used to execute `sim.sh` inside a docker container. Please, put your commands in `sim.sh` and just copy `run.sh` from the example. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version (if the docker container above is the latest version)

go2sh, 2018-06-22:
All phsyical types in vhdl store the value in the primary unit. For `TIME` this is fs (femto seconds). The LRM allows to select a secondary unit with a reduced accuracy (e.g. ns) to constrain the precision of the `TIME` type, based on your design or in general. The backing type of physical types should be at least a 32bit signed integer. With this, there is definitely a limit to values you can use with `time`. Although 9222 seems a little odd to me.

Paebbels, 2018-06-22:
GHDL is using a 64 bit integer to store `TIME` values. With a primary unit of `1 fs`, a 64 bit (signed) value can store circa 2.6 hours. Your 9222 seconds (divided by 3600 sec per hour) is circa `2.6 hr`.

There is currently no simulator (free or commercial) that supports more than 64 bit for type `TIME`.

With VHDL-2018, the minimum number of bits was raised to 64, because some synthesis still implement `TIME` with 32 bit, what is causing lots of problems.

You can initialize your constant with `time'high` to the greatest possible value: (2^63-1).

----------------------------
Why do you simulate in realtime?

CogPy, 2018-06-22:
I was converting a set of testbenches from Questa 10.6c to GHDL to improve regression test speeds. Hence this runtime error came up. I don't know if it's a feature or a bug; hence I haven't reported it as a bug.

Aforementioned simulator happily runs the testbench; I don't know what will happen if it actually simulated more than those ~5 hrs.

tgingold, 2018-06-23:
One possible fix is to allow to set the time precision to ps.

Paebbels, 2018-06-24:
@CogPy  
Can you check if `10000 sec` is a positive time in QuestaSim or if the literal is overflowing to a negative time?

Is the simulation time (accessible by `now`) really increasing to hours? Or are you talking about simulation execution time? A simulation exceeding such a simulation time frame would need to run >1 week on a good equipped computer.

go2sh, 2018-06-25:
FYI Modelsim sets the base time to 1ns if possible, depending on the model.

CogPy, 2018-06-25:
@Paebbels it's shown in the waveform as the expected value. 

The exception occurs after around 115 ms of simulation time.

tgingold, 2018-06-27:
@go2sh: that's clever!

tgingold, 2018-06-28:
Which backend are you using ?  Maybe the time resolution could be set at runtime if mcode is used.

CogPy, 2018-06-28:
I'm using a mcode backend. How can I set the resolution? The only mention
I've found doesn't work.

tgingold, 2018-07-01:
Currently this is not possible; but it could be implemented with mcode.  For gcc/llvm back-ends, that would be harder due to separate compilation.

tgingold, 2018-08-09:
I will soon commit a patch to add this feature.  Eg: to set time resolution to 1 sec:

ghdl -r --time-resolution=sec ent

(Only with mcode backend)