ghdl/ghdl #614: CONSTRAINT_ERROR : grt-processes.adb:1051

CogPy, 2018-06-22:
**Description**

Simulation fails.

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-220-ga3524ea3) [Dunoon edition]
Compiled with GNAT Version: 8.0.1 20180414 (experimental)
Target: x86_64-linux-gnu
In directory: /tmp/tmp.A894oiqdjx/
Command line:
ghdl -r --ieee=synopsys --workdir=/tmp/tmp.A894oiqdjx tb_xxxx
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-processes.adb:1051 overflow check failed
Call stack traceback locations:
0x5613ee8a6980 0x5613ee8a6c29 0x5613ee8a6cc2 0x5613ee83e042 0x5613ee8e4ee7 0x5613ee8e4fb1 0x5613eeb23db6 0x5613eeb0fc25 0x5613eeb007ff 0x5613eeb2dc38 0x5613ee83dc31 0x7fce974bbb95 0x5613ee83c908 0xfffffffffffffffe
******************************************************************
```

**Expected behaviour**
Not to fail.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:

- OS: Ubuntu 18.4
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: v0.35-220-ga3524ea3

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**

Sources not available.

tgingold, 2018-06-23:
So should we use a 64-bit integer for Nbr_Cycles ?

CogPy, 2018-06-23:
It's completely unrelated to the 5 hrs mentioned in #613, as this is an abort criterion that should never be reached.

Simulation runs for about 200' on a Xeon 2.5 GHz before it crashes (details off of my head, it's weekend after all).

@tgingold: preferably something along the line of Big_Numbers or wrap the counter if it gets too big. Depending on performance.

diogratia, 2018-06-23:
So I modified the program after letting it run for almost 6 hours.

```vhdl
entity nbr_cycles_test_min is
end entity;

architecture foo of nbr_cycles_test_min is
    constant start_time:  time := TIME'HIGH - integer'high *  fs - 100 fs;
begin
    process
    begin
        if now < start_time then
            wait for start_time;
        elsif now < time'high  then
            wait for 1 fs;
        else
            wait;
        end if;
    end process;
end  architecture;
```

I was suspicious the simulation wasn't ending with an exception so I set it up to end not longer after I'd expect Nbr_Cycles to overflow.       

Letting the new one run gave:
```
Macbook: time ghdl -r nbr_cycles_test_min --stats

total:          wall: 20381.770  user: 11940.540  sys: 8070.550
 elab:          wall:   0.000  user:   0.000  sys:   0.000
 internal elab: wall:   0.000  user:   0.000  sys:   0.000
 cycle (sum):   wall: 20381.770  user: 11940.540  sys: 8070.550
  processes:    wall: 5226.910  user: 3091.350  sys: 2011.310
  resume:       wall: 5129.950  user: 3021.630  sys: 2016.690
  update:       wall: 4974.770  user: 2862.040  sys: 2014.210
  next compute: wall: 5050.140  user: 2965.520  sys: 2028.340
Number of simple signals: 0
Number of signals with projected wave: 0
Number of non-driven simple signals: 0
Number of resolved simple signals: 0
Number of multi-sourced signals: 0
Number of signals whose activity is managed: 0
Number of drivers: 0
Number of direct drivers: 0
Number of signals per mode:
  B1:   0
  E8:   0
  E32:  0
  I32:  0
  I64:  0
  F64:  0

Propagation table length: 1
Propagation table count:
  end : 1
Number of delta cycles: 1
Number of non-delta cycles: -2147483549
Nbr of events: 0
Nbr of active: 0
Number of processes: 1
Number of sensitized processes: 0
Number of resumed processes: 2147483749
Average number of resumed processes per cycle: 
11940.544u 8070.553s 5:39:42.51 98.1%	0+0k 0+0io 3pf+0w
Macbook:
```

You can see an llvm version on MacOS doesn't get an exception for an overflow. 

The number of non-delta cycles demonstrates the uselessness of gathering statistics in holders that roll-over.

It couldn't hurt to make Nbr_Cycles a  64  bit integer, but then again the source the original poster was simulating would tell us if there's a practical issue or it's blue sky as in these code examples.

(I'd fix it anyway.)

CogPy, 2018-06-25:
This exception for my code occurs after around 115 ms of simulation time. 

Is this counter counting delta-cycles?

CogPy, 2018-06-26:
This rather two-handed approach to this bug let me run the tests succesfully.

```diff
diff --git a/src/grt/grt-processes.adb b/src/grt/grt-processes.adb
index e4248f0a..cf1ca7a2 100644
--- a/src/grt/grt-processes.adb
+++ b/src/grt/grt-processes.adb
@@ -1048,7 +1048,9 @@ package body Grt.Processes is
          Current_Delta := 0;

          --  Statistics.
-         Nbr_Cycles := Nbr_Cycles + 1;
+         if Nbr_Cycles < 1073741824 then
+            Nbr_Cycles := Nbr_Cycles + 1;
+         end if;

          --  For wave dumpers.
          Grt.Hooks.Call_Cycle_Hooks;
```

closed 2018-06-28 by tgingold

tgingold, 2018-06-28:
Why not, but doesn't really matter, as a 63-bit type is very very large.

commit 657fcfde (2018-06-28): grt: use 64-bit type for Nbr_Cycles.

Fix #614