ghdl/ghdl #616: Using a function to allocate a std_logic_vector slows down execution exponentially

jhizzle, 2018-06-29:
**Is your feature request related to a problem? Please describe.**

I've got a design that slow down considerably, and I believe exponentially, for certain code. I believe slow down is somehow related to to calling functions to allocate std_logic_vectors, but I'm not certain. I've provided a MWE below, what's really strange is that running the design is very slow, but uncommenting the variable assignment on line 35 actually speeds up the execution. This is not what I would expect. I was wondering if it was somehow related to memory allocation/lack of deallocation, or somehow related to issue #49. 

Comparison of simulation time vs time to simulate:

Simulation Run Time | Time to Simulate (Wall Clock Time)
---------------------|------------------------------------
10 us | 0.21 s
30 us | 1.77 s
50 us | 5.07 s

Here's `mwe.vhd`.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity mwe is
  generic (
    wait_us : integer := 40;
    clk_period : time := 10 ns
  );
end entity;

architecture sim of mwe is

  signal clk : std_logic := '0';
  signal runsim : boolean := true;

  function slv_all(constant width : in integer; constant value : in std_logic) return std_logic_vector is
    variable slv_v : std_logic_vector(width - 1 downto 0) := (others => value);
  begin
    return slv_v;
  end function;

  function slv_ones(constant width : in integer) return std_logic_vector is
  begin
    return slv_all(width, '1');
  end function;

begin

  p_clk: process
  begin
    while runsim loop
      clk <= '0';
      wait for clk_period / 2;
      clk <= '1';
      wait for clk_period / 2;
    end loop;
    wait;
  end process;

  p_check_requests: process

    function return_true return boolean is
      constant ones_c : std_logic_vector(31 downto 0) := slv_ones(32);
    begin
      return true;
    end function;

    variable ones_v : std_logic_vector(31 downto 0);
    variable result_v : boolean;
  begin
    wait until rising_edge(clk);

    while runsim loop
      wait until rising_edge(clk);

      result_v := return_true;

      -- uncommenting the following lines speeds up the design
      -- ones_v := slv_ones(32);

    end loop;

    wait;
  end process;

  p_main: process
  begin
    wait for wait_us * 1 us;
    runsim <= false;
    wait;
  end process;

end architecture;
```

**Describe the solution you'd like**
Faster, more consistent performance.

**Describe alternatives you've considered**
Uncommenting the `ones_v := slv_ones(32);` line.

**Additional context**
Tested ghdl gcc 0.35-dev (tarball) on Red Hat 7.5.
Tested ghdl llvm 0.34 (tarball), mcode 0.35 (from source) on Mac OSX.

```
#>> ghdl -a mwe.vhd
#>> ghdl -e mwe
#>> ghdl -r mwe
```

diogratia, 2018-06-29:
Using a fairly recent version of ghdl (GHDL 0.36-dev (v0.35-168-g252a916)) on MacOS (10.11.6 El Capitan) with an llvm backend and a longer `wait_us` (ghdl -r mwe -gwait_us=100, - substitutes numeric part of physical literal) shows that the memory utilization of ghdl steadily climbs while the simulation runs.

 (Watched with Activity Monitor.app, the increased run time allows for fumbling around to select the executing process and look at memory utilization.)

This bug would likely be due to how the expression stack is popped. The expression stack would nominally be cleared on statement boundaries.  Constant ones_c isn't utilized which may have something to do with it, preventing the expression stack from being popped (which the following assignment to ones_v would do).

The memory leak can also be eliminated by commenting out the constant ones_c declaration in function return_true. A procedure call would also not cause a leak (I tested it).

I imagine Tristan will appreciate your test case.

faf5678, 2018-07-22:
@tgingold 

i just stumbled onto this bug in my own code. I chased the problem through my code independently from OPs MWE and came to a more concise version i think. well it's likely that OP and me reference a related if not the same bug. isim has no problem with this and thus is several orders of magnitude faster for some kinds of code. (blink of an eye vs. coffee break).

```
GHDL 0.36-dev (v0.35-229-g18116ef1) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 llvm code generator
```

```
library ieee;
use ieee.std_logic_1164.all;

entity top_ent is end entity;

architecture default of top_ent is
	signal clk : std_logic := '0';

	function preproc(v : std_logic_vector) return std_logic_vector is
	begin
		return v;
	end function;
	
	function my_xor_reduce(v : std_logic_vector) return std_logic is
		variable ret : std_logic;

		-- this version is super quick
		--variable vv : std_logic_vector(v'range) := v;

		-- this version is super slow
		variable vv : std_logic_vector(v'range) := preproc(v);
	begin
		for i in vv'range loop
			ret := ret xor vv(i);
		end loop;
		return ret;
	end function;
	
begin
	clk <= not clk after 500 ps;
	process(clk)
		variable s : std_logic_vector(0 to 63) := (others => '0');
		variable out_accum : std_logic := '0';
	begin
		if rising_edge(clk) then
			for i in 0 to 2047 loop
				out_accum := out_accum xor my_xor_reduce(s);
			end loop;
			report "ticktock";
		end if;
	end process;
end architecture;
```

commit 63906cb1 (2018-08-07): Release secondary stack within elab_declaration_chain.  Fix #616
closed 2018-08-07 by tgingold