ghdl/ghdl #618: Undocumented deviation from the standard

vicencb, 2018-06-29:
Hi,
from (Vhdl-02, 7.3.5) or (Vhdl-08, 9.3.6) Type conversions, last paragraph:
```
An implicit conversion ... can only be applied if the operand is either
 a numeric literal or
 an attribute, or
 ... the division of ... a physical type ...
```
given that statement it looks like this would be invalid:
```vhdl
entity e is end entity;
architecture a of e is
  type t is range 0 to 1;
  constant c :t := 7 - 6;
begin
  assert c = 3 report "c /= 3" severity note;
end architecture;
```
but ghdl accepts it with no warning or error.

Making the type conversions explicit, the standard way would be
```vhdl
constant c :t := t(7) - t(6);
assert c = t(3) report "c /= 3" severity note;
```
which should generate out-of-bounds errors.

Am i missing something? If so, what?

Otherwise, if my reasoning is correct, then this is not a bug report, because it looks like a feature.
But this deviation from the standard could be documented somewhere, like e.g.:
https://ghdl.readthedocs.io/en/latest/references/ImplementationOfVHDL.html
Also, should this feature be controlled with `-frelaxed-rules` or similar flag?

Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2018-06-29:
For `constant c :t := 7 - 6;`.

The integer literals like `7` and `6` are always of type *universal_integer*. The used minus operation is `"-"(universal_integer; universal_integer) return universal_integer`. This is also called a `universal expression`. The expression result (of type *universal_integer*) is now implicitly converted to the integer type `t`. The range of *universal_integer* is only limited by the implementation. The range if only checked when the value gets assigned (so after conversion).

I think that `9.3.6` is not correct, because there are more operation that create universal results. Division of physical types is one possibility. See e.g. 9.5

When you search for *universal*:
- 5.1 Types - General
  > [...] or an implicit type conversion of a value of type universal_integer or universal_real to the corresponding value of another numeric type
- 5.2.3.1 Integer types - General  
   > Integer literals are the literals of an anonymous predefined type that is called universal_integer in this standard. Other integer types have no literals. However, for each integer type there exists an implicit conversion that converts a value of type universal_integer into the corresponding value (if any) of the integer type (see 9.3.6).
- See 9.2.5 Adding operators / 9.2.7 Multiplying operators

tgingold, 2018-07-01:
Indeed no error message is required here, however there could be a warning for expression like 'c = 3' which is never true.

vicencb, 2018-07-01:
Hi Paebbels, Tristan,
> The expression result (of type _universal_integer_) is now implicitly converted to the integer type `t`.

please, can you tell me where in the standard this implicit conversion is allowed. I do not see it in:
1. `5.1 Types - General` states that an implicit conversion is a basic operation.
1. `5.2.3.1 Integer types - General` states that there exists implicit conversions and refers to `9.3.6`.
1. `9.2.5 Adding operators` Implicit conversions are not mentioned.

None of these points mention where an implicit conversion is allowed.
`9.3.6` is the only point that i see that says where implicit conversions are allowed and an _expression_ is not listed. Just the opposite, it states that the aforementioned list (literal, attribute or physical division) are the only allowed ones.

Regards,
&nbsp;&nbsp;Vicenç.

tgingold, 2018-07-01:
See the last paragraph of lrm 2008 9.3.6 Types conversion

vicencb, 2018-07-01:
Just for reference, Vhdl-08 LRM, 9.3.6, Type conversions, last paragraph:
```
In certain cases, an implicit type conversion will be performed.
An implicit conversion
of an operand of type universal_integer to another integer type, or
of an operand of type universal_real to another floating-point type,
can only be applied if the operand is either
a numeric literal or
an attribute, or
if the operand is an expression consisting of the division of a value of a physical type by a value of the same type;
such an operand is called a convertible universal operand.
An implicit conversion of a convertible universal operand is applied if and only if
the innermost complete context determines a unique (numeric) target type for the implicit conversion, and
there is no legal interpretation of this context without this conversion.
```

Paebbels, 2018-07-02:
@vicencb I tried to explain that VHDL knows a lot more then just one operation (division of two physical types) that generates a new *universal* type.

The last paragraph should be:
> [...] if the operand is a universal expression [...]

*universal expression* includes
* identity of a value of any integer type
* negation of a value of any integer type
* addition of a value of any integer type and a value of the same type
* subtraction of a value of any integer type and a value of the same type
* multiplication of a value of any integer type and a value of the same type
* division of a value of any integer type and a value of the same type
* **division of a value of a physical type by a value of the same type**
* modulus of a value of any integer type and a value of the same type
* remainder of a value of any integer type and a value of the same type
* absolute value of a value of any integer type
* exponentiation of a value of any integer type and a value of type integer

vicencb, 2018-07-02:
Hi Paebbels,

> The last paragraph should be:

The purpose of the Vhdl-08 LRM, like any other standard document, is to be the reference for all conformant implementations in order to allow interoperability.
If each implementation re-writes paragraphs at will, the purpose is fully defeated.

Likewise, your list of expressions resulting in a universal type is nowhere to be seen in the standard (except for the one in bold).
5.2.3.1 and 5.2.5.1 state the existence of conversions from universal_integer to each integer type and from universal_real to each floatint-point type, **but not viceversa**.

The aim of the Vhdl standard is to define a Strongly-typed programming language.
If your implicit conversions were allowed, then you could assign any integer type to any other integer type, which is against strong-typing:
```vhdl
entity e is
  type tp1 is range 0 to 1;
  type tp2 is range 0 to 1;
  constant c1 :tp1 := 0;
  constant c2 :tp2 := c1;
  -- c1 of integer type tp1 is implicitly converted to universal_integer.
  -- First point in your list allows it.
  -- The universal_integer result is implicitly converted to integer type tp2.
  -- Your first post, 4th phrase, allows it.
end entity;
```
And this is, IMHO and by my interpretation** of the standard, correctly handled by ghdl:
```
ghdl -a e.vhd 
e.vhd:5:23:error: can't match 'c1' with type tp2
ghdl:error: compilation error
```

Regards,
&nbsp;&nbsp;Vicenç.

P.D.: If the standard is well written, there is only one interpretation.

tgingold, 2018-07-02:
> @vicencb I tried to explain that VHDL knows a lot more then just one operation (division of two physical types) that generates a new universal type.

That's true.

> The last paragraph should be:
> 
> [...] if the operand is a universal expression [...]

No, as it would be ambiguous.

Paebbels, 2018-07-04:
@vicencb  
*universal_integer* as handled as any other type in the LRM. Thus when the LRM writes:

`"+"(anonymous : type; anonymous : same_type) return same_type;`  
then it also applies to *universal_integer* and also to *universal_real*. This is how I created the long list of operations creating *universal_integer* results.

vicencb, 2018-07-04:
Hi Patrick,
agreed, the LRM has this general rule:
```vhdl
"-"(anonymous :any numeric type ; anonymous :same_type        ) return same_type        ;
```
which i also agreee it implies this particular one:
```vhdl
"-"(anonymous :universal_integer; anonymous :universal_integer) return universal_integer;
```
But **not this** from your list:
```vhdl
"-"(anonymous :any integer type ; anonymous :same_type        ) return universal_integer;
```
That comes fom 5.2.3.1, 7th paragraph and 9.2 which define several operators from the universal_integer type to the same universal_integer type.

&nbsp;

I still think that `constant c :t := 7 - 6;` from the initial post is against the LRM because neither option is allowed:
 1. `"-"[universal_integer,universal_integer return universal_integer](7, 6)` can not be implicitly converted. It is not in the list of **only** allowed implicit conversions in `9.3.6`.
 2. in `"-"[t,t return t](7, 6)` the 7 and 6 are out of range when implicitly converted to type `t`.
 3. `"-"[universal_integer,universal_integer return t]` does not exist.

Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2018-07-04:
In my eyes, LRM 9.3.6 is incomplete. It needs to state more operations then division of the same pyhsical types, otherwise the language can not work.

-------------------
Notification list: @NJDFan, @lievenlemiengre, @jimlewis

tgingold, 2018-07-05:
> I still think that constant c :t := 7 - 6; from the initial post is against the LRM because neither option is allowed:

No, it isn't:

> 	• in "-"[t,t return t](7, 6) the 7 and 6 are out of range when implicitly converted to type t.

No, they aren't. Ranges are only checked for the final value.
When you declare:

  type t is range 0 to 1;

this is roughly equivalent to (see 5.2.3):

  type t'base is range -(2**31) to 2*31 - 1; -- a 32 bit integer.
  subtype t is t'base range 0 to 1;

So 7 and 6 are valid as t'base values, hence 7 - 6 is a valid t expression.

vicencb, 2018-07-05:
Hello Tristan,
thank you for this helpful post. It finally answers my very first questions:
> Am i missing something? If so, what?

So, making the overloading and implicit-conversions explicit:
```vhdl
constant c :t := "-"[t'base, t'base return t'base](t'base(7), t'base(6));
```
and there is no conversion needed at all from `t'base` to `t`.

The LRM level of formality makes it very concise and unambiguous, but, at the same time makes it difficult to read and understand.

In the end, ghdl is standard compliant in that regard, so, closing the issue.

Thank you all,
&nbsp;&nbsp;Vicenç.

closed 2018-07-05 by vicencb

tgingold, 2018-07-05:
> So, making the overloading and implicit-conversions explicit:
> 
> constant c :t := "-"[t'base, t'base return t'base](t'base(7), t'base(6));
> and there is no conversion needed at all from t'base to t.

That's correct.  There is however a range check for the initialization.

> The LRM level of formality makes it very concise and unambiguous, but, at the same time makes it difficult to read and understand.

This is a delicate topic.  The language is strongly typed but still allows to use numeric literals for any numeric type.
Another delicate topic is visibility/overloading/overload resolution.

You don't really need to understand these delicate topics to use VHDL as the resulting behaviour is intuitive.