ghdl/ghdl #620: crash passing generic package function to generic package

TripRichert, 2018-07-04:
**Description**
ghdl is crashing when I instantiate two generic packages, passing a type and function from one as generics into another and using that function in a procedure.

**Context**

> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: 0.35 (v0.35) [Dunoon edition]
> Compiled with GNAT Version: 7.2.0
> Target: x86_64-linux-gnu
> In directory: /home/trip/Documents/projects/vhdl/nest_pkg/
> Command line:
> ghdl -e --std=08 test_tb
> Exception CONSTRAINT_ERROR raised
> Exception information:
> raised CONSTRAINT_ERROR : trans-chap8.adb:3051 access check failed
> Call stack traceback locations:
> 0x55c59505b783 0x55c595094d66 0x55c595062583 0x55c595062b77 0x55c595064248 0x55c5950643b3 0x55c5950cb2d9 0x55c595081982 0x55c595081aae 0x55c59506b70e 0x55c5950d30a3 0x55c595044fce 0x55c595042e77 0x55c59504fc79 0x55c5950fba90 0x55c5950ec369 0x55c5950dbd5f 0x55c595107ae4 0x55c594e1d66f 0x7f40af2d91bf 0x55c594e1c498 0xfffffffffffffffe
> ******************************************************************
> 

- OS: Ubuntu 17.10
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: ecd48092b465229ef36c4a00ac0df3f2ae6672e0

**Additional context**
I haven't used issue runner before. Tried to base my post on others. Hope this works.

**How to reproduce?**

```
#>> sim.sh
ghdl -a --std=08 nest_pkg/type_declaration_pkg.vhd
ghdl -a --std=08 nest_pkg/type_user_pkg.vhd
ghdl -a --std=08 nest_pkg/test_tb.vhd
ghdl -e --std=08 test_tb

#>> run.sh
docker run --rm -tv /$(pwd):/src -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Files**
[nest_pkg.zip](https://github.com/ghdl/ghdl/files/2164026/nest_pkg.zip)

**Checklist**
Before submitting your issue, please review the following checklist:

- [X] Add `GHDL Bug occurred` log block
- [X] Add a MWE
- [X] Try the latest version

1138-4EB, 2018-07-04:
> I haven't used issue runner before. Tried to base my post on others. Hope this works.

It is ok. It works as expected:

``` bash
$ curl -L https://raw.githubusercontent.com/1138-4EB/issue-runner/master/runner.sh | sh -s ghdl/ghdl#620
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   980  100   980    0     0   4375      0 --:--:-- --:--:-- --:--:--  4355
mkdir: created directory 'run-ghdl--ghdl--620'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   147  100   147    0     0    312      0 --:--:-- --:--:-- --:--:--   312
100  2206  100  2206    0     0   3446      0 --:--:-- --:--:-- --:--:--  3446
cmd: /usr/bin/python3 extract_sources.py ghdl/ghdl#620
Get sim.sh
Get run.sh
Get [nest_pkg.zip](https://github.com/ghdl/ghdl/files/2164026/nest_pkg.zip)
Extract nest_pkg.zip

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-227-gecd48092) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
ghdl -e --std=08 test_tb
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap8.adb:3051 access check failed
Call stack traceback locations:
0x55809a8606fb 0x55809a89a9a0 0x55809a867505 0x55809a867af9 0x55809a8691ca 0x55809a869335 0x55809a8d326e 0x55809a8871da 0x55809a887306 0x55809a870b8c 0x55809a8db093 0x55809a849a15 0x55809a846f85 0x55809a8548a3 0x55809a903fd4 0x55809a8f4acb 0x55809a8e3d7f 0x55809a90ffd2 0x55809a61f5c3 0x7f834dd62a85 0x55809a61e3e8 0xfffffffffffffffe
******************************************************************
```

However, I wonder why your bug report shows `0.35 (v0.35)` instead of `0.36-dev (v0.35-227-gecd48092)`, since both versions are apparently built from the same commit. Could you please tell us how did you build and install GHDL? I just want to ensure that the version you used is the same that you compiled, and not a different pre-existing one. Although it seems not to be relevant for the underlying bug in this issue.

On the issue itself, I think that this might be somehow expected because AFAIK generic types in packages and entities are not fully supported, yet. However, let's wait for @tgingold to confirm this.

TripRichert, 2018-07-05:
I found an easy workaround. Writing a function as a wrapper around the generic package function (and passing the wrapper function as the generic parameter) prevents the crash and runs as expected.

tgingold, 2018-07-25:
I can reproduce the crash; I have to investigate.

commit 2134f8df (2018-08-06): sem_association_subprogram: handle instantiated subprograms in marking used.

Fix #620
closed 2018-08-06 by tgingold