ghdl/ghdl #621: crash on improper attribute access

faf5678, 2018-07-07:
ghdl 0.35 crashes with this:

entity crash_entity is end entity;
architecture default of crash_entity is
	attribute s : string;
	function func return boolean is
	begin
		return s(1);
	end function;
begin
end architecture;

Paebbels, 2018-07-07:
Please add the reported error message from GHDL.

faf5678, 2018-07-07:
i was assuming this would be reproducible easily, but ok here it is:

```
sem_parenthesis_name: cannot handle IIR_KIND_ATTRIBUTE_DECLARATION (error.vhd:4:18)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (v0.35) [Dunoon edition]
Compiled with GNAT Version: 4.6
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/home/user/ghdl/bin/ghdl1-llvm -P/home/user/ghdl/lib/ghdl/v93/std/ -P/home/user/ghdl/lib/ghdl/v93/ieee/ -c -o error.o error.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Call stack traceback locations:
0x49ebaa 0x61b349 0x61da1e 0x630e64 0x631f2e 0x60ca7c 0x60dd52 0x60dec5 0x63d649 0x638feb 0x60f0cb 0x639ac9 0x63f954 0x4eb0bb 0x601ac6 0x6024ef 0x48741b 0x44a0fa 0x7fe2411e2ec3 0x44990a 0xfffffffffffffffe
******************************************************************
```

faf5678, 2018-07-07:
also as a note, i only now see that there is a line number given in the error message.

i guess the only real problem here is that the error message is confusing. it would be more useful without the additional crash message, cause that's all you see in that moment, not the one line that is actually helpful.

Paebbels, 2018-07-07:
@faf5678 I moved your error message into the original issue text. (You can edit own issues at any time.)

Anyhow, GHDL should write a error message without crashing even if the VHDL code is invalid.

1138-4EB, 2018-07-07:
> i guess the only real problem here is that the error message is confusing. it would be more useful without the additional crash message, cause that's all you see in that moment, not the one line that is actually helpful.

The `GHDL Bug occured` is not meant for users to understand it, but to make it easier for developers to analise and hopefully solve the problem. Ideally, internal bugs should not occur, and GHDL should provide only the meaningful line. But, because this is almost a single man project, we need to be considerate.

tgingold, 2018-07-07:
I have a fix for it, will commit soon.

commit 714dd824 (2018-07-08): sem_names: indexed attribute declaration is an error.

Fix #621
closed 2018-07-08 by tgingold