ghdl/ghdl #622: Man Pages

SethGower, 2018-07-16:
**Describe the solution you'd like**
I realize that the help output is there (`ghdl [-h, --help]`), but man pages would be nice.

Paebbels, 2018-07-16:
The long help pages (manual) can be found here: http://ghdl.readthedocs.io/en/latest/

A PDF version is offered, too: https://media.readthedocs.org/pdf/ghdl/latest/ghdl.pdf

1138-4EB, 2018-07-16:
We might generate 'man pages' with Sphinx itself: http://www.sphinx-doc.org/en/master/man/sphinx-build.html#options So the output should be equivalent to the HTML and PDF versions. I have not tested it, tho.

SethGower, 2018-07-16:
That would be awesome. I just would prefer in terminal man pages, as
opposed to opening a browser and finding them.

Paebbels, 2018-07-16:
Now I remember that there was a `man` builder to generate groff files :).

How do we integrate the results into the shipped files?  
Should we compile to groff on Travis-CI and package the outputs into the tarballs and ZIPs?

1138-4EB, 2018-07-16:
@SethGower, the following snippet uses sphinx to produce a man page:

``` bash
git clone -b fix-doc https://github.com/1138-4EB/ghdl/
cd ghdl

docker run --rm -itv /$(pwd):/src -w //src/doc btdi/sphinx:py2-featured sh -c "\
  sphinx-build -T -b man . ./_build/man"

nroff -man _build/man/ghdl.1
```

The output is a single file: [ghdl.1](https://github.com/ghdl/ghdl/files/2197821/ghdl.1.txt) (I added the extension, `.txt`, for GitHub to let me upload it)

> How do we integrate the results into the shipped files?
> Should we compile to groff on Travis-CI and package the outputs into the tarballs and ZIPs?

Adding it to buildthedocs (BTD) is straighforward: just add a sphinx-build line to https://github.com/buildthedocs/btd/blob/master/btd/build.sh#L72-L76. However:

- Some additional steps are required in order to add it to the tarballs/zips. I think that it should be added as a separate icon/file. See bottom-left in https://1138-4eb.github.io/ghdl-io/doc/fix-doc/ @Paebbels, please either assign this issue to me, open an issue in buildthedocs/btd or add a item to buildthedocs/btd#2.
- GHDL is not officially migrated to Travis-CI (BTD), yet. We still use readthedocs.org for CI of the docs. I don't know whether the 'man' builder can be activated in RTD.org.
- I have not checked the correctness of the generated `ghdl.1` file. It seems to be acceptable, as most of the content is readable. But some better checking would be desirable.

Paebbels, 2018-07-16:
RTFD does not support the `man` backend.

@1138-4EB I was not talking about moving the whole documentation to Travis-CI, but just the intermediate compilation step for `groff` outputs so we can package the man-pages into the binary packages for delivery.

SethGower, 2018-07-16:
Thanks for the responses guys! Just wanted to say that I love `ghdl`. I use it for school on my own laptop instead of testing everything in ModelSim on the schools computers (although I have to demo on MS). Just wanted to say how much I appreciate all the devs!

1138-4EB, 2018-07-16:
> @Paebbels
> (...) just the intermediate compilation step for groff outputs, so we can package the man-pages into the binary packages for delivery

Sure. Where in the packages should we put `ghdl.1` to ensure that `man` will find it? `$prefix/man/ghdl.1`?

@SethGower, thanks for your kind words.

Paebbels, 2018-07-17:
I believe there is a central `man` directory where we have to put that file(s).

According to this [AskUbuntu](https://askubuntu.com/questions/244809/how-do-i-manually-install-a-man-page) answer, we have to copy the `*.1` file(s) to `/usr/local/share/man/man1/`. Afterwards, you have to run `sudo mandb` to update the database of man.

1138-4EB, 2019-01-04:
Well, I had a look at the integration of man pages generation in the current CI system, but it is not straighforward:

I tried using the 'stages' feature in Travis to build the man pages in a first stage, and then execute the builds and tests. Unfortunately, according to the docs (see [Caches and build matrices](https://docs.travis-ci.com/user/caching/#caches-and-build-matrices)), the cache is unique for each of the builds. As a result, it is not possible to share the man pages created in the first stage with the builds in the second.

A feasible workaround is to use dockerhub or any other external service (github, amazon...) to save the intermediate artifacts between stages. However, no credentials are configured in this repo:

- In ghdl/ghdl GitHub releases is set up, which is a very specific procedure.
- In ghdl/docker DockerHub credentials are set up.

Another workaround is to build man pages in a separate repo and trigger release commits in order: doc first, builds after the former is finished.

A last solution is to use GitHub's API to manipulate the tarballs after the release has been pushed to GitHub.

@tgingold, @Paebbels what do you think?

1138-4EB, 2019-01-05:
Do you mean to generate the same man pages as many times as different tarballs we create? That will add at least a minute to each of the runs. Plus, images required to build GHDL itself are not the same as the one containing Sphinx.

Rather than doing that, I would suggest adding them manually to the release, so that the users that want so can download it as an additional package/tarball.

Paebbels, 2019-01-05:
I Debian, man pages are separate packages, aren't they?

tgingold, 2019-01-06:
That would be fine: generate the man page and make it available.

PR #733 (2019-01-06): Add man pages job to travis
Close #622.

This PR adds a parallel job in the Travis matrix, to build man pages for GHDL. When a release is tagged, file `ghdl.1` is added to the release artifacts, so it is pushed to GitHub along with the tarballs for different distributions.

Even though the file is only pushed to GitHub when a release is tagged, the content can be seen in every Travis build. See e.g. https://travis-ci.org/1138-4EB/ghdl/builds/475993763 and https://travis-ci.org/1138-4EB/ghdl/jobs/475993771
closed 2019-01-06 by tgingold