ghdl/ghdl #625: help wanted: ghdl -a takes infinite time

faf5678, 2018-07-21:
the file in question is 1700 LOC and given the nature of the problem i guess it would be difficult to try and produce a MWE. code is proprietary. isim compiles it and all its dependencies in 1 minute.

is there a way to let ghdl print some kind of progress to narrow down the problem. (for example which process it is currently working on or somesuch).

I had problems with this source file and isim before and ended up having to introduce a cache for some compile time computations (store the result of them in a generated vhdl file). That's to say it is indeed fairly convoluted. But as stated above isim compiles the source as-is in 1 minute.

tgingold, 2018-07-21:
Try with 'ghdl -a -v '.

Which back-end are you using ?  (the answer is given by ghdl -v)

faf5678, 2018-07-21:
`/home/user/ghdl/bin/ghdl1-llvm --work=work -Punimacro/ -Punisim/ -P/home/user/ghdl/lib/ghdl/v93/std/ -P/home/user/ghdl/lib/ghdl/v93/synopsys/ -c -o my_entity.o ../fpga/rtl/my_entity.vhd`

tgingold, 2018-07-21:
Ok, then try with: ghdl -a -v -v
If you use the mcode back-end it should be much faster to compile.

faf5678, 2018-07-21:
i have rebuild my ghdl project from scratch, including importing unisim and unimacro with

`GHDL 0.36-dev (v0.35-229-g18116ef1)` from the current docker image and the problem does not occur anymore.

if you care i could see if a clean reimport of everything breaks 0.35