ghdl/ghdl #626: problem passing anonymous vector as generic (v0.36-dev)

faf5678, 2018-07-21:
ghdl version: (docker image)

```
GHDL 0.36-dev (v0.35-229-g18116ef1) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 llvm code generator
```

problematic code:
(the snippet
`vec   => (0 to ix => '1')`
is what IMO should be valid vhdl but does not work)

```
library ieee;
use ieee.std_logic_1164.all;

entity inc_ent is
	generic (
		works : integer;
		vec : std_logic_vector);
end entity;

architecture default of inc_ent is
begin
	assert false report integer'image(works) & " " & integer'image(vec'length);
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity top_ent is end entity;

architecture default of top_ent is
	constant foo_v : std_logic_vector(0 to 12) := (others => '1');
begin
	g : for ix in 0 to 4 generate
		constant foo_v : std_logic_vector(0 to ix) := (others => '1');
	begin
		
		inst : entity work.inc_ent
			generic map (
				works => 0,
				vec   => (0 to ix => '1')
				);
		inst2 : entity work.inc_ent
			generic map (
				works => 1,
				vec => foo_v
				);
	end generate;
end architecture;
```

output of the above code

```
../error.vhd:12:8:@0ms:(assertion error): 0 32764
../error.vhd:12:8:@0ms:(assertion error): 1 1
../error.vhd:12:8:@0ms:(assertion error): 0 32764
../error.vhd:12:8:@0ms:(assertion error): 1 2
../error.vhd:12:8:@0ms:(assertion error): 0 32764
../error.vhd:12:8:@0ms:(assertion error): 1 3
../error.vhd:12:8:@0ms:(assertion error): 0 32764
../error.vhd:12:8:@0ms:(assertion error): 1 4
../error.vhd:12:8:@0ms:(assertion error): 0 32764
../error.vhd:12:8:@0ms:(assertion error): 1 5
```

tgingold, 2018-07-21:
I can reproduce the error.  It works correctly with mcode, but not with llvm.

diogratia, 2018-07-22:
All ghdl revisions here were built today with commit 6a24c6d

diogratia@Macbook: $GHDL_VERSION
GHDL_VERSION: Undefined variable.
diogratia@Macbook: $GHDL --version
GHDL 0.36-dev (v0.35-231-g6a24c6d) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 mcode code generator
Written by Tristan Gingold.

diogratia@Macbook: pushd ~/Desktop
~/Desktop ~/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/mcode
diogratia@Macbook: $GHDL -a inc
inc_ent.o     inc_ent.vhdl
diogratia@Macbook: $GHDL -a inc_ent.vhdl
diogratia@Macbook: $GHDL -e top_ent
diogratia@Macbook: $GHDL -r top_ent
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/mcode/ghdl_mcode:error: overflow detected
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/mcode/ghdl_mcode:error: simulation failed
diogratia@Macbook:

diogratia@Macbook: echo $PATH
/Users/diogratia/share/clang+llvm-3.8.0-x86_64-apple-darwin/bin:/Users/diogratia/share/gnat-gpl-2017-x86_64-darwin-bin/bin:.:/Users/diogratia/bin:/usr/bin:/usr/sbin:/bin:/sbin:/usr/local/bin:/usr/X11/bin/:/usr/libexec

Adacore gnat-gpl-2017-x86_64-darwin-bin

This may be an OS X centric issue

diogratia@Macbook: $GHDL --version
GHDL 0.36-dev (v0.35-231-g6a24c6d) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 mcode code generator
Written by Tristan Gingold.

diogratia@Macbook: echo $PATH
/Users/diogratia/share/clang+llvm-3.8.0-x86_64-apple-darwin/bin:/Users/diogratia/share/gnat-gpl-2015-x86_64-darwin-bin/bin:.:/Users/diogratia/bin:/usr/bin:/usr/sbin:/bin:/sbin:/usr/local/bin:/usr/X11/bin/:/usr/libexec

Adacore gnat-gpl-2015-x86_64-darwin-bin

diogratia@Macbook: $GHDL -a inc_ent.vhdl
diogratia@Macbook: $GHDL -e top_ent
diogratia@Macbook: $GHDL -r top_ent
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 8
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 1
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 8
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 2
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 8
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 3
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 8
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 4
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 8
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 5
diogratia@Macbook:

Wrong value for assertions starting with 0 Those 8 values should be 1 - 5 respectively

Both mcode versions pass the testsuite.

llvm -

diogratia@Macbook: $GHDL --version
GHDL 0.36-dev (v0.35-231-g6a24c6d) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
Written by Tristan Gingold.

diogratia@Macbook: echo $PATH
/Users/diogratia/share/clang+llvm-3.8.0-x86_64-apple-darwin/bin:/Users/diogratia/share/gnat-gpl-2015-x86_64-darwin-bin/bin:.:/Users/diogratia/bin:/usr/bin:/usr/sbin:/bin:/sbin:/usr/local/bin:/usr/X11/bin/:/usr/libexec
diogratia@Macbook:

diogratia@Macbook: $GHDL -a inc_ent.vhdl
diogratia@Macbook: $GHDL -e top_ent
diogratia@Macbook: $GHDL -r top_ent
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32705
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 1
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32705
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 2
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32705
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 3
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32705
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 4
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32705
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 5
diogratia@Macbook: $GHDL -r top_ent
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32736
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 1
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32736
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 2
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32736
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 3
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32736
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 4
inc_ent.vhdl:12:8:@0ms:(assertion error): 0 32736
inc_ent.vhdl:12:8:@0ms:(assertion error): 1 5

(OS X has page boundary randomization, these only tell us the address is relative to something else.)

This llvm version also fails the testsuite Issue #578 (left a comment):

GNA dir issue578:
analyze crash2.vhdl

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: erroneous memory access
Load address: 0x10a608000
Call stack traceback locations:
0x10b354ae7 0x7fff8adc6528
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/build/ghdl_llvm:error: compilation error
#################################################################
######### FAILURE: issue578
#################################################################

diogratia@Macbook: cd gna/issue578
diogratia@Macbook: $GHDL -a crash2.vhdl

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: erroneous memory access
Load address: 0x10c79d000
Call stack traceback locations:
0x10d4e9ae7 0x7fff8adc6528
/Users/diogratia/Desktop/ghdl/ghdl-0.36dev_624c6d1/ghdl/build/ghdl_llvm:error: compilation error
diogratia@Macbook:

(I run the testsuite for every build I install. These had a `ghdl --clean` between the different attempts.)

tgingold, 2018-07-22:
I am testing a fix for it.

commit d3b9380a (2018-07-22): elab_object_init: allocate bounds for unounded aggregate if needed.

Fix #626
closed 2018-07-22 by tgingold