ghdl/ghdl #628: cosmetic: allow concise console preamble

faf5678, 2018-07-23:
now this can be done with awk no problem, but i think it maybe nice to have (as a runtime option). i rarely ever use a waveform viewer to debug code and the console is a real work environment facing the user. that should warrant some comfort.

```
rtl/foo/bar_tb.vhd:234:40:@5ns:(report note): my assert message
```

what i am interested 98% of the time would be

```
bar_tb:234:5ns: my assert message
```

out of curiosity, what does the "40" represent int he original preamble?

faf5678, 2018-07-23:
on second thought the full flexibility of a custom text processor is better than half-flexibility from a command line option.

closed 2018-07-23 by faf5678

1138-4EB, 2018-07-23:
I think that it is still interesting to know what does `40` represent. @tgingold, is it explained anywhere?

Paebbels, 2018-07-23:
`40` (column in a line) identifies the start of the report statement, because you could have a second report statement at character / column 80 :). No one writes such ugly code, but it's possible and the message must be uniquely identifying the correct report statement.