ghdl/ghdl #629: please help: any way to trace a report statement up the stack

faf5678, 2018-08-02:
i have numerous of the typical metavalue error messages in numeric_std, but no idea where in my code the function call is made.

thanks.

faf5678, 2018-08-04:
couldn't this be built in by default in the binary releases? or specifically in docker images where an extra dependency does not hurt at all? i know i am lazy, but this seems like an unnecessary burden to have me wrestle with the ghdl build environment.

1138-4EB, 2018-08-04:
We can:

- Add a flag to [dist/travis/build.sh](https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh) in order to have libbacktrace configured.
  - Add a check to ensure that it is only processed with either LLVM or GCC.
- If the flag is provided, build libbracktrace and add `--with-backtrace-lib=` to the execution of `configure`.

I am not sure about which additional steps are required. Is it compulsory to build libbacktrace before building GHDL? Should it be https://github.com/ianlancetaylor/libbacktrace or https://github.com/gcc-mirror/gcc/tree/master/libbacktrace?

When using GCC backend, sources should be already downloaded (as we get the full GCC codebase). However, when using LLVM, they might need to be downloaded.