ghdl/ghdl #635: Ram Initialization Leads To Compiler Crash

forflo, 2018-08-16:
**Description**

I just tried to simulate a block ram entity which looks roughly like the following snippet
```vhdl
	type RamType is array(natural range <>) of std_logic_vector(busWidth - 1 downto 0);
	subtype RamRange is natural range 0 to ramSize;

	signal ram : RamType(RamRange) := (
		0 => "00000000000000000000000000000001",
        -- ...
```

**Expected behaviour**
Should just compile.

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-197-g319092de) [Dunoon edition]
Compiled with GNAT Version: 8.1.1 20180531
Target: x86_64-pc-linux-gnu
In directory: /home/florian/datenportal/fau/orka/VHDLTarget/build/tests/array/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/ghdl1 --std=08 -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ --anaelab testbench --ghdl-source=/home/florian/orka/VHDLTarget/tests/array/testbench.vhdl --ghdl-source=/home/florian/orka/VHDLTarget/tests/../runtime/ram.vhdl --ghdl-source=test.vhdl -quiet -o e~testbench.s e~testbench
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Base_Type
Call stack traceback locations:
0x13c55b1 0x6319e9 0x73bf88 0x73cf73 0x7372c9 0x7293f9 0x72be26 0x71cb80 0x705c96 0x6e22a9 0x6dcbb0 0x6d3965 0x6d26eb 0x60904b 0xafbf4e 0x6030f5 0x60dfab 0x74509b 0x601601 0x7fe3dc068001 0x605218 0xfffffffffffffffe
******************************************************************

Execution of /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Base_Type
Call stack traceback locations:
0x13c55b1 0x6319e9 0x73bf88 0x73cf73 0x7372c9 0x7293f9 0x72be26 0x71cb80 0x705c96 0x6e22a9 0x6dcbb0 0x6d3965 0x6d26eb 0x60904b 0xafbf4e 0x6030f5 0x60dfab 0x74509b 0x601601 0x7fe3dc068001 0x605218 0xfffffffffffffffe
```

- OS: Arch LInux
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: ded2825eb086493655d90b499c3dcf3abd9e5867

**How to reproduce?**
I assembled a working example that reproduces the error.

https://github.com/forflo/ghdlBugReport

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
  I don't know what you mean by that
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2018-08-16:
I can reproduce the crash.  I will investigate.

forflo, 2018-08-16:
Merci beaucoup!

tgingold, 2018-08-16:
With a fix, I get:
"""
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@100ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@100ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@100ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@100ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@200ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@200ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@200ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3034:7:@200ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
tb.vhdl:71:24:@28200ns:(assertion failure): Unexpected result: intput1 = 10; output = 81; outputExpected = 59
ghdl_mcode:error: assertion failed
  from: work.testbench(test).simulation.check at tb.vhdl:71
  from: process work.testbench(test).simulation at tb.vhdl:80
ghdl_mcode:error: simulation failed
"""

Is that correct ?

forflo, 2018-08-17:
Looks pretty good. `outputExpected` does not represent a valid expectaion. Thank you very much :).

commit 35647707 (2018-08-17): trans-chap7: use get_index_type.  Fix #635
closed 2018-08-17 by tgingold