ghdl/ghdl #637: Access types to arrays cannot access individual elements.

clauniel, 2018-08-23:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity example is
end entity;

architecture tb of example is
    type arrType is array (1 downto 0) of integer;
    type arrTypePtr is access arrType;
begin
    process
        variable ptr : arrTypePtr;
    begin
        ptr := new arrType'(10, 5); -- works
        ptr.all(0) := 5; -- crash
        ptr(0) := 5; --crash
    end process;

end architecture;

```

This results in a crash which the following output

```
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x56348daa1886 0x56348daf095d 0x56348daf04dc 0x56348daf05b9 0x56348daf0bca 0x56348daf107f 0x56348db526ef 0x56348db57a13 0x56348dbc14d1 0x56348db28398 0x56348db253e3 0x56348db32eb9 0x56348dbecdf7 0x56348dbd9fab 0x56348dbc9d23 0x56348dbf99d5 0x56348d8f1e9c 0x7f8bb7e98069 0x56348d8f0b68 0xfffffffffffffffe
```

This is similar to #473 but here the array type is constrained.

diogratia, 2018-08-23:
You'd do well to provide the complete error message or provide the first three lines of ghdl --version. It isn't clear from your report whether you have an analysis or run time failure, neither occurs for:
GHDL 0.36-dev (v0.35-267-g3bf65cc) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator

a recent commit built with llvm-3.8.0 and Adcore's 2015 gnat.

(Issue 473 is a -2008 feature, this isn't.)

clauniel, 2018-08-23:
Thanks, I just updated the OP. I forgot to make the process sensitized, the crash does not happen for an unsensitized process.

diogratia, 2018-08-24:
The crash appears to be occurring because of the reserved word `all` in the process sensitivity list when there are in fact no signals evaluated in the process. 5.4.1:

>An object declared to be of an access type shall be an object of class variable. An object designated by an access value is always an object of class variable.

There's nothing at example.vhd:15:13 (the second occurrence of the reserved word `all`) that can be expanded into the name of a signal. The search for sensitivity list signals isn't quite bullet proof.

Without a wait statement somewhere in the process and without adding the sensitivity list with the reserved word `all` your process doesn't stop execution during initialization. (And it memory leaks, to the point of bringing my Mac to non usability crossing 26 GB of virtual memory):

9.3.7 Allocators:

>For the evaluation of an allocator, the elaboration of the subtype indication or the evaluation of the qualified expression is first performed. The new object is then created, and the object is then assigned its initial value. Finally, an access value that designates the created object is returned.  
>  
>In the absence of explicit deallocation, an implementation shall guarantee that any object created by the evaluation of an allocator remains allocated for as long as this object or one of its subelements is accessible directly or indirectly; that is, as long as it can be denoted by some name.  
>  
>NOTE 1—Procedure deallocate is implicitly declared for each access type. This procedure provides a mechanism for explicitly deallocating the storage occupied by an object created by an allocator.  
>NOTE 2—An implementation may (but need not) deallocate the storage occupied by an object created by an allocator, once this object has become inaccessible.  

And that is perfectly 'legal' and not guaranteed to change in the remaining portion of my lifetime (noting there are no fully compliant -2008 tools whatsoever today).

P1076-2018 9.3.7:

>An implementation should deallocate the storage occupied by an object created by an allocator, once this object has become inaccessible. Deallocation may be delayed by the implementation.

where (1.3.1):

>... The word **should** is used to indicate a recommendation. ...

See [Language Change Specification for Garbage Collection](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_030) and the requirement Garbage Collection](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/GarbageCollection). At one time garbage collection was intended to become mandatory.

You could note there's a similar case with line access types and readline/writeline in textio, wherein no one is copying line access values (it's not useful with automatic deallocation of line storage on reading from or writing to a file, almost sure to cause a crash). 

<hr>

There appears to be a shortcoming in the -2008 standard in that the sensitivity list expanded from the reserved word `all` is not erroneous if `all` expands to an empty set other than syntactically:

11.3 Process statement:

>process_statement ::=   
>[ *process_*label : ]
>&nbsp;&nbsp;&nbsp;&nbsp;    [ **postponed** ] **process** [ ( process_sensitivity_list ) ] [ **is** ] 
> &nbsp;&nbsp;&nbsp;&nbsp;   ...  
>  
>process_sensitivity_list ::= **all** | sensitivity_list
> ...
>If a process sensitivity list appears following the reserved word process, then the process statement is assumed to contain an implicit wait statement as the last statement of the process statement part; this implicit wait statement is of the form  
>  
> **wait on** sensitivity_list ;

10.2 Wait statement:

>wait_statement ::=  
> &nbsp;&nbsp;&nbsp;&nbsp;[ label : ] **wait** [ sensitivity_clause ] [ condition_clause ] [ timeout_clause ] ;    
>  
> sensitivity_clause ::= on sensitivity_list  
>  
>sensitivity_list ::= signal_name { , signal_name }

Requiring at least one signal name syntactically, while the sensitivity list is generated semantically:

>&nbsp;&nbsp;&nbsp;&nbsp;— Otherwise, do not add to the sensitivity set.

Regardless, you'd anticipate a failure during simulation. There is however no semantic hook to hang your hat on for the failure. There's a need for vendor (ghdl) intervention here.

clauniel, 2018-08-24:
I don't think the memory leak is important. The crash happens during elaboration and not simulation. The same thing happens if do you actually add dependence on a sensitized signal and deallocate the access type, e.g

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity example is
end entity;

architecture tb of example is
    type arrType is array (1 downto 0) of integer;
    type arrTypePtr is access arrType;
    signal clk : std_logic := '0';
begin
    
    clk <= not clk after 10 ns;

    process (all)
        variable ptr : arrTypePtr;
    begin
        if rising_edge(clk) then
            ptr := new arrType'(10, 5); -- works
            ptr.all(0) := 5; -- crash
            ptr(0) := 5; --crash
            deallocate(ptr); -- no memory leak
        end if;
    end process;

end architecture;
```

diogratia, 2018-08-24:
This one analyzes, elaborates and simulates (and still leaks memory).

commit 6e0eb7b9 (2018-08-24): canon_extract_sensitivity: handle dereference.  Fix #637
closed 2018-08-24 by tgingold