ghdl/ghdl #638: 64 bit pointer can not be assigned to var of type access

1138-4EB, 2018-08-27:
**Description**
I tried some of the examples/extensions from [ygdes.com/GHDL](http://ygdes.com/GHDL/). With `runme2.sh` and `runme3.sh` in example `fb/scripts` I got a `NULL access dereferenced. simulation failed` error.

The source was that `mmap` in the C extension returned a 64 bit pointer, that was passed to the VHDL access type variable. The access type seems to be of 32 bits, so half of the pointer was lost. Then, when the shared variable was used, the error was produced.

I could work around it by passing flag `MAP_32BIT` to `mmap`. With this, the returned address fits and no error is produced. However, I'd like to know whether this is a limitation of the standard, an implementation decision or an error on my side.

- OS: Fedora 28
- Origin: ghdl/ghdl:fedora28-llvm

```
GHDL 0.36-dev (v0.35-271-g5cd3de46) [Dunoon edition]
 Compiled with GNAT Version: 8.1.1 20180712 (Red Hat 8.1.1-5)
 llvm code generator
Written by Tristan Gingold.
```

**How to reproduce?**

I uploaded a MWE to https://github.com/1138-4EB/hwd-ide/tree/access-error-reproducer/examples/access_error. It works as is:

``` bash
$ git clone -b access-error-reproducer https://github.com/1138-4EB/hwd-ide
$ cd hwd-ide
$ docker run --rm -tv /$(pwd)/examples/access_error:/src:z -w //src ghdl/ghdl:fedora28-llvm ./runme.sh
41fd3000
test.vhdl:12:7:@0ms:(report note): integer=0
test.vhdl:12:7:@0ms:(report note): integer=1
test.vhdl:12:7:@0ms:(report note): integer=2
test.vhdl:12:7:@0ms:(report note): integer=3
test.vhdl:12:7:@0ms:(report note): integer=4
```

However, if `MAP_32BIT` is removed from https://github.com/1138-4EB/hwd-ide/blob/access-error-reproducer/examples/access_error/access_mmap.c#L10, the output is:

``` bash
$ docker run --rm -tv /$(pwd)/examples/access_error:/src:z -w //src ghdl/ghdl:fedora28-llvm ./runme.sh
7f4d6c0e3000
./access_test:error: NULL access dereferenced
./access_test:error: simulation failed
```

tgingold, 2018-08-27:
Your get_p function returns an int32_t, so you removed half of the pointer.

1138-4EB, 2018-08-27:
As stupid as that :man_facepalming:. Thanks a lot. Sorry for bothering you.

closed 2018-08-27 by 1138-4EB