ghdl/ghdl #639: VHPI: first foreign declaration limits the return type

1138-4EB, 2018-08-27:
**Description**
In #638 I did not properly extract the MWE corresponding to exercise `fb` in http://ygdes.com/GHDL/.
In [`mmap2img`](https://github.com/1138-4EB/hwd-ide/tree/mmap2img/examples/VHPI/mmap2img) the same error is produced (`NULL access dereferenced. simulation failed`), but the C part is now hopefully correct.

Two VHDL functions are bind to the same C function:

- https://github.com/1138-4EB/hwd-ide/blob/mmap2img/examples/VHPI/mmap2img/mmap_ghdl.vhdl#L14-L16
- https://github.com/1138-4EB/hwd-ide/blob/mmap2img/examples/VHPI/mmap2img/mmap_ghdl.vhdl#L27-L29

Since the first one is defined in VHDL with a return type `integer`, the return value of the C function seems to be cast to 32 bits. As a result, when the alias is used (type `screen_p`), it fails. I cannot declare them in reverse order because the first one is used to retrieve the parameters used to the define `screen_p`.

The workaround I found is to first declare a dummy function that expects a custom type. See https://github.com/1138-4EB/hwd-ide/blob/mmap2img/examples/VHPI/mmap2img/mmap_ghdl.vhdl#L8-L10.

**Expected behaviour**
If one of the aliases requires the returned value to be of 64 bits, it should be kept, independently of which one is the first declaration.

**Context**

- OS: Fedora 28
- Origin: ghdl/ghdl:fedora28-llvm

```
GHDL 0.36-dev (v0.35-271-g5cd3de46) [Dunoon edition]
 Compiled with GNAT Version: 8.1.1 20180712 (Red Hat 8.1.1-5)
 llvm code generator
Written by Tristan Gingold.
```

**How to reproduce?**

Execute https://github.com/1138-4EB/hwd-ide/blob/mmap2img/examples/VHPI/mmap2img/scripts/runme.sh. Then remove/comment https://github.com/1138-4EB/hwd-ide/blob/mmap2img/examples/VHPI/mmap2img/mmap_ghdl.vhdl#L8-L10 and https://github.com/1138-4EB/hwd-ide/blob/mmap2img/examples/VHPI/mmap2img/mmap_ghdl.vhdl#L39-L41, and execute `runme.sh` again.

tgingold, 2018-08-28:
I will investigate, but it looks like you are looking for problems: why are you using the same function to return a pointer and a number ?  Why not defining two different functions ?

1138-4EB, 2018-08-28:
My work is based on example `GHDL/fb/` in ygdes.com/GHDL/, which was published in GNU/Linux Magazine France #130, september 2010. I found three options to 'fix' it:

- Use `MAP_32BIT` when calling `mmap`.
- Use two different functions, as you suggest.
- Use a dummy type, as commented in this issue.

So, this issue is for completeness only.

1138-4EB, 2018-08-31:
You mean withouth `MAP_32BIT`, with a single function and without the dummy type and function in `mmap_ghdl.vhdl`?

tgingold, 2018-09-02:
Ok thanks.  I now understand what happens.

For the compiler, there is only one definition of the imported get_p function.  So if you declare it with different prototypes/profiles, something weird may happen.

I think we'd better not try to fix that given your testcase is not fully valid.  So let's close it.

closed 2018-09-02 by 1138-4EB