ghdl/ghdl #640: Wrap GHDL simulation in a shared library

1138-4EB, 2018-09-02:
I am wrapping a GHDL simulation in a C program. I have successfully done it by building a `main.c` file with a `int main(int argc, char **argv)` function that calls a function defined in a different file (`wrapper.c`). `ghdl_main` is called from the latter. The build procedure is as follows:

``` bash
$ ghdl -a -fPIC ../screen_ghdl.vhdl ../test.vhdl
$ ghdl --bind -fPIC test_screen
$ gcc -c -fPIC ../wrapper.c -o wrapper.o
$ gcc ../main.c -o ../bin/main -Wl,wrapper.o -Wl,`ghdl --list-link test_screen`
```

Now I'd like to build a shared library `libwrapper.so`, and a program (`shared.c`) that uses it:

``` bash
$ gcc -g3 -fPIC -shared -o libwrapper.so -Wl,wrapper.o -Wl,`ghdl --list-link test_screen`
$ gcc -L$PWD -o ../bin/shared ../shared.c -lwrapper -ldl
```

However, the last command fails:

``` bash
/tmp/ccHFxRna.o: In function `main':
.../hwd-ide/examples/ghdl-so/obj/../shared.c:25: undefined reference to `ghdl_mm'
collect2: error: ld returned 1 exit status
```

Indeed, the three functions defined in `wrapper.c` are not listed with `nm -D libwrapper.so`. But, they are found with objdump:

``` bash
$ objdump -d libwrapper.so | grep "<ghdl_mm>"
000000000000572e <ghdl_mm>:
$ objdump -d libwrapper.so | grep "<get_p>"
0000000000005699 <get_p>:
    613a:	e8 5a f5 ff ff       	callq  5699 <get_p>
$ objdump -d libwrapper.so | grep "<get_m>"
00000000000056f7 <get_m>:
    619f:	e8 53 f5 ff ff       	callq  56f7 <get_m>
```

**Context**

- OS: Fedora 28
- Origin: built from sources (using ghdl/docker) by adding `OPT_FLAGS+=-fPIC` and `LIB_CFLAGS+=-fPIC` to the makefile.

```
GHDL 0.36-dev (v0.35-271-g5cd3de46-dirty) [Dunoon edition]
 Compiled with GNAT Version: 8.1.1 20180712 (Red Hat 8.1.1-5)
 llvm code generator
```

**How to reproduce?**

I uploaded the code to https://github.com/1138-4EB/hwd-ide/tree/ghdl-so/examples/ghdl-so. Executing `runme.sh` reproduces the steps above:

``` bash
|> Analyze VHDL sources (screen_ghdl.vhdl, test.vhdl)
|> Bind simulation unit
|> Build 'wrapper.o' for 'ghdl_mm'
|> Build 'main'
|> Run 'main'
> Call 'ghdl_mm'
>> ghdl_mm
>> ghdl_main
get_p(1): 14
get_p(0): 2002260
|> Build 'wrapper.so'
|> Build 'shared'
/tmp/ccHFxRna.o: In function `main':
...hwd-ide/examples/ghdl-so/obj/../shared.c:25: undefined reference to `ghdl_mm'
collect2: error: ld returned 1 exit status
```

1138-4EB, 2018-09-04:
Upon further inspection, this issue is related to argument `-Wl,--version-script=/opt/ghdl-llvm/lib/ghdl/grt.ver` provided by `ghdl --list-link test_screen`. This seems to limit the visible symbols to those listed there:

```
{
  global:
vpi_chk_error;
vpi_control;
vpi_free_object;
vpi_get;
vpi_get_str;
vpi_get_time;
vpi_get_value;
vpi_get_vlog_info;
vpi_handle;
vpi_handle_by_index;
vpi_handle_by_name;
vpi_iterate;
vpi_mcd_close;
vpi_mcd_name;
vpi_mcd_open;
vpi_put_value;
vpi_register_cb;
vpi_register_systf;
vpi_remove_cb;
vpi_scan;
vpi_vprintf;
vpi_printf;
  local:
	*;
};
```

If `ghdl_mm` is added to a copy of it, and this is used instead, the example above works. E.g.:

``` bash
cat /opt/ghdl-llvm/lib/ghdl/grt.ver | sed -e 's#^\(.*local:\)#ghdl_mm;\n\1#g' > ghdl.ver

gcc -g3 -fPIC -shared -o libwrapper.so -Wl,wrapper.o -Wl,`ghdl --list-link test_screen | sed -e 's#/opt/ghdl-llvm/lib/ghdl/grt.ver#ghdl.ver#g'`
```

However, if find this procedure error prone, as it requires to manually add the symbols expected to be visible.

1138-4EB, 2018-09-04:
Indeed, I achieved that with https://github.com/1138-4EB/ghdl/commit/d666872b3126ebf0465c821ea4d49e4e7101d1e6. So with the GHDL I am using (LLVM backend) all the libraries that belong to GHDL are built with `-fPIC`. The objects produced with `ghdl -a` are also done with `-fPIC`.

The issue I have is about the visibility of symbols. This affects any ELF created with `-Wl,$(ghdl --list-link...)`, no matter if it is a 'regular' executable or a shared library:

``` bash
$ gcc ../main.c -o main_wrapper_nosym -Wl,wrapper.o -Wl,`ghdl --list-link test_screen`
$ nm -D main_wrapper_nosym | grep ghdl_mm
# No output
$ nm -D main_wrapper_withsym | grep ghdl_mm
000000000040209a T ghdl_mm
$ gcc ../main.c -o main_wrapper_withsym -Wl,wrapper.o -Wl,`ghdl --list-link test_screen | sed -e 's#/opt/ghdl-llvm/lib/ghdl/grt.ver#ghdl.ver#g'`
```

tgingold, 2018-09-05:
I think there are two solutions:

* either you create a libgrt.so and you link the vhdl code with it (instead of with libgrt.a)
* or you remove the version script option when you create your shared library.

I don't know what is the easiest...

1138-4EB, 2018-09-05:
Would it be reasonable to add a `--no-version-script` optional arg to `ghdl --list-link`, in order to have it automatically removed from the list?
Alternatively, a `--add-global-sym="<symbol1,symbol2,..."` that added these to the default version script could also help.

Anyway, please leave this issue as a remainder to add some reference in the docs:

- https://ghdl.readthedocs.io/en/latest/references/CommandReference.html#list-link-list-link
- https://ghdl.readthedocs.io/en/latest/references/ImplementationOfVHDL.html#starting-a-simulation-from-a-foreign-program

tgingold, 2018-09-19:
Why about filtering the output of --list-link with sed ?
It is not a definitive solution, but a work-around.

(Sorry for the late reply).

1138-4EB, 2018-09-19:
> Why about filtering the output of --list-link with sed ?
> It is not a definitive solution, but a work-around.

Yes. It works. That's what I'm doing ATM. See:

- Comment above: https://github.com/ghdl/ghdl/issues/640#issuecomment-418476573
- Example in hwd-ide: https://github.com/1138-4EB/hwd-ide/blob/examples-vhpi/examples/VHPI/sharedobject/runme.sh#L31-L34

> (Sorry for the late reply).

No problem at all. This not a problem/bug, just a possible enhancement. So, please, prioritize other issues.

1138-4EB, 2018-10-10:
I found a possible bug related to `ghdl --list-link`:

``` vhdl
# test.vhd
entity test is end entity;
architecture arch of test is begin end arch;
```

``` bash
$ ghdl -a test.vhd
$ ghdl --bind test
$ ghdl --list-link test
```

Output on `x86_64`:

``` bash
e~test.o
/opt/ghdl-llvm/lib/ghdl/std/v93/std_standard.o
test.o
/opt/ghdl-llvm/lib/ghdl/libgrt.a
-ldl
-lm
-Wl,--version-script=/opt/ghdl-llvm/lib/ghdl/grt.ver
-Wl,--export-dynamic
-L./
-lz
-ldl
```

Output on `arm64`:

``` bash
e~test.o
/opt/ghdl-llvm/lib/ghdl/std/v93/std_standard.o
test.o
/opt/ghdl-llvm/lib/ghdl/libgrt.a
-ldl
-lm
-Wl,--version-script=/opt/ghdl-llvm/lib/ghdl/grt.ver
-Wl,--export-dynamic
-L./
-lz
```

Output on `arm32`:

``` bash
e~test.o
/usr/local/lib/ghdl/std/v93/std_standard.o
test.o
/usr/local/lib/ghdl/libgrt.a
-L./
-lz
```

The possible bug is that `-ldl` is added twice on `x86_64`, once on `arm64` and none on `arm32`. Also, `-lm` is missing on `arm32`. As a result, a runtime error is produced on `arm32` during elaboration, when trying to use `dlopen`. This breaks both `ghdl -e` and VUnit.

As a workaround, I manually added `-ldl` to the output of `--list-link` when embedding GHDL in a foreign language. It works ok.

@tgingold, should I open a separate issue for this?

tgingold, 2018-10-10:
The libraries to be added are defined in src/grt/Makefile.inc, and that depends on the architecture.
So for arm/aarch64 you have to adjust the makefile (if needed).

closed 2019-04-24 by tgingold

1138-4EB, 2019-04-24:
Now that #801 is merged, I will add the following hint to the docs:

> One of the arguments returned by `--list-link` is `-Wl,--version-script=PREFIX/lib/ghdl/grt.ver`, where PREFIX is the installation path of GHDL. This will hide most of the symbols when the target executable binary is built. In some contexts, where the binary is to be loaded dynamically, the user might want additional symbols to be accessible. There are two possible approaches to have it done:
> * Filter the output of `--list-link` with e.g. `sed`.
> * Provide an additional non-anonymous version script: `-Wl,-Wl,--version-script=file.ver`.

---

I think it would still be useful to have a `--no-version-script` as an option to `--list-link`. This would avoid the requirement to use `sed` when the user does not want to use any version script at all. This is useful to let all the symbols be visible without having to provide an additional `custom.ver` file where additional symbols are listed explicitly.

Let me elaborate on this: the current default `grt.ver` includes a wildcard (`*`) that makes all the symbols which are not explicitly listed to be local (hidden). This prevents the user from using a wildcard in the additional custom version script. Thus, it is a requirement to filter the output of `--list-link` or to explicitly list the names of all the desired symbols.

Nevertheless, GHDL is usable ATM, without requiring the modification of any global or system file. We can keep this issue closed.

tgingold, 2020-04-10:
So let's try to implement this feature.
First, I will put the shared objects commands (version script and export dynamic) in a separate .lst file.  Then add an option to not include it.

PR #801 (2019-04-24): add some version to grt.ver (#800)
Close #800 and Close #640.

As commented in #800, this PR adds a version (`ANY`) to `grt.ver`, so it is possible to add other visible symbols to the binary generated with `ghdl -e [args] -Wl,-Wl,--version-script=custom.ver`.