ghdl/ghdl #641: Creating an object from constrained object that was formerly unconstrained causes an "is unconstrained" error.

Paebbels, 2018-09-02:
**Description**
I'm trying to use the AXI4 Lite model from OSVVM with GHDL. It uses unconstrained and partially constrained record types. The creation of objects (e.g. a signal) from such type is working, but when another signal is created from such an object with `'subtype` an "is unconstrained" error is emitted.

**Expected behavior**
Is should allow creating an object with `'subtype` from a formerly unconstrained type.

I suspect the constraint is not propagated correctly. As comparison, Aldec Riviera-PRO has no problems with that code.

**Context**

    $ ghdl -a --std=08 /c/Temp/ghdl/UnconstrainedRecordUsingSubtype.vhdl
    C:/Temp/ghdl/UnconstrainedRecordUsingSubtype.vhdl:27:15: declaration of signal "data1" with unconstrained array subtype "std_logic_vector" is not allowed
    C:/Temp/ghdl/UnconstrainedRecordUsingSubtype.vhdl:28:34: prefix must denote an object
    C:/Temp/ghdl/UnconstrainedRecordUsingSubtype.vhdl:28:34: a type mark must be a simple or expanded name

- OS: Windows
- Origin:
  - [x] Built from sources.  
         latest 0.36-dev

**Additional context**
While `'subtype` in `line 27` is not working, but the workaround in `line 26` with `'range` is working. Thus, the constraint exists but is not used with `'subtype`. It seems as if `'subtype` behaves like `'base`.

Another strange behavior is shown in `line 28`. A selected name is not accepted as a prefix, while an intermediate alias is working. See `line 27`.

**How to reproduce?**

```vhdl
#>> ent.vhd
-- Helper package
library ieee;
use     ieee.std_logic_1164.all;

package p is
	type BusRecord is record
		Address : std_logic_vector;
		Data    : std_logic_vector;
	end record;
end package;

-- DUT
library ieee;
use     ieee.std_logic_1164.all;
use     work.p.all;

entity e is
	port (
		Axi : inout BusRecord
	);
end entity;

architecture a of e is
	alias  AxiData is Axi.Data;

	signal Address : std_logic_vector(Axi.Address'range);
	signal Data1   : AxiData'subtype;    -- line 27: declaration of signal "data1" with unconstrained array subtype "std_logic_vector" is not allowed
	signal Data2   : Axi.Data'subtype;   -- line 28: prefix must denote an object; a type mark must be a simple or expanded name
begin

end architecture;

-- Top Level
library ieee;
use     ieee.std_logic_1164.all;
use     work.p.all;

entity test is
end entity;

architecture tb of test is
	signal AxiBus : BusRecord(
		Address(7 downto 0),
		Data(31 downto 0)
	);
begin
	DUT: entity work.e
		port map (
			Axi => AxiBus
		);
end architecture;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run test

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

Paebbels, 2018-09-02:
According to Riviera-PRO, the intermediate alias declaration is not needed to use `'subtype`. Thus, select names should be supported as prefix too, right?

JimLewis, 2020-06-19:
@csantosb  Is there a schedule to have this issue fixed.   If it will take time, and @Paebbels work around addresses all of the issues with OSVVM models, I may want to update OSVVM so they work with the tools as they are.   For a commercial EDA tool, it would be a requirement for me to fix it as vendors are slow to fix bugs and even with a vendor fix, a user may not be empowered to install the latest version of the tool.

@Paebbels  Does the alias method fix all of the issues in the OSVVM AXI model?   Should I make these changes in the released version of OSVVM?

csantosb, 2020-06-20:
@JimLewis Not a tight schedule from my side. It would be nice to get it working, though.

JimLewis, 2020-06-20:
@csantosb  I was wondering if and/or when GHDL would have the feature fixed?   If it is going to take some time to fix GHDL, then I will look into working around GHDL's issue in code.   Note it does make the code more clunky, but working models are better than pretty ones.   

My OSVVM hat says working around tool issues is good as it provides models for the community.  OTOH, my VHDL hat says it is bad to do this as it promotes the myth that VHDL is more verbose than other languages - which with VHDL-2008 is no longer true.

csantosb, 2020-06-21:
@JimLewis I see your point, this is rather usual in vhdl, unfortunately. My approach in this situation is to branch the lrm compliant code, fix, document the problem and open an issue upstream, hoping it gets fixed soon. In the meanwhile, I move forward, keeping the backup branch for the day the issue gets fixed. Mostly works for minor issues easy to work around.

JimLewis, 2020-06-21:
@csantosb Currently in fire drill mode working on other tasks.   I am guessing that I am two weeks out for an update of OSVVM to work around this issue.

diogratia, 2020-06-22:
**@Paebbels Does the alias method fix all of the issues in the OSVVM AXI model? Should I make these changes in the released version of OSVVM?**

According to 6.4.1 an element of an object is an object.

According to 6.6.2 Object aliases c):

>1) If the subtype indication is absent  
>>— If the alias designator denotes a slice of an object, then the slice of the object is viewed as if it were of the subtype specified by the slice.  
>>— If the name is an external name, then the object is viewed as if it were of the subtype spec- ified in the external name.  
>>— Otherwise, the object is viewed as if it were of the subtype specified in the declaration of the object denoted by the name.  

The last clause tells us the error for line 27 is valid, the subtype declaration is unconstrained. Further d) applies the above specifically to the present circumstances:

>d) When the prefix of an attribute name denotes the alias defined by the alias declaration, subrules 1), 2), and 3), of rule c) apply.

Here ghdl is following the standard.

**As comparison, Aldec Riviera-PRO has no problems with that code.**

This appears to be standard non-compliant behavior which Annex D tells us may be non-portable.

Correcting the alias declaration to include a subtype indication:

```vhdl
    -- alias AxiData   is Axi.Data;
    alias  AxiData: std_logic_vector(Axi.Data'range) is Axi.Data;
```
allows the alias to be used as expected.

JimLewis, 2020-06-22:
@diogratia  Do any objects exist without constraints?   Only ports can be unconstrained and their subtype is determined by the rules of  5.3.2.2 Index constraints and discrete ranges.

Paebbels, 2020-08-05:
Related issue: https://github.com/ghdl/ghdl/issues/1128

Paebbels, 2021-05-12:
Further tracking also at https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/167.

JimLewis, 2021-05-13:
@diogratia 
> This appears to be standard non-compliant behavior which Annex D tells us may be non-portable.

I don't see anything in Annex D that refers to object aliases to arrays ports.   So can you explain why you make this sort of claim?

JimLewis, 2021-05-13:
@tgingold   I have created a number of test cases to try to find the root cause of this issue.   I pushed them to my GitLab repository:   https://gitlab.com/synthworks/ghdl_alias_subtype_issue

Quick summary:  It sure looks like object aliases mangle type information when the named object is either a port whose type is a record type with unconstrained elements or a signal whose type is derived using 'subtype and the port is a record type with unconstrained elements.    It is interesting as in the case of the signal, if 'range is used instead of 'subtype to constrain the type, then the object alias works just fine.

What I see is that:
* Ports whose type is has unconstrained elements work.
* Deriving subtypes from these ports using 'subtype works.
* Using either of the the declarative region (by using 'range or 'length) works fine.
* Accessing elements of the first two items works fine.
* Create an object alias without constraints to either of the first two and use it, causes GHDL to crash.
* Create an object alias with constraints to either of the first two results in a constraints error - unfortunately these tests cases are not directly executable (you have to fiddle with the commented out code in test cases 2b and 2g)

See the repository and test cases for details.   Hopefully this will help clear this issue as I really want the OSVVM AXI models to start working in GHDL.   I also provided additional analysis comments in the 1076 issues page at https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/167

Well I am going to bed - mainly because I have to be up in 4 hours.

tgingold, 2021-05-13:
I have fixed the crash for unbounded records.
All the tests have been added as regressions tests and are now passing.

Thanks!

JimLewis, 2021-05-13:
@tgingold  Thanks for the quick update.   Now that I have a decent setup for running GHDL scripts, I hope to be able to chase down issues like this.   

Look forward to getting to further testing of OSVVM VC and either see what next or to see them fully operational.

JimLewis, 2021-05-31:
@Paebbels   From my perspective, this has been fixed.   Do you concur?   Maybe it should be closed?

Paebbels, 2021-05-31:
If OSVVM without workarounds for GHDL, yes.
@JimLewis so intermediate workarounds are removed?

JimLewis, 2021-08-07:
@Paebbels   The models have been cleaned up significantly and no longer have any of ugly workarounds they previously had.    There are some issues remaining, but I have worked around them and have not had time to go back and chase them.

umarcor, 2021-08-16:
I'm closing this, since the reproducer is working now, and the discussion about the language is tracked elsewhere.

closed 2021-08-16 by umarcor