ghdl/ghdl #644: '_' is not supported in normal (binary) bit-string literals

Paebbels, 2018-09-02:
**Description**
The `'_'` character was introduced to add a visibility separator in long bit-string literals. While this new VHDL-2008 feature is supported in hex, octal, decimal and (explicit) binary bit-string literals, it is not supported in ordinary binary bit-string literals:

```vhdl
constant C_COLOR_2 : std_logic_vector(31 downto 0) := "000000000000_10101010101010101010";
```

**Expected behavior**
This should be supported.

**Context**
```
$ ghdl -a --std=08 /c/Temp/ghdl/SeparatorInBitStringLiteral.vhdl
C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:8:62: type bit does not define character '_'
C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:8:62: string length does not match that of anonymous integer subtype defined at C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:8:40
C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:10:62: type std_ulogic does not define character '_'
C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:10:62: string length does not match that of anonymous integer subtype defined at C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:10:46
```

- OS: Windows
- Origin:
  - [x] Built from sources.  
      latest 0.36-dev

**Additional context**
While the syntax in `line 8` and `line 10` is not supported yet, the syntax is accepted for `line 14` if the bit-string is specified as *binary*. Note, the length `32` is not needed. The behavior is the same.

```vhdl
constant C_COLOR_2 : std_logic_vector(31 downto 0) :=    "000000000000_10101010101010101010"; -- line 10
constant C_COLOR_6 : std_logic_vector(31 downto 0) := 32b"000000000000_10101010101010101010"; -- line 14
```

**How to reproduce?**

```
#>> ent.vhd
library ieee;
use     ieee.std_logic_1164.all;

entity test is
end entity;

architecture a of test is
	constant C_COLOR_0 : bit_vector(31 downto 0)       := "000000000000_10101010101010101010";    -- line 8:  type bit does not define character '_'
	constant C_COLOR_1 : bit_vector(31 downto 0)       := x"0000_00_00";                          --          string length does not match that of anonymous integer subtype defined at C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:8:40
	constant C_COLOR_2 : std_logic_vector(31 downto 0) := "000000000000_10101010101010101010";    -- line 10: type std_ulogic does not define character '_'
	constant C_COLOR_3 : std_logic_vector(31 downto 0) := x"0000_00_00";                          --          string length does not match that of anonymous integer subtype defined at C:/Temp/ghdl/SeparatorInBitStringLiteral.vhdl:8:40
	constant C_COLOR_4 : std_logic_vector(31 downto 0) := 32d"0000_00_00";
	constant C_COLOR_5 : std_logic_vector(31 downto 0) := 32o"0000_00_00";
	constant C_COLOR_6 : std_logic_vector(31 downto 0) := 32b"000000000000_10101010101010101010"; -- line 14: here is is supported
begin
end architecture;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run test

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

Paebbels, 2018-09-02:
I labeled this issue as a bug, because the underlying VHDL-2008 should be fully implemented.

diogratia, 2018-09-03:
It's not just -2008. See IEEE Std 1076-1993:

> 13.7 Bit string literals
> 
> A bit string literal is formed by a sequence of extended digits (possibly none)enclosed between two quotations used as bit string brackets, preceded by a base specifier.
> 
>      bit_string_literal ::=  base_specifier " [ bit_value ] "
> 
>      bit_value ::=  extended_digit { [ underline ] extended_digit }
> 
>      base_specifier ::=  B | O | X
> 
>**An underline character inserted between adjacent digits of a bit string literal does not affect the value of this literal.** The only letters allowed as extended digits are the letters A through F for the digits ten through fifteen. A letter in a bit string literal (either an extended digit or the base specifier) can be written either in lowercase or in uppercase, with the same meaning.
> 
> If the base specifier is 'B', the extended digits in the bit value are restricted to 0 and 1. If the base specifier is 'O', the extended digits in the bit value are restricted to legal digits in the octal number system, i.e.,the digits 0 through 7. If the base specifier is 'X', the extended digits are all digits together with the letters A through F.
> 
> **A bit string literal has a value that is a string literal consisting of the character literals '0' and '1'. If the base specifier is 'B', the value of the bit string literal is the sequence given explicitly by the bit value itself after any underlines have been removed.**
> 
> If the base specifier is 'O' (respectively 'X'), the value of the bit string literal is the sequence obtained by replacing each extended digit in the bit_value by a sequence consisting of the three (respectively four) values representing that extended digit taken from the character literals '0' and '1';as in the case of the base specifier 'B', underlines are first removed. Each extended digit is replaced according to the table on the following page:  
>  
> Extended digit	Replacement when the base specifier is	Replacement when the base specifier is
>  
> ```none
>               'O'       'X'
>     0 	      000	0000
>     1	      001	0001
>     2	      010	0010
>     3	      011       0011
>     4	      100       0100
>     5	      101       0101
>     6	      110       0110
>     7	      111       0111
>     8	    (illegal)	1000
>     9	    (illegal)	1001
>     A	    (illegal)	1010
>     B	    (illegal)	1011
>     C	    (illegal)	1100
>     D	    (illegal)	1101
>     E	    (illegal)	1110
>     F	    (illegal)	1111
> ```
> 
> The length of a bit string literal is the length of its string literal value.
> 
> Example:
> ```
>      B"1111_1111_1111"         --  Equivalent to the string literal "111111111111"
>      X"FFF"                    --  Equivalent to B"1111_1111_1111"
>      O"777"                    --  Equivalent to B"111_111_111"
>      X"777"                    --  Equivalent to B"0111_0111_0111"
> 
>      constant c1: STRING := B"1111_1111_1111";
> 
>      constant c2: BIT_VECTOR := X"FFF";
> 
>      type MVL is ('X', '0', '1', 'Z');
>      type MVL_VECTOR is array (NATURAL range <>) of MVL;
>      constant c3: MVL_VECTOR := O"777";
> 
>      assert    c1'LENGTH = 12 and
>                c2'LENGTH = 12 and
>                c3 = "111111111";
> ```

diogratia, 2018-09-04:
String literals and bit string literals are separate lexical elements. Their value is subject to semantic interpretation.

String literal values is determined  according to 15.7  String literals:

>A string literal has a value that is a sequence of character values corresponding to the graphic characters of the string literal apart from the quotation mark itself. If a quotation mark value is to be represented in the sequence of character values, then a pair of adjacent quotation marks shall be written at the corresponding place within the string literal. (This means that a string literal that includes two adjacent quotation marks is never interpreted as two adjacent string literals.)

A bit string value is determined according 15.8 Bit string literals:

>A bit string literal has a value that is a string literal. The string literal is formed from the bit value by first obtaining a simplified bit value, consisting of the bit value with underline characters removed, and then obtaining an expanded bit value. Finally, the string literal value is obtained by adjusting the expanded bit value, if required.

(Obtaining the expanded bit value and adjusting it (for length) are described in the paragraphs following.)

It's dangerous to accept any implementation as being completely compliant with a particular revision of the VHDL standard. In 1995 the VHDL Technology Group (VHDL International) released a VHDL validation suite. It suffered from 'golden implementation bias' where a not quite compliant implementation was accepted as the authority.  They took every sentence of the standard separately marked as testable or un-testable and related testable sentences to test cases in a Microsoft Access Database. I'd take umbrage at a lot of sentences marked not testable personally (I extracted a lot of the records).

There hasn't been a publicly released effort since. The nebulousness of standard and revision compliance is useful from a marketing perspective and also demonstrates why vendors were historically responsive to interoperability issues. You're compliant as long as no one complains. It leaves a holistic understanding of the standard as the court of first resort for a non-portable design description.

tgingold, 2018-09-04:
Yes, GHDL is correct: underscore is not allowed in string for bit and logic type.

Paebbels, 2018-09-04:
Thanks. I relabeled from "Bug" to "LRM Reading".

closed 2018-09-04 by Paebbels