ghdl/ghdl #646: Crash when using string in record

1138-4EB, 2018-09-08:
```
#>> ent.vhd
entity ent is
end entity;

architecture a of ent is

  type tb_cfg_t is record
    value : string;
  end record tb_cfg_t;

  constant tb_cfg: tb_cfg_t := ( value => "hello" );

begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;

#>> sim.sh
ghdl -a ent.vhd
ghdl --elab-run ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

``` bash
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-271-g5cd3de46-dirty) [Dunoon edition]
Compiled with GNAT Version: 8.1.1 20180712 (Red Hat 8.1.1-5)
Target: x86_64-redhat-linux
In directory: /home/user/vhdl/test/py/
Command line:
/opt/ghdl-llvm/bin/ghdl1-llvm --workdir=/home/user/vhdl/test/py/vunit_out/ghdl/libraries/lib --work=lib --std=08 -P/home/user/vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/ -P/home/user/vhdl/test/py/vunit_out/ghdl/libraries/JSON/ -P/home/user/vhdl/test/py/vunit_out/ghdl/libraries/osvvm/ -P/home/user/vhdl/test/py/vunit_out/ghdl/libraries/lib/ -P/opt/ghdl-llvm/lib/ghdl/ieee/v08/ -P/opt/ghdl-llvm/lib/ghdl/ -c -o /home/user/vhdl/test/py/vunit_out/ghdl/libraries/lib/tb_axis_hsconv2.o /home/user/vhdl/test/py/vunit_out/preprocessed/lib/tb_axis_hsconv2.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs_utils.adb:1157
Call stack traceback locations:
0x7f6929171331 0x4cc42b 0x608974 0x617458 0x5b3470 0x5b3ac7 0x5b3bc6 0x5bd834 0x5a7938 0x627ef6 0x56dbd7 0x62e15a 0x62e9eb 0x63029a 0x408cb8 0x7f6928788119 0x407c28 0xfffffffffffffffe
******************************************************************
```

Paebbels, 2018-09-08:
It's a missing VHDL-2008 feature and related to many other issues regarding unconstrained or partially constrained types (especially records).

When compiling without the VUnit part, you are compiling as VHDL-93, right?

1138-4EB, 2018-09-08:
I expected it not to be supported, but I thought it would produce a error, not a crash.

You are ok, I didn't realize but I was compiling without `std=08` when not using VUnit.

Paebbels, 2018-11-22:
@tgingold is this also fixed by the latest commits regarding unconstrained records?

1138-4EB, 2018-11-23:
It is still failing, but with a different bug report:

``` bash
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-421-g69c5ab15) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
/usr/local/bin/ghdl --elab-run --std=08 --work=lib --workdir=/src/vunit_out/ghdl/libraries/lib -P/src/vunit_out/ghdl/libraries/vunit_lib -P/src/vunit_out/ghdl/libraries/lib tb_axis_hsconv2 tb -gtb_path=/src/ -grunner_cfg=active python runner : true,enabled_test_cases : test,output path : /src/vunit_out/test_output/lib.tb_axis_hsconv2.test_ed8652434df7168baf5b3a5e3bec756b7c038f20/,tb path : /src/,use_color : true --assert-level=error
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:49
Call stack traceback locations:
0x564c75b18267 0x564c75b27cec 0x564c75b0d572 0x564c75b0d977 0x564c75b0da80 0x564c75b15342 0x564c75af2897 0x564c75b3abda 0x564c75ad88b5 0x564c75b38b75 0x564c75ae371c 0x564c75b3dafc 0x564c75b465ae 0x564c75b44d99 0x564c75993d4a 0x564c758a26ab 0x7f3c1d4da2df 0x564c758a1478 0xfffffffffffffffe
******************************************************************
```

commit 8d517a18 (2018-12-08): sem_expr: add constraints to record aggregates.

Fix #646
closed 2018-12-08 by tgingold